Hieromon/AutoConnect #1: Fatal exception 3(LoadStoreErrorCause) when using Simple example (any example actually)

FlaviuN, 2018-03-27:
I was trying the library and it crashes for me using the any of the examples (Simple example used bellow) on every wifi connect. Any ideas? Can you help me use interpret the stack?

Thank you.

MY CONFIGURATION:

> Hardware: nodemcu v3 12e (Lolin)
> Core Version: 2.4.1 stable
> 
> Settings in IDE
> ide 1.8.5
> 
> Module: nodemcu 1.0 12e
> Flash Size: 4MB/1MB
> IwIP Variant: v2 low memory (tried high bandwith as well)
> CPU Frequency: 80Mhz
> Flash Frequency: 80Mhz
> Upload Using: SERIAL
> 
> AutoConnect version: 1.9.2
> PageBuilder: 1.0.1
> Example: Simple.ino

SKETCH
```
/*
  Simple.ino, Example for the AutoConnect library.
  Copyright (c) 2018, Hieromon Ikasamo
  https://github.com/Hieromon/AutoConnect

  This software is released under the MIT License.
  https://opensource.org/licenses/MIT
*/

#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <time.h>
#include <AutoConnect.h>

ESP8266WebServer Server;
AutoConnect      Portal(Server);

#define TIMEZONE    (3600 * 9)    // Tokyo
#define NTPServer1  "ntp.nict.jp" // NICT japan.
#define NTPServer2  "time1.google.com"

void rootPage() {
  String  content = 
    "<html>"
    "<head>"
    "<meta name=\"viewport\" content=\"width=device-width, initial-scale=1\">"
    "</head>"
    "<body>"
    "<h2 align=\"center\" style=\"color:blue;margin:20px;\">Hello, world</h2>"
    "<h3 align=\"center\" style=\"color:gray;margin:10px;\">{{DateTime}}</h3>"
    "<p style=\"padding-top:10px;text-align:center\">" AUTOCONNECT_LINK(COG_32) "</p>"
    "</body>"
    "</html>";
  static const char *wd[7] = { "Sun","Mon","Tue","Wed","Thr","Fri","Sat" };
  struct tm *tm;
  time_t  t;
  char    dateTime[26];

  t = time(NULL);
  tm = localtime(&t);
  sprintf(dateTime, "%04d/%02d/%02d(%s) %02d:%02d:%02d.",
    tm->tm_year + 1900, tm->tm_mon + 1, tm->tm_mday,
    wd[tm->tm_wday],
    tm->tm_hour, tm->tm_min, tm->tm_sec);
  content.replace("{{DateTime}}", String(dateTime));
[... 150 lines cut ...]
> 0x4020c3b0: AutoConnect::_classifyHandle(HTTPMethod, String) at /Users/user123/Documents/Arduino/libraries/AutoConnect/src/AutoConnect.cpp line 496
> 0x4020bf7f: std::_Function_handler    (AutoConnect*, std::_Placeholder1>, std::_Placeholder2>)> >::_M_invoke(std::_Any_data const&, HTTPMethod, String) at /Users/user123/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 601
> 0x40210510: String::String(String const&) at /Users/user123/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/WString.cpp line 41
> 0x4020bf48: std::_Function_handler    (AutoConnect*, std::_Placeholder1>, std::_Placeholder2>)> >::_M_invoke(std::_Any_data const&, HTTPMethod, String) at /Users/user123/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 2054
> 0x4020f040: PageBuilder::canHandle(HTTPMethod, String) at /Users/user123/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 2464
> 0x40210510: String::String(String const&) at /Users/user123/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/WString.cpp line 41
> 0x4020b8e1: ESP8266WebServer::_parseRequest(WiFiClient&) at /Users/user123/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WebServer/src/Parsing.cpp line 130
> 0x40210dd0: esp_yield() at /Users/user123/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/core_esp8266_main.cpp line 87
> 0x4020f008: PageBuilder::canHandle(HTTPMethod, String) at /Users/user123/Documents/Arduino/libraries/PageBuilder/src/PageBuilder.cpp line 45
> 0x4020a5ff: ESP8266WebServer::handleClient() at /Users/user123/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WebServer/src/ESP8266WebServer.cpp line 298
> 0x4020ee54: DNSServer::processNextRequest() at /Users/user123/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/DNSServer/src/DNSServer.cpp line 51
> 0x40210dd0: esp_yield() at /Users/user123/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/core_esp8266_main.cpp line 87
> 0x4020c80c: AutoConnect::handleClient() at /Users/user123/Documents/Arduino/libraries/AutoConnect/src/AutoConnect.cpp line 276
> 0x4020c986: AutoConnect::begin(char const*, char const*) at /Users/user123/Documents/Arduino/libraries/AutoConnect/src/AutoConnect.cpp line 139
> 0x40212124: std::_Function_base::_Base_manager ::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at /Users/user123/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 1934
> 0x40212110: std::_Function_handler ::_M_invoke(std::_Any_data const&) at /Users/user123/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 2069
> 0x4010020c: _umm_free at /Users/user123/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/umm_malloc/umm_malloc.c line 1291
> 0x4020c9bc: AutoConnect::begin() at /Users/user123/Documents/Arduino/libraries/AutoConnect/src/AutoConnect.cpp line 57
> 0x4020717b: setup() at /Users/user123/Documents/Arduino/libraries/AutoConnect/examples/Simple/Simple.ino line 55
> 0x40207054: rootPage() at /Users/user123/Documents/Arduino/libraries/AutoConnect/examples/Simple/Simple.ino line 22
> 0x40212124: std::_Function_base::_Base_manager ::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at /Users/user123/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 1934
> 0x40212110: std::_Function_handler ::_M_invoke(std::_Any_data const&) at /Users/user123/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 2069
> 0x40210e14: loop_wrapper() at /Users/user123/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/core_esp8266_main.cpp line 118
> 
>

Hieromon, 2018-03-28:
I have tested ESP-12E as NodeMCU V2, but no tested NodeMCU V3 as Lolin. So I would like to ask you some questions to solve this problem.

1. Your configuration is NodeMCU 1.0 12e and the Flash Frequency is 80MHz. There is no Flash Frequency for 80MHz in the choice of NodeMCU 1.0 with Arduino IDE 1.8.5.
So, I tried NodeMCU V2 as  Generic ESP8266 Module with the following configuration, but no problem occurred.
How did you select Flash Frequency 80MHz?

[image]

2. Stack dump says that this exception occurred when the captive portal was activated and tried to generate the first page. And the stack dump says too that this problem has occurred in the copy constructor of the String object of the argument indicating the page URL.
I think that it is probably pointing to an address that does not exist.
The location as an appropriate logic is Line 527 of AutoConnect.cpp.

AutoConnect.cpp:527:

if ((_currentPageElement = _setupPage(uri))! = nullptr) {
  _uri = String(uri);  <- exception?

A parameter "uri" is a String instance inherited from the handleClient() of ESP8266WebServer. It may have been an exception because the URI has expired and the heap has been freed.
Would you modify AutoConnect.cpp as follows and try again to identify the problem?

from AutoConnect.cpp:527

527: //_uri = String(uri);  <- comment out
528:   char buf[64];  <- insert
529:   uri.toCharArray(buf, 64); <- insert
530:   _uri = String(buf); <- insert

I would be happy if you let me know when things have changed.

FlaviuN, 2018-03-28:
1. Flash Frequency: 80Mhz is a mistake on my end (copy paste error). Sorry about that.

Here are my settings:
[image]

2. Made the suggested changes, but it still crashes

Using the same config from above with AutoConnect 1.9.3 (from GitHub) + suggested changes.

LOGS:

> method: GET url: /_ac search: 
> Fatal exception 3(LoadStoreErrorCause):
> epc1=0x40214224, epc2=0x00000000, epc3=0x00000000, excvaddr=0x402467wifi evt: 7
> 7e, depc=0x00000000
> 
> Exception (3):
> epc1=0x40214224 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4024677e depc=0x00000000
> 
> ctx: cont 
> sp: 3fff1690 end: 3fff1b60 offset: 01a0
> 
> >>>stack>>>
> 3fff1830:  3fff0a54 00000134 3fff1880 40210591  
> 3fff1840:  00000000 3fff0570 3fff1880 402105de  
> 3fff1850:  4024677e 3fff0570 3fff1870 402104b0  
> 3fff1860:  40100428 00000030 3fff2a04 4020e0dd  
> 3fff1870:  00000000 00000000 00000000 40210514  
> 3fff1880:  3fff2a24 0000000f 00000004 4021055f  
> 3fff1890:  3fff0a54 0000021d 3fff18d0 40210591  
> 3fff18a0:  3fff1940 3fff18d0 3fff18d0 40210698  
> 3fff18b0:  3ffea4a7 3fff0698 3fff0570 00000000  
> 3fff18c0:  3fff1940 3fff0698 3fff0570 4020c544  
> 3fff18d0:  3fff316c 0000000f 00000004 00000097  
> 3fff18e0:  40249a38 00000010 00000010 402019f8  
> 3fff18f0:  3ffea4a7 00000010 3fff19e4 00000001  
> 3fff1900:  3fff312c 00000010 3fff196c 40210514  
> 3fff1910:  3fff312c 00000004 3fff196c 4021055f  
> 3fff1920:  00000011 00000001 3fff196c 00000001  
> 3fff1930:  3fff0570 00000001 4020c4e8 4020c087  
> 3fff1940:  3fff314c 0000000f 00000004 402106c0  
> 3fff1950:  00000001 4020c050 3fff2574 4020f1f0  
> 3fff1960:  00000000 00000000 00000000 00000000  
> 3fff1970:  00000000 00000000 3fff06c4 402106c0  
> 3fff1980:  3ffea4a7 3fff0a78 3fff06c4 4020b9e9  
> 3fff1990:  3fff312c 0000000f 00000004 00000001  
> 3fff19a0:  3fff0670 00000001 00000008 00000000  
> 3fff19b0:  00000000 3fff1a20 3fff1a20 3ffe96e4  
> 3fff19c0:  40002514 3fffdd3c 3fff07a4 00000001  
> 3fff19d0:  3fff0670 00000001 3fff3d44 0000000f  
> 3fff19e0:  00000000 3fff3d24 0000000f 00000001  
> 3fff19f0:  3fff2e8c 0000000f 00000004 3fff3c84  
> 3fff1a00:  0000000f 00000003 3fff2d44 0000001f  
> 3fff1a10:  00000011 00000007 3fff0a78 40210075  
> 3fff1a20:  3fff06ec 00000000 3fff0708 00000006  
> 3fff1a30:  0000077f 4020f1b8 00000001 3fff2574  
> 3fff1a40:  00000000 3fff2e4c 3fff06c4 3fff0670  
> 3fff1a50:  00000001 3fff06ec 3fff06c4 4020a707  
> 3fff1a60:  40107378 00000000 00001388 4020f004  
> 3fff1a70:  00000000 3fff2e4c 40210fd0 3fff0b40  
> 3fff1a80:  00000001 3fff07a4 3fff0570 4020c96c  
> 3fff1a90:  00000001 3fff07a4 3fff0570 4020cb31  
> 3fff1aa0:  40107530 01f4a8c0 40107530 01f4a8c0  
> 3fff1ab0:  40107530 00000000 40107530 00000000  
> 3fff1ac0:  40107530 00000000 40212324 40212310  
> 3fff1ad0:  00000000 00000000 000003dc 4010020c  
> 3fff1ae0:  3fffdad0 00000000 3fff3f5c 3fff0b38  
> 3fff1af0:  3fffdad0 00000000 3fff0a78 4020cb6c  
> 3fff1b00:  feefeffe 00000000 3fff0a78 4020717b  
> 3fff1b10:  40207054 feefeffe 40212324 40212310  
[... 47 lines cut ...]
> 0x40210514: String::changeBuffer(unsigned int) at /Users/fnegrean/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/WString.cpp line 151
> 0x4021055f: String::reserve(unsigned int) at /Users/fnegrean/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/WString.cpp line 141
> 0x4020c4e8: AutoConnect::_classifyHandle(HTTPMethod, String) at /Users/fnegrean/Documents/Arduino/libraries/AutoConnect-0.9.3/src/AutoConnect.cpp line 513
> 0x4020c087: std::_Function_handler    (AutoConnect*, std::_Placeholder1>, std::_Placeholder2>)> >::_M_invoke(std::_Any_data const&, HTTPMethod, String) at /Users/fnegrean/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 601
> 0x402106c0: String::String(String const&) at /Users/fnegrean/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/WString.cpp line 41
> 0x4020c050: std::_Function_handler    (AutoConnect*, std::_Placeholder1>, std::_Placeholder2>)> >::_M_invoke(std::_Any_data const&, HTTPMethod, String) at /Users/fnegrean/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 2054
> 0x4020f1f0: PageBuilder::canHandle(HTTPMethod, String) at /Users/fnegrean/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 2464
> 0x402106c0: String::String(String const&) at /Users/fnegrean/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/WString.cpp line 41
> 0x4020b9e9: ESP8266WebServer::_parseRequest(WiFiClient&) at /Users/fnegrean/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WebServer/src/Parsing.cpp line 130
> 0x40210075: Print::write(char const*) at /Users/fnegrean/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/Print.h line 60
> 0x4020f1b8: PageBuilder::canHandle(HTTPMethod, String) at /Users/fnegrean/Documents/Arduino/libraries/PageBuilder/src/PageBuilder.cpp line 45
> 0x4020a707: ESP8266WebServer::handleClient() at /Users/fnegrean/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WebServer/src/ESP8266WebServer.cpp line 298
[... cut ...]
> 0x40212324: std::_Function_base::_Base_manager ::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at /Users/fnegrean/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 1934
> 0x40212310: std::_Function_handler ::_M_invoke(std::_Any_data const&) at /Users/fnegrean/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 2069
> 0x40211014: loop_wrapper() at /Users/fnegrean/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/core_esp8266_main.cpp line 118

FlaviuN, 2018-03-29:
Seems like the error occurs in PageElement* AutoConnect::_setupPage(String uri) inside AutoConnectPage.cpp.

After reducing the size of _PAGE_STAT (commenting about half the content), the error then happens on the next line.

Hieromon, 2018-03-30:
> Seems like the exception occurs in PageElement* AutoConnect::_setupPage(String uri) inside 

You are right. The stack dump I saw that says the crash point is the constructor of the string object.

I prepared a dedicated branch for the problem countermeasures: '**measures/\_setupPage\_crash**'.
I would like to you use the branch **measures/\_setupPage\_crash** and I will share it to modify as measures.

I uploaded an alternate source code with a change to clarify the root of the problem to "**measures/\_setupPage\_crash**" as AutoConnect.cpp.
I hope you try this modified code. But it is to identify where the bug is, not a formal fix.
My idea is that the problem is in the parameter passed to _setupPage, So I made the following modifications.
1. Trace the parameter as "**uri**" before _setupPage.
2. Discards PROGMEM of AutoConnectPage.cpp, the page templates allocate to RAM. (temporarily for this crash.)

This modification works fine for Simple.ino with my environment NodeMCU V2(ESP12e), Arduino IDE 1.8.5 for Win, Core 2.4.1. Debug Level is HTTP_CLIENT+HTTP_SERVER.
AutoConnect debug messages with the prefix "[AC]" will be displayed on the serial monitor.

FlaviuN, 2018-04-03:
Thanks again for looking into the issue.

I downloaded the code from the **measures/_setupPage_crash** branch.

Here are the latest logs:

> SDK:2.2.1(cfd48f3)/Core:2.4.1/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1)
> 
> bcn 0
> del if1
> mode : sta(dc:4f:22:1c:24:fd)
> [AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0) [AC] DHCP client(STARTED)
> [AC] WiFi.begin()
> [AC] Connecting....................................................................................................[AC] time out IP:0.0.0.0
> [AC] http server started
> mode : sta(dc:4f:22:1c:24:fd) + softAP(de:4f:22:1c:24:fd)
> add if1
> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
> bcn 100
> [AC] SoftAP esp8266ap/12345678 CH(1) H(0) IP:192.168.244.1
> [AC] DNS server started
> add 1
> aid 1
> station: cc:20:e8:32:c7:e1 join, AID = 1
> New client
> method: GET url: /hotspot-detect.html search: 
> [AC] /hotspot-detect.html
> [AC] __uri:/hotspot-detect.html
>  [AC] PageElement:0
> [AC] Page[] allocated
> headerName: Host
> headerValue: captive.apple.com
> headerName: Connection
> headerValue: close
> headerName: User-Agent
> headerValue: CaptiveNetworkSupport-355.50.1 wispr
> args: 
> Request: /hotspot-detect.html
>  Arguments: 
> request handler not found
> New client
> method: GET url: /_ac search: 
> [AC] captive.apple.com/_ac
> [AC] __uri:/_ac
>  [AC] PageElement:0x3fff2344
> [AC] Page[/_ac] allocated
> headerName: Host
> headerValue: 192.168.244.1
> headerName: Connection
> headerValue: close
> headerName: User-Agent
> headerValue: CaptiveNetworkSupport-355.50.1 wispr
> args: 
> Request: /_ac
>  Arguments: 
> [AC] 192.168.244.1/_ac
> Fatal exception 3(LoadStoreErrorCause):
> epc1=0x40213be0, epc2=0x00000000, epc3=0x00000000, excvaddr=0x40246665, depc=0x00000000
> 
> Exception (3):
> epc1=0x40213be0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40246665 depc=0x00000000
> 
> ctx: cont 
> sp: 3fff0b90 end: 3fff1060 offset: 01a0
> 
> >>>stack>>>
> 3fff0d30:  00000018 3fff48b8 3ffeff54 3fff0dd4  
> 3fff0d40:  40246665 3fff4684 00000018 401007ac  
> 3fff0d50:  3ffeff54 00000219 00000001 4020ef8a  
> 3fff0d60:  3fff3ae0 3fff48b8 3fff48b8 4020b97f  
[... 75 lines cut ...]
> 0x402100e8: String::~String() at /Users/fnegrean/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/WString.cpp line 120
> 0x4020f558: PageBuilder::_sink(int, ESP8266WebServer&) at /Users/fnegrean/Documents/Arduino/libraries/PageBuilder/src/PageBuilder.cpp line 90
> 0x4010020c: _umm_free at /Users/fnegrean/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/umm_malloc/umm_malloc.c line 1291
> 0x401009a8: free at /Users/fnegrean/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/umm_malloc/umm_malloc.c line 1739
> 0x4020f5e0: PageBuilder::handle(ESP8266WebServer&, HTTPMethod, String) at /Users/fnegrean/Documents/Arduino/libraries/PageBuilder/src/PageBuilder.cpp line 107
> 0x402102c8: String::operator=(String const&) at /Users/fnegrean/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/WString.cpp line 215
> 0x4020f5a4: PageBuilder::handle(ESP8266WebServer&, HTTPMethod, String) at /Users/fnegrean/Documents/Arduino/libraries/PageBuilder/src/PageBuilder.cpp line 101
> 0x40209e22: ESP8266WebServer::_handleRequest() at /Users/fnegrean/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WebServer/src/ESP8266WebServer.cpp line 590
> 0x4020fcad: Print::write(char const*) at /Users/fnegrean/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/Print.h line 60
> 0x4020ee0c: PageBuilder::canHandle(HTTPMethod, String) at /Users/fnegrean/Documents/Arduino/libraries/PageBuilder/src/PageBuilder.cpp line 45
> 0x4020fd9c: Print::println() at /Users/fnegrean/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/Print.cpp line 163
> 0x4020a090: ESP8266WebServer::handleClient() at /Users/fnegrean/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WebServer/src/ESP8266WebServer.cpp line 303
> 0x4020ec55: DNSServer::processNextRequest() at /Users/fnegrean/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/DNSServer/src/DNSServer.cpp line 51
> 0x40210bf0: esp_yield() at /Users/fnegrean/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/core_esp8266_main.cpp line 87
> 0x4020c480: AutoConnect::handleClient() at /Users/fnegrean/Documents/Arduino/libraries/AutoConnect-measures-_setupPage_crash/src/AutoConnect.cpp line 293
> 0x4020c7b2: AutoConnect::begin(char const*, char const*) at /Users/fnegrean/Documents/Arduino/libraries/AutoConnect-measures-_setupPage_crash/src/AutoConnect.cpp line 156
> 0x402102c8: String::operator=(String const&) at /Users/fnegrean/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/WString.cpp line 215
> 0x40210143: String::changeBuffer(unsigned int) at /Users/fnegrean/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/WString.cpp line 151
> 0x4010020c: _umm_free at /Users/fnegrean/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/umm_malloc/umm_malloc.c line 1291
> 0x4020c800: AutoConnect::begin() at /Users/fnegrean/Documents/Arduino/libraries/AutoConnect-measures-_setupPage_crash/src/AutoConnect.cpp line 57
[... cut ...]
> 0x40211c68: std::_Function_base::_Base_manager ::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at /Users/fnegrean/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 1934
> 0x40211c54: std::_Function_handler ::_M_invoke(std::_Any_data const&) at /Users/fnegrean/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 2069
> 0x40210c34: loop_wrapper() at /Users/fnegrean/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/core_esp8266_main.cpp line 118

Hieromon, 2018-04-05:
Thank you for the trial results. Still, the problem is String-related.
My idea is that the heap has been destroyed because the crash points move each time the code is changed.
And it always involves casting from const char to String.
Have you successfully executed an example of PageBuilder or ESP8266WebServer's examples with a large sketch size? It is advisable to completely erase the ESP8266 flash.
There are cases where a fatal exception (3) triggers in the state of Flash. Fatal exception (3) occurs when the processor detects a physical error for flash access.
https://www.esp8266.com/viewtopic.php?f=160&t=13020
https://arduino.stackexchange.com/questions/32819/fatal-exception-error-on-esp8266-arduino

I will investigate the std::vector that the problem is always involved in. If the problem is std:: vector or the memory leak, I think that the PageBuilder's examples will also crash.

Hieromon, 2018-04-10:
I don't have few guesses about the large sketch, but how about this one?
https://github.com/letscontrolit/ESPEasy

I am currently studying the possibility of memory leaks, but there is no problem yet. So, there are two possibilities I currently recognize.

1. AutoConnect bug
Some memory leaks occur in the interface between AutoConnect and PageBuiler, or the String pointer was destroyed.
If the problem cause is that case occurrence will not depend on a particular module. But this case does not occur in my environment IDE1.8.5, Core2.4.1 and other modules such as NodeMCU V2 and Heltec WiFi Kit 8. Also, it does not seem to happen below a certain size.

2. Failure of the individual module you have
An exception occurs when accessing a specific flash address or RAM address.

FlaviuN, 2018-04-10:
Hi there,

I tried https://github.com/letscontrolit/ESPEasy and it works without any problems. Connected to the AP, captivate triggered, provided the credentials and esp8266 connected to my local wifi. Then I was able to load the full ESP_easy mega page on the local network without any problems.

I have multiple NodeMCU v3 units (6 units) and they all present the same problem.
I have ordered a couple of Wemos d1 minis. They should arrive soon and will be testing with those as well and let you know the outcome.

Thanks

Hieromon, 2018-04-11:
Ok, I ordered NodeMCU V3 Lolin w/CH430 USB bridge and WeMos D1 mini, too. It will arrive from China after three weeks.
Nodemcu officially does not specify V3, It seems that USB serial bridge changed CP2102 to CH430 and pins out for USB power out and the other for an additional GND. But the difference will not matter much for this problem, I think.

Thank you for cooperating for solving.

Hieromon, 2018-04-12:
I made some modifications to raise the trace density of the crash points.
Perhaps it may be possible to catch the problem addresses of the flash with this modified trace.
Please try this one. The revision code of the trace was uploaded to the measures/_setupPage_crash branch.
I also added a countermeasure branch to PageBuilder. To execute the new trace revision, please clone the trace code from the PageBulder measures/AutoConnect_crash branch.

Hieromon, 2018-04-13:
Thank you for your results report.
The moved crash point appears to be occurring in EEPROM.begin of AutoConnectCredential::_allocateEntry.
Revised trace code excluded PROGMEM, and this time it occurred in EEPROM.begin. After all the problem seems to be flash access.
Flash memory must be read using 32bit aligned addresses else a processor exception will be triggered order within the 32bit values are. Perhaps the immediate cause is this. But I do not yet make sure the why it does not occur in my module and it happens in your module. It may be a difference in the compilation environment rather than the execution module difference.
So, you can confirm the flash chip ID with esptool.py and I would like to know which flash chip it is.

> esptool.py --port ESP8266_PORT flash_id

Flash chip ID table: https://github.com/jhcloos/flashrom/blob/master/flashchips.h

Anyway, please try the menu "Configure New AP". If the expected cause is correct, I think that an exception will occur at the storing of EEPROM the SSID that was established after the connection.

I updated the trace code slightly, so please update your branch measures/_setupPage_crash for AutoConnect and branch measures/AutoConnect_crash for PageBuilder.

Hieromon, 2018-04-21:
NodeMCU Lolin and Wemos d1 mini arrived at my hand. I tested both modules and they worked fine.
The results of previous tests say that this problem is related to your environment.
Have you tested with the WeMos that ordered you said?
And did you successfully clear all of the flash and read the flash chip ID using with esptool?
I hope you will report the results.

[image]

FlaviuN, 2018-04-24:
Sorry for the lack of communication lately. I was away for a few days.
I have not received the wemos yet. Should arrive soon.
I am glad you are not seeing the problem. Maybe my setup is problematic.

I will perform the suggested tests and capture the chip id in the following days.

Thanks again.

FlaviuN, 2018-05-10:
My appologies for the long pause in communications.

In the mean time I have cleared and reinstalled all arduino related tools on my mac in an effort to solve any environment related problems.

Here are my results for esptool.py --port ESP8266_PORT flash_id:

> esptool.py v2.3.1
> Connecting....
> Detecting chip type... ESP8266
> Chip is ESP8266EX
> Features: WiFi
> Uploading stub...
> Running stub...
> Stub running...
> Manufacturer: c8
> Device: 4016
> Detected flash size: 4MB
> Hard resetting via RTS pin...

I didn't see an exception when accessing "Configure New AP".
I just received the Wemos D1 so I will test with that tomorrow.

> [AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0) [AC] DHCP client(STARTED)
> [AC] WiFi.begin()
> [AC] Connecting....................................................................................................[AC] time out IP:0.0.0.0
> [AC] http server started
> [AC] SoftAP esp8266ap/12345678 CH(1) H(0) IP:192.168.244.1
> [AC] DNS server started
> New client
> method: GET url: /hotspot-detect.html search: 
> [AC] /hotspot-detect.html
> [AC] __uri:/hotspot-detect.html
>  [AC] PageElement:0
> [AC] Page[] allocated
> [AC] After elements:0
> headerName: Host
> headerValue: captive.apple.com
> headerName: Connection
> headerValue: close
> headerName: User-Agent
> headerValue: CaptiveNetworkSupport-355.50.1 wispr
> args: 
> Request: /hotspot-detect.html
>  Arguments: 
> request handler not found
> New client
> method: GET url: /_ac search: 
> [AC] captive.apple.com/_ac
> [AC] __uri:/_ac
>  [AC] PageElement:0x3fff617c
> [AC] Mold:{{HEAD}}<title>AutoConnect statistics</title><style type="text/css">{{CSS_BASE}}{{CSS_TABLE}}{{CSS_LUXBAR}}</style></head><body style="padding-top:58px;"><div class="container">{{MENU}}<div><table class="info" style="border:none;"><tbody><tr><td>Established connection</td><td>{{ESTAB_SSID}}</td></tr><tr><td>Mode</td><td>{{WIFI_MODE}}({{WIFI_STATUS}})</td></tr><tr><td>IP</td><td>{{LOCAL_IP}}</td></tr><tr><td>GW</td><td>{{GATEWAY}}</td></tr><tr><td>Subnet mask</td><td>{{NETMASK}}</td></tr><tr><td>SoftAP IP</td><td>{{SOFTAP_IP}}</td></tr><tr><td>AP MAC</td><td>{{AP_MAC}}</td></tr><tr><td>STA MAC</td><td>{{STA_MAC}}</td></tr><tr><td>Channel</td><td>{{CHANNEL}}</td></tr><tr><td>dBm</td><td>{{DBM}}</td></tr><tr><td>Chip ID</td><td>{{CHIP_ID}}</td></tr><tr><td>CPU Freq.</td><td>{{CPU_FREQ}}MHz</td></tr><tr><td>Flash size</td><td>{{FLASH_SIZE}}</td></tr><tr><td>Free memory</td><td>{{FREE_HEAP}}</td></tr></tbody></table></div></div></body></html>
> [AC] Token(0):HEAD
> [AC] Token(1):CSS_BASE
> [AC] Token(2):CSS_TABLE
> [AC] Token(3):CSS_LUXBAR
> [AC] Token(4):MENU
> [AC] Token(5):ESTAB_SSID
> [AC] Token(6):WIFI_MODE
> [AC] Token(7):WIFI_STATUS
> [AC] Token(8):LOCAL_IP
> [AC] Token(9):SOFTAP_IP
> [AC] Token(10):GATEWAY
> [AC] Token(11):NETMASK
> [AC] Token(12):AP_MAC
> [AC] Token(13):STA_MAC
> [AC] Token(14):CHANNEL
> [AC] Token(15):DBM
> [AC] Token(16):CPU_FREQ
> [AC] Token(17):FLASH_SIZE
> [AC] Token(18):CHIP_ID
[... 453 lines cut ...]
> headerValue: http://192.168.244.1/_ac/config
> headerName: Connection
> headerValue: keep-alive
> headerName: Upgrade-Insecure-Requests
> headerValue: 1
> headerName: Accept
> headerValue: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
> headerName: Accept-Language
> headerValue: en-ca
> headerName: User-Agent
> headerValue: Mozilla/5.0 (iPhone; CPU iPhone OS 11_3 like Mac OS X) AppleWebKit/605.1.15 (KHTML, like Gecko) Mobile/15E302
> headerName: DNT
> headerValue: 1
> headerName: Accept-Encoding
> headerValue: gzip, deflate
> args: 
> Request: /_ac/fail
>  Arguments: 
> [AC] 192.168.244.1/_ac/fail
> [AC] /_ac/fail exists, PageBuilder uri:/_ac/fail
> [AC] Elements:1
> [AC] Mold(0):{{HEAD}}<title>AutoConnect statistics</title><style type="text/css">{{CSS_BASE}}{{CSS_TABLE}}{{CSS_LUXBAR}}</style></head><body style="padding-top:58px;"><div class="container">{{MENU}}<div><table class="info" style="border:none;"><tbody><tr><td>Connection Failed</td><td>{{STATION_STATUS}}</td></tr></tbody></table></div></div></body></html>
> 
> ```
>

Hieromon, 2018-05-11:
Thank you for your report.
The flash memory chip equipped is the same as my owned model and the HTML code seems to be generated correctly.
The meaning the menu action of the "Configure new AP" worked fine is that the access to the flash worked. Even master branch code with HTML modal placed in PROGMEM may work correctly.
I updated the library in the last few days. The latest version is 0.9.4.
It is worth returning your PageBuilder to the master branch and trying it with the latest AutoConnect 0.9.4.

guru-florida, 2018-10-12:
I am having the same issue. Has this been resolved?

FYI. I read the entire thread and I see there was a lot of effort between the two of you. If there is any way I can help debug let me know. I did enable AC_DEBUG and the issue seems to be around the same _classifyHandle() function.

guru-florida, 2018-10-12:
FYI. I checked out the <AutoConnect>measures/_setupPage_crash and <PageBuilder>/measures/AutoConnect_crash repos and it works. The only issue is the ESP8266 didn't automatically restart. I got the "wet reset" line in the debug output but no reset. Hit the physical reset button and it came up with the established credentials.

Hieromon, 2018-10-16:
Hi guru-florida, thank you cooperation.
This problem is not resolved yet, but your exam results are very informative. The problem still seems to be related to PROGMEM.
I may not fully understand the PROGMEM implementation of ESP8266, and I have not been reached to the cause. If you are familiar with the PROGMEM implementation of ESP8266, I welcome any guides.

ageurtse, 2018-12-16:
Hello, 

Trying to help out, i'm running a macbook with Sloeber (arduino eclips development)
running the above code and i have direct Exception (3).

please need help to resolve it.

ageurtse, 2018-12-16:
After googling around, here is the solution for the above piece of code:
[https://devask.nl/questions/51531696/progmem-error-on-macos-high-sierra-arduino-mac-185](https://devask.nl/questions/51531696/progmem-error-on-macos-high-sierra-arduino-mac-185
)

Now find out how to solve this in your lib.

ageurtse, 2018-12-17:
I did try that yesterday, just before posting the link with the solution. 
With the FPSTR in front the code run's fine. 

But what if the string isn't in progmem, that is not what i tryed, i'm now unable to try of work.

Hieromon, 2018-12-17:
Ok, If FPSTR is missed, 4-byte alignment for the flash string is not performed. My environment that is not Mac seems to have been very lucky.
I posted measures to the branch as [measures/AutoConnect_crash](https://github.com/Hieromon/PageBuilder/tree/measures/AutoConnect_crash) and [measures/_setupPage_crash](https://github.com/Hieromon/AutoConnect/tree/measures/_setupPage_crash), but I can not confirm it in my lucky environment.

ageurtse, 2018-12-17:
I will try, just use this brance and try to run one of the examples.

ageurtse, 2018-12-17:
Here is my decoded stack dump:

```
0x40201b70: realloc_loc at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/heap.c line 140
0x4021163c: String::operator=(__FlashStringHelper const*) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 252
0x4010020c: _umm_free at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/umm_malloc/umm_malloc.c line 1295
0x401009e4: free at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/umm_malloc/umm_malloc.c line 1755
0x402115f0: String::String(unsigned int, unsigned char) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 86
0x4020ff83: AutoConnect::_setupPage(String) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/xtensa-lx106-elf/include/c++/4.8.2/functional line 1336
0x4021160c: String::String(unsigned int, unsigned char) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 90
0x4020ffbe: AutoConnect::_setupPage(String) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/xtensa-lx106-elf/include/c++/4.8.2/tuple line 140
0x4021163c: String::operator=(__FlashStringHelper const*) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 252
0x40211687: String::concat(char const*, unsigned int) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 287
0x402116b9: String::concat(String const&) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 262
0x402117c0: operator+(StringSumHelper const&, String const&) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 365
0x4020e082: AutoConnect::end() at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/libraries/AutoConnect/0.9.5/src/AutoConnect.cpp line 241
0x4021163c: String::operator=(__FlashStringHelper const*) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 252
0x40211687: String::concat(char const*, unsigned int) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 287
0x4020e008: AutoConnect::_stopPortal() at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/libraries/AutoConnect/0.9.5/src/AutoConnect.cpp line 427
0x4020d973: AutoConnect::_loadAvailCredential() at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/libraries/AutoConnect/0.9.5/src/AutoConnect.cpp line 404
0x402117e8: operator+(StringSumHelper const&, char const*) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 372
0x4020d93c: AutoConnect::_loadAvailCredential() at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/libraries/AutoConnect/0.9.5/src/AutoConnect.cpp line 399
0x40207670: PageBuilder::canHandle(HTTPMethod, String) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/xtensa-lx106-elf/include/c++/4.8.2/functional line 2464
0x402117e8: operator+(StringSumHelper const&, char const*) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 372
0x4020ce69: ESP8266WebServer::_parseRequest(WiFiClient&) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WebServer/src/Parsing.cpp line 56
0x40207638: PageBuilder::canHandle(HTTPMethod, String) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/libraries/PageBuilder/1.2.0/src/PageBuilder.cpp line 52
0x4020bb87: ESP8266WebServer::urlDecode(String const&) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WebServer/src/Parsing.cpp line 583
0x4020d79c: AutoConnectConfig::AutoConnectConfig() at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/libraries/AutoConnect/0.9.5/src/AutoConnect.h line 139
0x4021219c: std::_Sp_counted_deleter  ::_Deleter   >, std::allocator , (__gnu_cxx::_Lock_policy)0>::~_Sp_counted_deleter() at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/xtensa-lx106-elf/include/c++/4.8.2/bits/shared_ptr_base.h line 343
0x4020e564: AutoConnect::begin(char const*, char const*, unsigned long) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/libraries/AutoConnect/0.9.5/src/AutoConnect.cpp line 142
0x4020e931: AutoConnect::_handleNotFound() at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/libraries/PageBuilder/1.2.0/src/PageBuilder.h line 114
[... cut ...]
0x402135d4: EEPROMClass::write(int, unsigned char) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/libraries/EEPROM/EEPROM.cpp line 110
0x40212220: SPIFFSImpl::info(fs::FSInfo&) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/spiffs_api.h line 99
```

other info:
SDK:2.2.1(cfd48f3)/Core:2.4.2/lwIP:1.4.0rc2/BearSSL:6d1cefc

as library i use https://github.com/Hieromon/PageBuilder/tree/measures/AutoConnect_crash
i hope this helps out, if not, please give a detail instruction what to do.

Hieromon, 2018-12-18:
Sorry, I was leaking code to fix. I posted revise the fix to the branch [measures/_setupPage_crash](https://github.com/Hieromon/AutoConnect/tree/measures/_setupPage_crash).

ageurtse, 2018-12-18:
After booting i get an exception 9, i now have to work tonight i will try a bit more.

ageurtse, 2018-12-18:
Here is the error that i get:

New client
:ref 2
:ur 2
method: GET url: /_ac search: 
[AC] captive.apple.com/_ac
Fatal exception 3(LoadStoreErrorCause):
epc1=0x40215ab8, epc2=0x00000000, epc3=0x00000000, excvaddr=0x4024ad99, depc=0x00000000

Exception (3):
epc1=0x40215ab8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4024ad99 depc=0x00000000

ctx: cont 
sp: 3ffffa90 end: 3fffffd0 offset: 01a0

and the decoded stack:
`Decoding stack results
0x40201b70: realloc_loc at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/heap.c line 140
0x4021163c: String::changeBuffer(unsigned int) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 156
0x4010020c: _umm_free at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/umm_malloc/umm_malloc.c line 1295
0x401009e4: free at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/umm_malloc/umm_malloc.c line 1755
0x402115f0: String::invalidate() at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 141
0x4020ff83: AutoConnect::_setupPage(String) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/libraries/AutoConnect/0.9.5/src/AutoConnectPage.cpp line 1007
0x4021160c: String::~String() at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 125
0x4020ffbe: AutoConnect::_setupPage(String) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/xtensa-lx106-elf/include/c++/4.8.2/functional line 1336
0x4021163c: String::changeBuffer(unsigned int) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 156
0x40211687: String::reserve(unsigned int) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 146
0x402116b9: String::copy(char const*, unsigned int) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 175
0x402117c0: String::operator=(String const&) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 220
0x4020e082: AutoConnect::_classifyHandle(HTTPMethod, String) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/libraries/AutoConnect/0.9.5/src/AutoConnect.cpp line 564
0x4021163c: String::changeBuffer(unsigned int) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 156
0x40211687: String::reserve(unsigned int) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 146
0x4020e008: AutoConnect::_classifyHandle(HTTPMethod, String) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/libraries/AutoConnect/0.9.5/src/AutoConnect.cpp line 551
0x4020d973: std::_Function_handler    (AutoConnect*, std::_Placeholder1>, std::_Placeholder2>)> >::_M_invoke(std::_Any_data const&, HTTPMethod, String) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/xtensa-lx106-elf/include/c++/4.8.2/functional line 601
0x402117e8: String::String(String const&) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 41
0x4020d93c: std::_Function_handler    (AutoConnect*, std::_Placeholder1>, std::_Placeholder2>)> >::_M_invoke(std::_Any_data const&, HTTPMethod, String) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/xtensa-lx106-elf/include/c++/4.8.2/functional line 2054
0x40207670: PageBuilder::canHandle(HTTPMethod, String) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/xtensa-lx106-elf/include/c++/4.8.2/functional line 2464
0x402117e8: String::String(String const&) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 41
0x4020ce69: ESP8266WebServer::_parseRequest(WiFiClient&) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WebServer/src/Parsing.cpp line 130
0x40207638: PageBuilder::canHandle(HTTPMethod, String) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/libraries/PageBuilder/1.2.0/src/PageBuilder.cpp line 52
0x4020bb87: ESP8266WebServer::handleClient() at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WebServer/src/ESP8266WebServer.cpp line 298
0x4020d79c: DNSServer::processNextRequest() at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/libraries/DNSServer/src/DNSServer.cpp line 51
0x4021219c: esp_yield() at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/core_esp8266_main.cpp line 91
[... cut ...] lf-gcc/2.5.0-2-59d892c8/xtensa-lx106-elf/include/c++/4.8.2/functional line 1934
0x402135d4: std::_Function_handler ::_M_invoke(std::_Any_data const&) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/xtensa-lx106-elf/include/c++/4.8.2/functional line 2069
0x40212220: loop_wrapper() at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/core_esp8266_main.cpp line 122
`

Hieromon, 2018-12-19:
Thank you for feedback. After your confirmation with the sloeber, I include the fix to the master of v0.9.7 as next version.

ageurtse, 2018-12-19:
I'm having trouble installing the plugin within Sloeber, try it later again. 
If some one could help out on how to install the library the correct way it could be fine.

ageurtse, 2018-12-19:
Back home, and finaly managed to install the library 

but still having a exception 3 error. 

here is the log:

SDK:2.2.1(cfd48f3)/Core:2.4.2/lwIP:1.4.0rc2/BearSSL:6d1cefc
 
bcn 0
del if1
usl
mode : sta(5c:cf:7f:53:8d:2b)
add if0
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0) [AC] DHCP client(STARTED)
[AC] WiFi.begin()
[AC] Connecting....................................................................................................[AC] time out IP:0.0.0.0
[AC] http server started
mode : sta(5c:cf:7f:53:8d:2b) + softAP(5e:cf:7f:53:8d:2b)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
bcn 0
del if1
add if1
dhcp server start:(ip:192.168.244.1,mask:255.255.255.0,gw:192.168.244.1)
bcn 100
[AC] SoftAP esp8266ap/12345678 CH(1) H(0) IP:192.168.244.1
[AC] DNS server started
add 1
aid 1
station: b0:19:c6:e7:06:d9 join, AID = 1
[AC] /hotspot-detect.html
[AC] Page[] allocated
[AC] captive.apple.com/_ac
[AC] Page[/_ac] allocated
[AC] 192.168.244.1/_ac
[AC] /_ac already allocated
Fatal exception 3(LoadStoreErrorCause):
epc1=0x40214864, epc2=0x00000000, epc3=0x00000000, excvaddr=0x4024a89e, depc=0x00000000

Exception (3):
epc1=0x40214864 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4024a89e depc=0x00000000

ctx: cont 
sp: 3ffffa40 end: 3fffffd0 offset: 01a0
```
Decoding stack results
0x402106fb: String::changeBuffer(unsigned int) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 156
0x402106fb: String::changeBuffer(unsigned int) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 156
0x40210747: String::reserve(unsigned int) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 146
0x40210779: String::copy(char const*, unsigned int) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 175
0x402107c6: String::String(char const*) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 36
0x401009e4: free at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/umm_malloc/umm_malloc.c line 1755
0x4020e86a: AutoConnect::_token_HEAD(PageArgument&) at /Users/ageurtse/Arduino/libraries/AutoConnect-measures-_setupPage_crash/src/AutoConnectPage.cpp line 751
0x40212841: std::_Function_handler    (AutoConnect*, std::_Placeholder1>)> >::_M_invoke(std::_Any_data const&, PageArgument&) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/xtensa-lx106-elf/include/c++/4.8.2/functional line 2058
0x4020770e: PageElement::build(char const*, std::vector   >, PageArgument&) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.h line 117
0x4020ce43: std::_Function_base::_Base_manager    (AutoConnect*, std::_Placeholder1>)> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/xtensa-lx106-elf/include/c++/4.8.2/functional line 1910
0x402107c6: String::String(char const*) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 36
0x40207ae5: PageBuilder::build(PageArgument&) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.h line 117
0x4020b755: ESP8266WebServer::_parseArguments(String) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WebServer/src/Parsing.cpp line 283
0x40207cff: PageBuilder::_sink(int, ESP8266WebServer&) at /Users/ageurtse/Arduino/libraries/PageBuilder-master/src/PageBuilder.cpp line 97
0x402106b8: String::invalidate() at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 141
0x402106d4: String::~String() at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 125
0x4020d3c0: AutoConnect::_classifyHandle(HTTPMethod, String) at /Users/ageurtse/Arduino/libraries/AutoConnect-measures-_setupPage_crash/src/AutoConnect.cpp line 513
0x4020ced2: std::_Function_handler    (AutoConnect*, std::_Placeholder1>, std::_Placeholder2>)> >::_M_invoke(std::_Any_data const&, HTTPMethod, String) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/xtensa-lx106-elf/include/c++/4.8.2/functional line 2058
0x401009e4: free at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/umm_malloc/umm_malloc.c line 1755
0x402106b8: String::invalidate() at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 141
0x40207e98: PageBuilder::handle(ESP8266WebServer&, HTTPMethod, String) at /Users/ageurtse/Arduino/libraries/PageBuilder-master/src/PageBuilder.cpp line 143
[... cut ...] gcc/2.5.0-2-59d892c8/xtensa-lx106-elf/include/c++/4.8.2/functional line 1934
0x402123a8: std::_Function_handler ::_M_invoke(std::_Any_data const&) at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/xtensa-lx106-elf/include/c++/4.8.2/functional line 2069
0x402112d0: loop_wrapper() at /Applications/Sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/core_esp8266_main.cpp line 122

```

ageurtse, 2018-12-19:
Even in Arduino it won't run anymore, don't know why !?

This is what i get from arduino.
....................................................................[AC] time out IP:0.0.0.0
[AC] http server started
[AC] SoftAP esp8266ap/12345678 CH(1) H(0) IP:192.168.244.1
[AC] DNS server started
[AC] /hotspot-detect.html
[AC] Page[] allocated
[AC] captive.apple.com/_ac
[AC] Page[/_ac] allocated
[AC] 192.168.244.1/_ac
[AC] /_ac already allocated

Exception (3):
epc1=0x4021420c epc2=0x00000000 epc3=0x00000000 excvaddr=0x4024a079 depc=0x00000000

ctx: cont 
sp: 3ffffa50 end: 3fffffd0 offset: 01a0

....................................................................[AC] time out IP:0.0.0.0
[AC] http server started
[AC] SoftAP esp8266ap/12345678 CH(1) H(0) IP:192.168.244.1
[AC] DNS server started
[AC] /hotspot-detect.html
[AC] Page[] allocated
[AC] captive.apple.com/_ac
[AC] Page[/_ac] allocated
[AC] 192.168.244.1/_ac
[AC] /_ac already allocated

Exception (3):
epc1=0x4021420c epc2=0x00000000 epc3=0x00000000 excvaddr=0x4024a079 depc=0x00000000

ctx: cont 
sp: 3ffffa50 end: 3fffffd0 offset: 01a0

Hieromon, 2018-12-19:
Did the measures work fine once in Arduino, yesterday?

ageurtse, 2018-12-19:
Yes, just prior of the post it works fine. 
but know compiling goes wel, but running te sketch gives the exception(3) error.

Hieromon, 2018-12-19:
I'm not familiar with the sloeber, I was just looking for clues from the sloeber' RP issue.
Generation of board parameters at compile time seems to be wrong.
https://github.com/Sloeber/arduino-eclipse-plugin/issues/1047

... I thought it, but you be the success.

Thank you so much, I appreciate your devotional help.

ageurtse, 2018-12-19:
Well, take a look at sloeber, it's a great product. 
as far as i can see far better then arduino. 

next step is custom parameters :)

Hieromon, 2018-12-19:
This topic will close when merged with v0.9.7. Custom parameter handling support will be done soon by v0.9.7.

closed 2019-03-01 by Hieromon

OttoTrezza, 2020-04-16:
just put a delay(100) into the main loop!