Hieromon/AutoConnect #2: Example for mqtt sucribe

cruizg, 2018-04-16:
Please, I'm doing the example of mqttRSSI, but I need to do some action in the esp8266, for that I have to subscribe but your example is only to publish.
Best Regards

Hieromon, 2018-04-17:
If ESP8266 is just a WiFi client, AutoConnect is just a substitute for WiFi.begin. It does not affect mqtt subscribe and simply implements the callback function according to a common method.
In order to use PubSubClient library with subscribing, implement a call back function and specify it in the constructor of PubSubClient.
For example, to quote [an example contained in the PubSubClient library](https://github.com/knolleary/pubsubclient/blob/master/examples/mqtt_publish_in_callback/mqtt_publish_in_callback.ino), exclude the EthernetClient declaration, define the AutoConnect variable, and execute AutoConnect::begin instead of Ethernet.begin.

cruizg, 2018-04-21:
Very Thanks

closed 2018-04-21 by cruizg