Hieromon/AutoConnect #3: cursor position

cruizg, 2018-04-21:
When I select the SSID, the cursor should be positioned in the text box of the key, because if we have more than 10 networks (for example), we would have to scroll to write the key but that would not be necessary if the function for example: document.getElementById ('password'). focus ();

cruizg, 2018-04-21:
a question about it, if I want to turn off the AP mode, how could I do it?

Hieromon, 2018-04-22:
Even if the connection succeeds with the first WiFi.begin() in AutoConnect, if it is different from the expected SSID, do you want to start the reconfigure the connection?

Also about autofocus to the password field, I already improved the library and It will be released soon as 0.9.4.

cruizg, 2018-04-22:
I have an esp8266 module and I configure it with your library for the AP that exposes, after configuring it it does not make sense for the AP to continue active, given that someone could access that AP and change some configuration that's why I think that after configure the AP this disappears has the module esp8266 already be connected to WiFi, but if for some reason the esp8266 is restarted or the WiFi key is changed the esp8266 should have the ability with your Liberia to be able to enable the AP again to reconfigure it .

cruizg, 2018-04-22:
Cambiar a español
Another thing your module when I start the AP I notice that it is good and it works but I think you should have the way to start with a dynamic SSID (for example the Mac or chip name) as well as start it with a key or without a password, that way the bookstore WiFi anafe, eh opted for you Liberia because in my opinion it is better developed. About autofocus, very very thanks.

Hieromon, 2018-04-22:
I have no objection to improving the library, but clear specifications are necessary.
I feel that you have an important idea about the AutoConnect AP mode when starting the captive portal.
But please make one topic with one issue thread. If the theme is compounded it will be confusing.
I organized it as follows.

1. Specify dynamically determined SSID without Password when AutoConnect captive portal started. issue #4.
2. Please, you explain more clearly for turning off the AP mode at AutoConnect started.

I posted for the dynamic SSID with issue #4 and you can revise the requirement in the thread. Please, you post the "Turning off the AP" with a new issue.
A branch is made for each enhancement. Its issue whose improvement has been completed will be merged into the master branch and closed.

An enhancement of this issue as autofocus to the password field has been pushed to the branch "feature/autofocus_password_field" and you can try this one.
If there is no problem I will merge it into the master branch.

commit b0d8f0f0 (2018-04-22): resolve #3
closed 2018-05-06 by Hieromon