Hieromon/AutoConnect #4: Specify dynamically determined SSID

Hieromon, 2018-04-22:
This topic from #3 .
> Another thing your module when I start the AP I notice that it is good and it works but I think you should have the way to start with a dynamic SSID (for example the Mac or chip name) as well as start it with a key or without a password, that way the bookstore WiFi anafe, eh opted for you Liberia because in my opinion it is better developed.

Hieromon, 2018-04-22:
Hi, cruizg

I posted this issue on behalf of you.
About Dynamic SSID, my understanding is as follows. You should correct it if necessary.

When AutoConnect launches Captive Portal, the SSID as an access point is fixed to "esp8266ap/12345678", but it can be determined dynamically based on chip ID etc, and the dynamic SSID has no password.
Is it right?

Hieromon, 2018-04-24:
Incidentally, for dynamically determined SSID from a chip ID, it can be realized by the following implementation without changing the AutoConnect library current version 0.9.3.

```arduino
AutoConnect  Portal;
AutoConnectConfig  ACconfig;

void setup() {
  // Determine AP name from MAC of the ESP8266 module.
  // It provides "esp{three digits of MAC}free" without password as AP name.
  ACconfig.apid = String("esp") + String(ESP.getChipId(), HEX) + String("free");
  ACconfig.psk = String("");
  Portal.config(ACconfig);
  Portal.begin();
}
```

cruizg, 2018-04-27:
Yes, is correct

Hieromon, 2018-04-28:
Ok, Since the request is already supported in the current version, the issue is closed.

closed 2018-04-28 by Hieromon