Hieromon/AutoConnect #5: Turning off the AP

cruizg, 2018-04-27:
When the device (esp8266) has already connected to the Wi-Fi network, the AP mode should be turned off, since some intruder could connect to the AP and change the configuration of the device (Esp8266).
When the key has been changed in the modem the AP should start again in order to update the key and after the connection is established it should be switched off again

Hieromon, 2018-05-01:
Hello, cruizg

I was reviewing your enhancement request few days, but there is a point as not well for me.
AutoConnect has established a connection once, it will not be in AP mode after the second time for the same access point in the WLAN. However, it is necessary to reset the module from AutoConnect menu or manually.
And if the key of the access point saved previously by ESP8266 has been changed, start AP mode appears in association with STA mode. This is such the specification of AutoConnect current version.
Is your request different from this behavior?

cruizg, 2018-05-04:
yes , is correct.

Hieromon, 2018-05-06:
I supported the option to reconnect to a known access point in conjunction with issue #6.
This feature might be useful for your situations. Pls refer https://hieromon.github.io/AutoConnect/advancedusage/index.html#automatic-reconnect.

Hieromon, 2018-08-27:
Did the problem be solved by the support of AutoReconnect function?
If so, please close this issue.

Hieromon, 2018-09-24:
Since there was no response even after one month passed, closed this issue as enhanced.

closed 2018-09-24 by Hieromon