Hieromon/AutoConnect #9: #define max issue

sspence65, 2018-06-20:
Arduino: 1.8.5 (Windows 10), Board: "Adafruit HUZZAH ESP8266, 80 MHz, 115200, 4M (3M SPIFFS)"

In file included from C:\Users\xxx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClient.h:25:0,

                 from C:\Users\xxx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:39,

                 from C:\Users\xxx\Arduino\libraries\AutoConnect\src\AutoConnectPage.cpp:10:

c:\users\xxx\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_vector.h: In member function 'std::vector<_Tp, _Alloc>::size_type std::vector<_Tp, _Alloc>::_M_check_len(std::vector<_Tp, _Alloc>::size_type, const char*) const':

C:\Users\xxx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Arduino.h:253:18: error: expected unqualified-id before '(' token

 #define max(a,b) ((a)>(b)?(a):(b))

                  ^

c:\users\xxx\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_bvector.h: In member function 'std::vector<bool, _Alloc>::size_type std::vector<bool, _Alloc>::_M_check_len(std::vector<bool, _Alloc>::size_type, const char*) const':

C:\Users\xxx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Arduino.h:253:18: error: expected unqualified-id before '(' token

 #define max(a,b) ((a)>(b)?(a):(b))

                  ^

exit status 1
Error compiling for board Adafruit HUZZAH ESP8266.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Hieromon, 2018-06-25:
I have found a similar problem on https://github.com/esp8266/Arduino/issues/2549 and https://github.com/platformio/platformio-core/issues/646.
The macros of max and min in Arduino.h are incompatible with std::vector.
AutoConnect uses std::vector. On the other hand, arduino.h of esp8266 core 2.3.0 avoids the definition of min and max macros for std::vector usage.

Arduino/cores/esp8266/Arduino.h#L250
> #ifndef _GLIBCXX_VECTOR
> // arduino is not compatible with std::vector
> #define min(a,b) ((a)<(b)?(a):(b))
> #define max(a,b) ((a)>(b)?(a):(b))
> #endif

In other words, in esp8266 arduino core 2.3.0, the max and the min macro will not be defined after ```#include <vector>```, and this problem will occur if the Arduino.h is included before including std::vector.

But the include sequence when compiling AutoConnect.cpp is as follows. (esp8266 arduion core 2.3.0)
AutoConnect.cpp#L10: ```#include "AutoConnect.h"``` ->
AutoConnect.h#L13: ```#include <vector>``` ->
xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/vector#L56: ```#define _GLIBCXX_VECTOR 1```
AutoConnect.h#L17: ```#include <ESP8266WiFi.h>``` ->
ESP8266WiFi.h#L39: ```#include "WiFiClient.h"``` ->
WiFiClient.h#L25: ```#include "Arduino.h"``` ->
Arduino.h#L250: ```#ifndef _GLIBCXX_VECTOR```

I have no idea why Arduino.h#L253 was evaluated by the compiler.
Also, it is not defined in the latest Arduino core.
https://github.com/esp8266/Arduino/blob/641c5cdc61e5967c682a24e2df1c3b1765d60dab/cores/esp8266/Arduino.h#L265-L271.

If your sketch wants to include Arduino.h before including AutoConnect.h (I do not know the reason), you can improve AutoConnect.h as follows.

AutoConnect.h#L13
``` arduino
#ifdef Arduino_h
// arduino is not compatible with std::vector
#undef min
#undef max
#endif
#include <vector>
```
I prepared a branch for measures/define_max as above. You can try this branch. But there is a possibility that it does not work well.

Hieromon, 2018-10-19:
Since there was no response even after one month passed, closed this issue.
When some reports are posted, reopen it.

closed 2018-10-19 by Hieromon