Hieromon/AutoConnect #10: Crash when specify a custom config

schmurtzm, 2018-08-15:
Hi,

I'm trying to change the name of the SSID and password for SoftAP but each time I try to specify a config I have a big instability on my esp.

Here my code, may be I've done something wrong.... But each time I activate the line "Portal.config("Test", "testtest"); " the ESP connect to my AP and crash every 5 seconds. with this message : 

> HTTPUpdateServer ready! Open http://test.local/update in your browser
> WiFi connected:MyAP
> IP:192.168.1.223
> 
>  ets Jan  8 2013,rst cause:4, boot mode:(3,7)
> 
> wdt reset
> load 0x4010f000, len 1384, room 16 
> tail 8
> chksum 0x2d
> csum 0x2d
> v614f7c32
> ~ld

If you have a complete example with a custom SoftAP config I'm interested ! :) 

---------------------------------
`#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <AutoConnect.h>
#include <ESP8266WebServer.h>
#include <PubSubClient.h>
#include <ESP8266mDNS.h>
#include <ESP8266HTTPUpdateServer.h>
#include <Adafruit_NeoPixel.h>
#include <ArduinoJson.h>
#include <EasyNeoPixels.h>

ESP8266WebServer Server;
AutoConnect      Portal(Server);
AutoConnectConfig  Config("TestTest", "testtest");
//AutoConnectConfig  Config;

WiFiClient   wifiClient;
PubSubClient mqttClient(wifiClient);
ESP8266HTTPUpdateServer httpUpdater;

void setup() {

  delay(1000);
  Serial.begin(115200);
  Serial.println();

  Server.on("/", rootPage);

  // update on web page
  httpUpdater.setup(&Server);
  Serial.printf("HTTPUpdateServer ready! Open http://pluie.local/update in your browser\n");
  

// Portal.config("Test", "testtest");   // ->> iF I activate this the ESP crash 
// Portal.config(Config);

  Serial.print("WiFi ");
  if (Portal.begin()) {
    Serial.println("connected:" + WiFi.SSID());
    Serial.println("IP:" + WiFi.localIP().toString());
  } else {
    Serial.println("connection failed:" + String(WiFi.status()));
    while (1) {
      delay(100);
      yield();
    }
  }

  //  We add a DNS name Test.local
  if (MDNS.begin("Test")) {
    MDNS.addService("http", "tcp", 80);
  }

}

void loop() {
  Portal.handleClient();
}

Hieromon, 2018-08-21:
If you enable the debug option, you may get useful information. Debugging options are on both ESP8266 Core and AutoConnect.

- Enable the debug option for the ESP8266 core on Arduino IDE, see attached screen shot
[image]
- Enable the debug option for the AutoConnect library, see the below
https://hieromon.github.io/AutoConnect/advancedusage/index.html#debug-print

Hieromon, 2018-10-19:
Since there was no response even after one month passed, closed this issue as enhanced.
When a debug log is posted, reopen it.

closed 2018-10-19 by Hieromon