Hieromon/AutoConnect #12: How to use Autoconnect with with ESP8266 which need to have Webserver port 502 open for Modbus TCP Master

JoeyUson, 2018-09-23:
I'm new in ESP8266 and manage to work the Modbus TCP serve and connect to wifi using the stardard connection. Now I would like to use AutoConnect but wondering how to run the webserver serving port 502 which is the standard port for this communication protocol.

Hieromon, 2018-09-24:
If you want just change the HTTP port of ESP8266WebServer, there are two ways in the current release.

- Change forcibly by defining the macro of HTTP port number  
- Prepare an ESP8266WebServer instance that owns the specified port

But, if you want to run a captive portal on port 502 with HTTP, it depends on the specification of the captive portal's detection client. For example, iOS on the iPhone cannot do it.   
Is there no problem even if HTTP and Modbus packets are exchanged on the same port anyway? I think that the Arduino implementation of Modbus's function will conflict with ESP8266WebServer for HTTP, but is my recognition wrong?

JoeyUson, 2018-10-07:
I did the second one and prepare not to disturb the original setting for port 80 which is allocated for internet explorer. Thanks a lot it is working having a webserver with port 502 in ESP8266

JoeyUson, 2018-10-07:
issue close

closed 2018-10-07 by JoeyUson