Hieromon/AutoConnect #13: Can anyone help me make this code compatible with AutoConnect?

marrobHD, 2018-09-30:
Thats my Code:

`/*
  _______                 _       _   _   _           _       _       _______                 _                       _
  |__   __|               | |     | \ | | (_)         | |     | |     |__   __|               | |                     | |
    | |      ___    ___  | |__   |  \| |  _    __ _  | |__   | |_       | |      ___    ___  | |__    _ __     ___   | |   ___     __ _   _   _
    | |     / _ \  / __| | '_ \  | . ` | | |  / _` | | '_ \  | __|      | |     / _ \  / __| | '_ \  | '_ \   / _ \  | |  / _ \   / _` | | | | |
    | |    |  __/ | (__  | | | | | |\  | | | | (_| | | | | | | |_       | |    |  __/ | (__  | | | | | | | | | (_) | | | | (_) | | (_| | | |_| |
    |_|     \___|  \___| |_| |_| |_| \_| |_|  \__, | |_| |_|  \__|      |_|     \___|  \___| |_| |_| |_| |_|  \___/  |_|  \___/   \__, |  \__, |
                                               __/ |                                                                               __/ |   __/ |
                                              |___/                                                                               |___/   |___/

*/

#define FASTLED_INTERRUPT_RETRY_COUNT 0
#define FASTLED_ALLOW_INTERRUPTS 0

#include <ArduinoJson.h>
#include <ESP8266WiFi.h>
#include <PubSubClient.h>
#include "FastLED.h"
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

/************ WIFI and MQTT Information (CHANGE THESE FOR YOUR SETUP) ******************/
const char* ssid = "FRITZ!Box 6490 Cable"; //type your WIFI information inside the quotes
const char* password = "010166Hamburg";
const char* mqtt_server = "192.168.178.56";
const char* mqtt_username = "homeassistant-1";
const char* mqtt_password = "hass1";
const int mqtt_port = 1883;

/**************************** FOR OTA **************************************************/
#define SENSORNAME "porch" //change this to whatever you want to call your device
#define OTApassword "1234" //the password you will need to enter to upload remotely via the ArduinoIDE
int OTAport = 8266;

/************* MQTT TOPICS (change these topics as you wish)  **************************/
const char* light_state_topic = "technight/porch";
const char* light_set_topic = "technight/porch/set";

const char* on_cmd = "ON";
const char* off_cmd = "OFF";
const char* effect = "solid";
String effectString = "solid";
String oldeffectString = "solid";

/****************************************FOR JSON***************************************/
const int BUFFER_SIZE = JSON_OBJECT_SIZE(10);
#define MQTT_MAX_PACKET_SIZE 512

/*********************************** FastLED Defintions ********************************/
#define NUM_LEDS    186
#define DATA_PIN    5
//#define CLOCK_PIN 5
#define CHIPSET     WS2811
#define COLOR_ORDER BRG

byte realRed = 0;
byte realGreen = 0;
byte realBlue = 0;

byte red = 255;
byte green = 255;
byte blue = 255;
byte brightness = 255;

#define qsubd(x, b)  ((x>b)?b:0)                   // Digital unsigned subtraction macro. if result <0, then => 0. Otherwise, take on fixed value.
#define qsuba(x, b)  ((x>b)?x-b:0)                          // Analog Unsigned subtraction macro. if result <0, then => 0
[... 1600 lines cut ...]
      break;
 
    default:                                                          // Middle of the ripples.
      leds[(center + step + NUM_LEDS) % NUM_LEDS] += ColorFromPalette(currentPalette, colour, myfade / step * 2, currentBlending);   // Simple wrap from Marc Miller
      leds[(center - step + NUM_LEDS) % NUM_LEDS] += ColorFromPalette(currentPalette, colour, myfade / step * 2, currentBlending);
      step ++;                                                         // Next step.
      break;
  } // switch step
 
} // ripple()
void rainbow_march() {                                        // The fill_rainbow call doesn't support brightness levels
  thishue++;                                                  // Increment the starting hue.
  fill_rainbow(leds, NUM_LEDS, thishue, deltahue);            // Use FastLED's fill_rainbow routine.
} // rainbow_march()
void rainbow_beat() {
 
  uint8_t beatA = beatsin8(17, 0, 255);                        // Starting hue
  uint8_t beatB = beatsin8(13, 0, 255);
  fill_rainbow(leds, NUM_LEDS, (beatA + beatB) / 2, 8);        // Use FastLED's fill_rainbow routine.
 
} // rainbow_beat`

Thanks for help!

~Marlon

Hieromon, 2018-10-12:
I don't have time to look into your code in detail. Looking at the code overview, it seems that AutoConnect can be embedded in setup_wifi(). However, I did not investigate all of your code. A detailed explanation of how to embed AutoConnect is documented. Please refer to https://hieromon.github.io/AutoConnect

1. Insert ```#include <AutoConnect.h>``` to the bottom of include headers.
2. Define ```AutoConnect Portal;``` before ```void setup()``` function.
3. Comment out the below step.
```
L266: WiFi.mode(WIFI_STA);
L267: WiFi.begin(ssid, password);
```
4. Insert a new line as ```Portal.begin()``` at behind of the commented out line.
5. Insert ```Portal.handleClient();``` behind of ```showleds();``` in ```void loop()``` function.

Hieromon, 2018-11-12:
Since there was no response even after one month passed, closed this issue.
When some reports are posted, reopen it.

closed 2018-11-12 by Hieromon