Hieromon/AutoConnect #17: Custom parameters ?

hardlog, 2018-10-19:
Hello,

This is a great Tuorial and a great library. Thank you!
Is it possible to transmit "Custom Parameters" via the Autoconnect menu (/ _ac)? I would like to be able to enter the address and topic of the MQTT Broker in the Autoconnect menu to use it later in my Sketch. 

Thank you Flo

Hieromon, 2018-10-19:
There are many requests for the custom data entry function in part of AutoConnect 's menu transition like as your recommendation. It will be similar to the user own code as the exit routines or callback functions.
I have been thinking from time to time and have some ideas.

1. Add custom parameters to the Configure new AP screen
2. Add user's own screen created by PageBuilder library to menu transition

I have not decided which one to implement. Both have advantages and disadvantages.
Its enhancement will be the main title of the next update.

cruizg, 2018-10-24:
You stole the question, I was about to write the same, the idea is that with your library we can send in addition to the user and password, additional parameters and saved in the ESP8266 / ESP32 for example MQTT values or values sent from a mobile app that wants send additional to user and password parameters that register the device and then use it in the sckets.

Hieromon, 2018-10-25:
That's right. I have already started designing it. It is expected to be released within a few weeks.
That release can prepare the add-on page adding input fields or buttons optionally and invoke them from the AutoConnect menu. You will be able to retrieve the inputted values in add-on page from sketches and can respond by the customized page.
However, saving/loading of user parameters will be further scheduled.

ageurtse, 2018-12-16:
Need this to for my project.

Naboo2604, 2018-12-25:
Hi, thank you for this very cool library. i build IoT weather sensors for family and friends and use your lib for self-configuration. it would be great if you implement custom fields, so i can use it for location data and custom name in the autoconfig process. 
thanks again and merry christmas,
markus

ageurtse, 2018-12-25:
When looking at the mqtt examples in https://github.com/Hieromon/AutoConnect/tree/enhance/AutoConnectAux there are some custom fields.

Keep in mind that this branche is the development branche, so there could be some isues.

JoeyUson, 2019-03-31:
Any idea how to include in the parameters the Server IP, USER and PASSWORD for use later in connection to mySQL database resides in other server or Raspi

Hieromon, 2019-03-31:
What you want to do is similar to the example **mqttRSSI** in the library. However, the following two conditions will be added to it.

1. To available the MySQL settings page, ESP8266/ESP32 has already established a connection to AP with STA mode. In other words, the menu of MySQL parameters settings does not appear while processing in Captive portal.
2. The response of the MySQL parameters setting page will return the result of the connection to MySQL.

Is it right? If so, you can control menu display dynamically using the [**menu**](https://hieromon.github.io/AutoConnect/apiaux.html#menu) function and the sketch as like this:
```cpp
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <AutoConnect.h>
#include <MySQL_Connection.h>

const static char* mysqlSettingPage PROGMEM = R"(
{
  "title": "MySQL Setting",
  "uri": "/mysql_setting",
  "menu": false,
  "element": [
    {
      "name": "header",
      "type": "ACText",
      "value": "<h2>MySQL broker settings</h2>",
      "style": "text-align:center;color:#2f4f4f;padding:10px;"
    },
    {
      "name": "server",
      "type": "ACInput",
      "value": "",
      "label": "Server",
      "placeholder": "MySQL server IP address"
    },
    {
      "name": "user",
      "type": "ACInput",
      "label": "user"
    },
    {
      "name": "password",
      "type": "ACInput",
      "label": "password"
    },
    {
      "name": "connect",
      "type": "ACSubmit",
      "value": "Connect",
      "uri": "/mysql_conn"
    },
    {
      "name": "discard",
      "type": "ACSubmit",
      "value": "Discard",
      "uri": "/ac"
    }
  ]
}
)";

const static char* mysqlConnectPage PROGMEM = R"(
{
  "title": "MySQL Setting",
  "uri": "/mysql_conn",
  "menu": false,
  "element": [
    {
      "name": "caption",
      "type": "ACText",
      "value": "<h4>MySQL connection</h4>",
      "style": "text-align:center;color:#2f4f4f;padding:10px;"
    },
    {
      "name": "Result",
      "type": "ACText"
    },
    {
      "name": "OK",
      "type": "ACSubmit",
      "value": "OK",
      "uri": "/"
    }
  ]
}
)";

AutoConnect     portal;
AutoConnectAux  mysqlSetting;
AutoConnectAux  mysqlConnect;

WiFiClient client;
MySQL_Connection conn((Client *)&client);

void handleRoot() {
  String  content =
    "<html>"
    "<head>"
    "<meta name=\"viewport\" content=\"width=device-width, initial-scale=1\">"
    "</head>"
    "<body>"
    "Your sketch application there."
    "<p style=\"padding-top:5px;text-align:center\">" AUTOCONNECT_LINK(COG_24) "</p>"
    "</body>"
    "</html>";
  ESP8266WebServer&  webServer = portal.host();
  webServer.send(200, "text/html", content);
}

String onConnect(AutoConnectAux& aux, PageArgument& args) {
  // here, obtain parameters to connect MySQL
  IPAddress server;
  server.fromString(mysqlSetting.getElement<AutoConnectText>("server").value);
  String user = mysqlSetting.getElement<AutoConnectText>("user").value;
  String password = mysqlSetting.getElement<AutoConnectText>("password").value;

  unsigned long tm = millis();
  while (conn.connect(server, 3306, user.c_str(), password.c_str()) != true) {
    if (millis() - tm > 10000)
      break;
    delay(200);
    Serial.print('.');
  }
  AutoConnectText& result = aux.getElement<AutoConnectText>("Result");
  if (conn.connected())
    result.value = "Connected";
  else
    result.value = "Timeout";
  return String();
}

void setup() {
  delay(1000);
  Serial.begin(115200);
  Serial.println();
  mysqlSetting.load(mysqlSettingPage);
  mysqlConnect.load(mysqlConnectPage);
  mysqlConnect.on(onConnect);
  if (portal.begin()) {
    // Activate the settings page only
    mysqlSetting.menu(true);
  }
}

void loop() {
  if (conn.connected()) {
    // here, your actual sketch with using MySQL
  }
  portal.handleClient();
}
```
The above code has not been tested.

JoeyUson, 2019-04-02:
> @JoeyUson What you want to do is similar to the example **mqttRSSI** in the library. However, the following two conditions will be added to it.
> 
> 1. To available the MySQL settings page, ESP8266/ESP32 has already established a connection to AP with STA mode. In other words, the menu of MySQL parameters settings does not appear while processing in Captive portal.
> 2. The response of the MySQL parameters setting page will return the result of the connection to MySQL.
> 
> Is it right? If so, you can control menu display dynamically using the [**menu**](https://hieromon.github.io/AutoConnect/apiaux.html#menu) function and the sketch as like this:
> 
> ```c++
> #include <ESP8266WiFi.h>
> #include <ESP8266WebServer.h>
> #include <AutoConnect.h>
> #include <MySQL_Connection.h>
> 
> const static char* mysqlSettingPage PROGMEM = R"(
> {
>   "title": "MySQL Setting",
>   "uri": "/mysql_setting",
>   "menu": false,
>   "element": [
>     {
>       "name": "header",
>       "type": "ACText",
>       "value": "<h2>MySQL broker settings</h2>",
>       "style": "text-align:center;color:#2f4f4f;padding:10px;"
>     },
>     {
>       "name": "server",
>       "type": "ACInput",
>       "value": "",
>       "label": "Server",
>       "placeholder": "MySQL server IP address"
>     },
>     {
>       "name": "user",
>       "type": "ACInput",
>       "label": "user"
>     },
>     {
>       "name": "password",
>       "type": "ACInput",
>       "label": "password"
>     },
>     {
>       "name": "connect",
>       "type": "ACSubmit",
>       "value": "Connect",
>       "uri": "/mysql_conn"
>     },
>     {
>       "name": "discard",
>       "type": "ACSubmit",
>       "value": "Discard",
>       "uri": "/ac"
>     }
>   ]
> }
> )";
> 
> const static char* mysqlConnectPage PROGMEM = R"(
> {
>   "title": "MySQL Setting",
>   "uri": "/mysql_conn",
>   "menu": false,
>   "element": [
>     {
>       "name": "caption",
>       "type": "ACText",
>       "value": "<h4>MySQL connection</h4>",
>       "style": "text-align:center;color:#2f4f4f;padding:10px;"
>     },
>     {
>       "name": "Result",
>       "type": "ACText"
>     },
>     {
>       "name": "OK",
>       "type": "ACSubmit",
>       "value": "OK",
>       "uri": "/"
>     }
>   ]
> }
> )";
> 
> ESP8266WebServer  webServer;
> AutoConnect     portal(webServer);
> AutoConnectAux  mysqlSetting;
> AutoConnectAux  mysqlConnect;
> 
> WiFiClient client;
> MySQL_Connection conn((Client *)&client);
> 
> void handleRoot() {
>   String  content =
>     "<html>"
>     "<head>"
>     "<meta name=\"viewport\" content=\"width=device-width, initial-scale=1\">"
>     "</head>"
>     "<body>"
>     "Your sketch application there."
>     "<p style=\"padding-top:5px;text-align:center\">" AUTOCONNECT_LINK(COG_24) "</p>"
>     "</body>"
>     "</html>";
>   webServer.send(200, "text/html", content);
> }
> 
> String onConnect(AutoConnectAux& aux, PageArgument& args) {
>   // here, obtain parameters to connect MySQL
>   IPAddress server;
>   server.fromString(mysqlSetting.getElement<AutoConnectText>("server").value);
>   String user = mysqlSetting.getElement<AutoConnectText>("user").value;
>   String password = mysqlSetting.getElement<AutoConnectText>("password").value;
> 
>   unsigned long tm = millis();
>   while (conn.connect(server, 3306, user.c_str(), password.c_str()) != true) {
>     if (millis() - tm > 10000)
>       break;
>     delay(200);
>     Serial.print('.');
>   }
>   AutoConnectText& result = aux.getElement<AutoConnectText>("Result");
>   if (conn.connected())
>     result.value = "Connected";
>   else
>     result.value = "Timeout";
>   return String();
> }
> 
> void setup() {
>   delay(1000);
>   Serial.begin(115200);
>   Serial.println();
>   mysqlSetting.load(mysqlSettingPage);
>   mysqlSetting.menu(false);
>   mysqlConnect.load(mysqlConnectPage);
>   mysqlConnect.menu(false);
>   mysqlConnect.on(onConnect);
>   webServer.on("/",handleRoot);
>   portal.begin();
> }
> 
> void loop() {
>   // MySQL setting menu control
>   bool mysqlMenu = WiFi.status() == WL_CONNECTED;
>   mysqlSetting.menu(mysqlMenu);
> 
>   if (conn.connected()) {
>     // here, your actual sketch with using MySQL
>   }
> 
>   portal.handleClient();
> }
> ```
> 
> The above code has not been tested.

Just new in using AutoConnect with MySQL, however I add  "portal.join({ mysqlSetting });" just after mysqlConnect.on(onConnect); and the setting menu appear in the menu selection but after filling the form it give 403 error code

Hieromon, 2019-04-03:
At first, can you determine the state of LAN transparency with a sniffer tool such as Wireshark?  403 response has nothing to do with AutoConnect.

Hieromon, 2019-04-04:
@JoeyUson I reviewed the code I presented. Your point is partially correct. The above code does not have a custom web page joined to the portal. The code to add is below.

```cpp
portal.join({ mysqlSetting, mysqlConnect });
```
However, the cause of the 403 response is unknown.

JoeyUson, 2019-04-04:
Thanks a lot for the reply. I checked the sketch for MySQLconnect and use
the supplied sketch (simplified) and I'm having problem with connecting to
database and so nothing related to AutoConnect.

Have a good day.

JoeyUson, 2019-04-04:
I already solve MySQL database and it is only the user grant access. Used MySQL sample database sketch and able to insert data at regular interval from ESP8266.

Now return back to our Autoconnect sketch above from
changing the   "(conn.connect(server, 3306, user.c_str(), password.c_str()) != true)" with "(conn.connect(server_addr,3306,"joey","secret") != true )" works but the former give 

**error shown bellow from serial** 
[AC] Element<server> type mismatch<3>
[AC] Element<user> type mismatch<3>
[AC] Element<password> type mismatch<3>

Hieromon, 2019-04-05:
@JoeyUson 
Sorry, I took a mistake. The correct code is:
```cpp
  server.fromString(mysqlSetting.getElement<AutoConnectInput>("server").value);
  String user = mysqlSetting.getElement<AutoConnectInput>("user").value;
  String password = mysqlSetting.getElement<AutoConnectInput>("password").value;
```

JoeyUson, 2019-04-05:
**//Here the complete code for the test I did for AutoConnect and MySQL connection after the changes 
// It is working now , Thanks a lot for the assistance**

// AutoConnect with MySQL database connection

#include <AutoConnect.h>
#include <MySQL_Connection.h>
#include <MySQL_Cursor.h>

const static char* mysqlSettingPage PROGMEM = R"(
{
  "title": "MySQL Setting",
  "uri": "/mysql_setting",
  "menu": false,
  "element": [
    {
      "name": "header",
      "type": "ACText",
      "value": "<h2>MySQL broker settings</h2>",
      "style": "text-align:center;color:#2f4f4f;padding:10px;"
    },
    {
      "name": "server",
      "type": "ACInput",
      "value": "",
      "label": "Server",
      "placeholder": "MySQL server IP address"
    },
    {
      "name": "user",
      "type": "ACInput",
      "label": "user"
    },
    {
      "name": "password",
      "type": "ACInput",
      "label": "password"
    },
    {
      "name": "connect",
      "type": "ACSubmit",
      "value": "Connect",
      "uri": "/mysql_conn"
    },
    {
      "name": "discard",
      "type": "ACSubmit",
      "value": "Discard",
      "uri": "/_ac"
    }
  ]
}
)";

const static char* mysqlConnectPage PROGMEM = R"(
{
  "title": "MySQL Setting",
  "uri": "/mysql_conn",
  "menu": false,
  "element": [
    {
      "name": "caption",
      "type": "ACText",
      "value": "<h4>MySQL connection</h4>",
      "style": "text-align:center;color:#2f4f4f;padding:10px;"
    },
    {
      "name": "Result",
      "type": "ACText"
    },
    {
      "name": "OK",
      "type": "ACSubmit",
      "value": "OK",
      "uri": "/"
    }
  ]
}
)";

ESP8266WebServer	webServer;
AutoConnect			portal(webServer);
AutoConnectAux		mysqlSetting;
AutoConnectAux		mysqlConnect;

WiFiClient client;

// database related
MySQL_Connection conn((Client *)&client);
MySQL_Cursor* cursor;
// Sample query
const char INSERT_SQL[] = "INSERT INTO test_arduino.hello_arduino (message) VALUES ('Hello, Arduino!')";

void handleRoot() {	// this is the content of the root page to be display as default
	String  content =
		"<html>"
		"<head>"
		"<meta name=\"viewport\" content=\"width=device-width, initial-scale=1\">"
		"</head>"
		"<body>"
		"Your sketch application there."
		"<p style=\"padding-top:5px;text-align:center\">" AUTOCONNECT_LINK(COG_24) "</p>"
		"</body>"
		"</html>";
	webServer.send(200, "text/html", content);
}

String onConnect(AutoConnectAux& aux, PageArgument& args) {
	// here, obtain parameters to connect MySQL
	IPAddress server;
	
	server.fromString(mysqlSetting.getElement<AutoConnectInput>("server").value);
	String user = mysqlSetting.getElement<AutoConnectInput>("user").value;
	String password = mysqlSetting.getElement<AutoConnectInput>("password").value;

	unsigned long tm = millis();

	while (conn.connect(server, 3306, user.c_str(), password.c_str()) != true) {
		if (millis() - tm > 500)
			break;
		delay(200);
		Serial.print('.');
	}

	AutoConnectText& result = aux.getElement<AutoConnectText>("Result");
	if (conn.connected()) {
		result.value = "Connected";
		// create MySQL cursor object if connected
		cursor = new MySQL_Cursor(&conn);
	}
	else
		result.value = "Timeout";
	return String();
}

void setup() {
	delay(1000);
	Serial.begin(115200);
	Serial.println();
	mysqlSetting.load(mysqlSettingPage);
	mysqlSetting.menu(false);			// hide the myssqlSetting from menu initially
	mysqlConnect.load(mysqlConnectPage);
	mysqlConnect.menu(false);			// should always be hidden and call by mysqlsetting only
	mysqlConnect.on(onConnect);			// to be executed for database connection

	portal.join({ mysqlSetting,mysqlConnect});	// added in the standard menu items
	webServer.on("/", handleRoot);
	portal.begin();
}

void loop() {
	// MySQL setting menu control
	bool mysqlMenu = WiFi.status() == WL_CONNECTED;
	mysqlSetting.menu(mysqlMenu);				// will display the mysqlSetting once connected to WiFi

	static unsigned long tm = millis();

	if (conn.connected()) 					// if still connected to database
		if (millis() - tm > 5000) {
			cursor->execute(INSERT_SQL);		// Insert Hello Message
			tm = millis();
		}

	portal.handleClient();
}

Hieromon, 2019-04-06:
You can save the inputted parameter as server IP, user, and password for MySQL connection, to SPIFFS on the ESP module. The [AutoConnectAux::saveElement](https://hieromon.github.io/AutoConnect/apiaux.html#saveelement) function outputs the specified elements to the stream along with the elements value.
For example, to save the server IP address, user, password after conn.connect is successful, you can use the following code:
```cpp
#include <FS.h>
#define PARAM_FILE      "/param.json"

// In onConnect function

  if (conn.connected()) {
    saveParams(PARAM_FILE);    // Insert this line
    result.value = "Connected";
    // create MySQL cursor object if connected
    cursor = new MySQL_Cursor(&conn);
  }

// Add new function to save parameter as the below:
void saveParams(const char* paramFile) {
  SPIFFS.begin();
  File param = SPIFFS.open(paramFile, "w");
  mysqlSetting.saveElement(param, { "server", "user", "password" });
  param.close();
  SPIFFS.end();
}
```
And, load this parameter file in advance after resetting.

```cpp
void setup() {
  delay(1000);
  Serial.begin(115200);
  Serial.println();
  mysqlSetting.load(mysqlSettingPage);

  loadParams(PARAM_FILE);       // Load the parameters in advance

  mysqlSetting.menu(false);	// hide the myssqlSetting from menu initially
  mysqlConnect.load(mysqlConnectPage);
...
}

// Add new function to load the stored parameter as the below:
void loadParams(const char* paramFile) {
  SPIFFS.begin();
  File param = SPIFFS.open(paramFile, "r");
  if (param) {
    bool rc = mysqlSetting.loadElement(param);    // Load the elements with parameters
    if (rc)
      Serial.println(String(paramFile) + " loaded");
    else
      Serial.println(String(paramFile) + " failed to load");
    param.close();
  }
  SPIFFS.end();
}
```
But this is still not enough. It's smart that the menu is available when you time out with parameters loaded after reset as you said. Therefore we can take the further measures.
1. Make server IP, user, password to global variables. It is accessible at any time.
2. The loaded parameters are stored in their global variables and retrieved as needed.
3. The MySQL connection is executed by another function. This function takes parameters from the above global variables.
4. In the loop function, we do the sketch application. Activate the menu if the MySQL connection lost.

Here, the final code looks like this
```cpp
#if defined(ARDUINO_ARCH_ESP8266)
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#elif defined(ARDUINO_ARCH_ESP32)
#include <WiFi.h>
#include <WebServer.h>
#include <SPIFFS.h>
[... 209 lines cut ...]
  portal.begin();
}

void loop() {
  // MySQL setting menu control
  bool mysqlMenu = WiFi.status() == WL_CONNECTED;
  // will display the mysqlSetting once connected to WiFi
  mysqlSetting.menu(mysqlMenu);

  static unsigned long tm = millis();
  if (conn.connected()) { 					// if still connected to database
  	if (millis() - tm > 5000) {
  		cursor->execute(INSERT_SQL);  // Insert Hello Message
  		tm = millis();
  	}
  }
  else if (mysqlMenu) {
    // WiFi connection has been established, but MySQL connection lost
    // Try to reconnect
    connectMySQL(3000);
  }

  portal.handleClient();
}
```

JoeyUson, 2019-04-06:
I remove the delay(300) inside the loop of conn.connect.. to reduce the delay in connecting the WiFi. It seems the mySQLconnect is racing with WiFI connection and now it takes longer time to connect the WiFi. It works and the save server,user and password are retain even restart of esp8266.

Hieromon, 2019-04-07:
@JoeyUson I can not evaluate it because your adjustment depends on the MySQL library but I think you understand AutoConnect's parameter loading and saving mechanism.
May I close this topic?

JoeyUson, 2019-04-07:
Yeh you can, its more in fine tuning..

closed 2019-04-07 by Hieromon