Hieromon/AutoConnect #19: Stackoverflow when accessing _ac page

MFreeze, 2018-11-05:
Hi I'm using your library on a LoLin D32 Pro (ESP32 WROVER).

The start of the Access Point is OK and I can get the root page on my smartphone. However as soon as I try to access (automatically or manually) the _ac page, I get the following error.

`[AC] 192.168.244.1/_ac
***ERROR*** A stack overflow in task loopTask has been detected.
abort() was called at PC 0x4008f554 on core 1
0x4008f554: vApplicationStackOverflowHook at /home/mfreeze/esp/esp-idf/components/esp32/panic.c:680

Backtrace: 0x4008f310:0x3ffbd710 0x4008f53d:0x3ffbd730 0x4008f554:0x3ffbd750 0x4008ca17:0x3ffbd770 0x4008e3b0:0x3ffbd790 0x4008e366:0x3ffbd918
0x4008f310: invoke_abort at /home/mfreeze/esp/esp-idf/components/esp32/panic.c:680

0x4008f53d: abort at /home/mfreeze/esp/esp-idf/components/esp32/panic.c:680

0x4008f554: vApplicationStackOverflowHook at /home/mfreeze/esp/esp-idf/components/esp32/panic.c:680

0x4008ca17: vTaskSwitchContext at /home/mfreeze/esp/esp-idf/components/freertos/tasks.c:3564

0x4008e3b0: _frxt_dispatch at /home/mfreeze/esp/esp-idf/components/freertos/portasm.S:406

0x4008e366: _frxt_int_exit at /home/mfreeze/esp/esp-idf/components/freertos/portasm.S:206

Rebooting...
`

I'm not using the Arduino IDE (I use the toolchain instead) and I don't have anything else running on the board.

I'm using the code provided in the Readme:

`
/* WeatherStation for ValueBugs Project
 *  Written to be run on ESP32 board
 *  Connect to WiFi
 *  Get measures from sensors : 2 BME260 sensor (temperature/humidity/pressure) + 1 DS18B20 (temperature) + 1 HL-69 (moisture)
 *  Write values on influxdb
 *  Print values on screen
 *  
 *  
 */

/*-----------------------------------------------------------------------------
 *  Libraries
 *-----------------------------------------------------------------------------*/
/* {{{ -------- Libraries -------- */
#include <WiFi.h>
#include <WebServer.h>
#include <AutoConnect.h>
/* }}} */

/*-----------------------------------------------------------------------------
 *  Constants
 *-----------------------------------------------------------------------------*/
/* {{{ -------- Constants -------- */
WebServer Server;
AutoConnect Portal(Server);
/* }}} */

/*-----------------------------------------------------------------------------
 *  Functions
 *-----------------------------------------------------------------------------*/
/* {{{ -------- Functions -------- */
void rootPage() {
    char content[] = "Hello, world";
    Server.send(200, "text/plain", content);
}
/* }}} */

void setup()
{
    /* Initialisation du port série */
    Serial.begin(115200);

    Server.on("/", rootPage);
    if (Portal.begin()) {
        Serial.println("WiFi connected: " + WiFi.localIP().toString());
    }
}

void loop()
{
    Portal.handleClient();

    //delay(5000); //Delay 5 sec.
}
`

Hieromon, 2018-11-05:
D32 Pro has not been tested in my environment, but I tested it by choosing LOLIN D32 Pro in the board manager using M5Stack core with PSRAM like D32 Pro. It worｋｓ.
I think that arduino-esp32 is not stable yet, it seems the same case as espressif/arduino-esp32#1025.
Is the SDK version of esp-idf in your environment the same as mine?
`Serial.println(String("SDK:") + String(ESP.getSdkVersion()));`
```
SDK:v3.2-dev-39-gaaf12390
```

MFreeze, 2018-11-05:
No :/

SDK:v3.2-dev-1429-g7efbe8178

Ok I try to see if I can get an older version.

MFreeze, 2018-11-05:
If you use the git repo, could you give me the hash of the commit corresponding to your SDK version? 

Thanks in advance

Hieromon, 2018-11-06:
https://github.com/espressif/arduino-esp32/commit/3222e64
It's just at LOLIN D32 Pro supports.

bfaliszek, 2018-12-14:
I'm not really sure if my error is related to this, but after uploading Simple.ino, immediately after trying to connect to the esp8266ap network, the following error appears and ESP8266 (NodeMCU) restarts.

```
Exception (3):
epc1=0x40214bc4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4024645a depc=0x00000000

ctx: cont 
sp: 3ffffa80 end: 3fffffd0 offset: 01a0

>>>stack>>>
3ffffc20:  40103269 40210d23 7fffffff 00000000  
3ffffc30:  00000022 00000008 3ffef8b8 40216030  
3ffffc40:  3ffffcf0 3ffffcc0 00000008 00000001  
3ffffc50:  00000010 3ffffcc0 3ffffcc0 40210d23  
3ffffc60:  4024645a 00000004 3ffffcc0 40210d6f  
3ffffc70:  00000000 3ffef8b8 3ffffcc0 40210da1  
3ffffc80:  00000000 3ffef8b8 3ffffcc0 40210dee  
3ffffc90:  4024645a 40209b57 3fff14ac 40210cfc  
3ffffca0:  00000020 3ffef8b8 3fff14ac 40209b92  
3ffffcb0:  00000000 00000000 00000000 40210d23  
3ffffcc0:  3fff14c4 0000000f 00000004 40210d6f  
3ffffcd0:  3ffef9e0 3ffffd50 3ffffd10 40210da1  
3ffffce0:  3ffef9e0 3ffffd10 3ffffd10 40210ea8  
3ffffcf0:  3ffe98b7 3ffffd50 3ffef8b8 3ffefc28  
3ffffd00:  3ffef9e0 3ffffd50 3ffef8b8 40207bc6  
3ffffd10:  3fff1494 0000000f 00000007 40210d23  
3ffffd20:  3fff119c 00000007 3ffffd7c 40210d6f  
3ffffd30:  00000014 3ffffe18 3ffffd7c 00000001  
3ffffd40:  3ffef8b8 00000001 40207b4c 4020749f  
3ffffd50:  3fff11b4 0000000f 00000007 40210ed0  
3ffffd60:  00000001 40207468 3fff06ec 4020f824  
3ffffd70:  00000000 00000000 00000000 00000000  
3ffffd80:  00000000 00000000 3ffefa0c 40210ed0  
3ffffd90:  3ffe98b7 3ffefa34 3ffefa0c 4020ee69  
3ffffda0:  3fff119c 0000000f 00000007 3ffefa0c  
3ffffdb0:  3ffefa34 00000001 3ffefc28 00000030  
3ffffdc0:  00000000 00000000 00000001 3ffe8540  
3ffffdd0:  3ffefaec 3ffef9b8 3ffefc28 00000030  
3ffffde0:  00000000 3ffefa50 3fff1184 0000000f  
3ffffdf0:  00000000 3fff0bac 0000000f 00000001  
3ffffe00:  3fff1a0c 0000000f 00000007 3fff1a4c  
3ffffe10:  0000000f 00000003 3fff0b84 0000001f  
3ffffe20:  00000014 4bc6a7f0 00028038 3ffefc28  
3ffffe30:  3ffefa50 00000000 3ffe8540 3ffefa34  
3ffffe40:  00000001 00000001 4020f7ec 3fff06ec  
3ffffe50:  00000000 3fff1d84 3ffefa0c 3ffefc28  
3ffffe60:  00000001 3ffefa34 3ffefa0c 4020de33  
3ffffe70:  3ffef9b8 3ffefaec 3fff0ffc 4020f639  
3ffffe80:  4023c6d7 3ffefc28 40211850 3fffefb0  
3ffffe90:  3ffef9b8 3ffefaec 3ffef8b8 402080ac  
3ffffea0:  3ffef9b8 3ffefaec 3ffef8b8 4020848d  
3ffffeb0:  3fff0b04 00ffffff 01f4a8c0 3ffefd00  
3ffffec0:  00000000 00000000 00000000 3fff06b4  
3ffffed0:  0000000f 00000007 3fff0b4c 0000000f  
3ffffee0:  00000007 3fff0b34 0000000f 00000007  
3ffffef0:  3fff0b1c 0000000f 00000007 3fff1424  
3fffff00:  0000000f 00000007 40213190 01f4a8c0  
3fffff10:  40213190 00000000 40213190 00000000  
3fffff20:  40213190 00000000 00000000 40210d23  
3fffff30:  00000001 00000001 00000000 00000000  
3fffff40:  3fff1214 3fff122c 00000001 00000000  
3fffff50:  40213190 01f4a8c0 3ffef930 3ffefd00  
3fffff60:  3fffdad0 3ffef8b8 3ffefc28 402084ee  
3fffff70:  feefeffe 3ffef8b8 3ffefc28 4020a940  
3fffff80:  4020a810 feefeffe 402128b0 4021289c  
3fffff90:  00000000 00000000 00000000 feefeffe  
3fffffa0:  feefeffe feefeffe feefeffe feefeffe  
3fffffb0:  feefeffe 00000000 3ffefcf9 402118d4  
3fffffc0:  feefeffe feefeffe 3ffe8540 40100739  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
```

Hieromon, 2018-12-15:
You first go to the Exception Decoder, and I can not analyze the problem only with the information you provided. The nodemcu 1.0 is a major module and I also develop it naturally.
If it happens in both ESP32 and ESP8266, there will be your environmental factors. For example, in Arduino IDE 1.8.7, the behavior of LwIP v2 is sometimes wrong, and in ESP32 WiFiClient has a problem that the connection is disconnected.
You indicate at least where your crash point is in your environment, please.

Hieromon, 2018-12-29:
Since there was no response even after two weeks passed, closed this issue.
When environment information reports are posted, reopen it.

closed 2018-12-29 by Hieromon