Hieromon/AutoConnect #23: json response instead of html

m-menon, 2018-11-29:
Is there a possibility of sending json response (a REST API or something) instead of html? This would give developers different ways of integrating AutoConnect into their projects. Thanks

m-menon, 2018-12-07:
This new feature is cool. I'll perhaps be able to adapt the above for my use case. Thanks

Hieromon, 2018-12-19:
It's same as #17.

ageurtse, 2019-02-09:
Hieromon,

i can't find a good documentation on how to use the data type, yes there is an example. but not for a dropdownbox. and not for integers.

It would be nice if you release this version there is a help page with all the datatypes and how to use them.

ageurtse, 2019-02-09:
maybe it is also posible to give a position for the menu item, i don't like it at the bottom of the menu.
Home should be the last menu item. so if possible please change it.

Hieromon, 2019-02-09:
@ageurtse For the input box, it may need the ability to specify the format with custom fields. It is convenient to give the form as the regular expression and check the error with CSS.
Apparently, I need to consider that implementation and it will also need documentation on format specification. I am not good at English, but I will make a documentation effort so that users can understand it.

Hieromon, 2019-02-09:
@ageurtse I'm wondering about my decision for custom menu order. I'm feeling something wrong with the order too.

That’s weird, I think so too.:
```
Configure new AP
Open SSIDs
Disconnect
Reset...
HOME
[Custom menu item #1]
[Custom menu item #2]
```

Your suggestion:
```
Configure new AP
Open SSIDs
Disconnect
Reset...
[Custom menu item #1]
[Custom menu item #2]
HOME
```

Is the position of `Reset...` wrong?
Relocate the `Reset...` (`Reboot...`):
```
Reset...
Configure new AP
Open SSIDs
Disconnect
[Custom menu item #1]
[Custom menu item #2]
HOME
```
It's a strange, let change according to your advice

ageurtse, 2019-02-09:
why not give the menu items an order number 

like:

```
100 Configure new AP
110 [Custom menu item #1]
200 Open SSIDs
300 Disconnect
400 Reset...
420 [Custom menu item #2]
500 HOME
```

ageurtse, 2019-02-09:
> @ageurtse For the input box, it may need the ability to specify the format with custom fields. It is convenient to give the form as the regular expression and check the error with CSS.
> Apparently, I [...]

This i don't understand what you mean, i think it will end up great :)
If i need to read a pre made manual, i'm glad to do so. just let me know.

Hieromon, 2019-02-10:
@ageurtse 
> why not give the menu items an order number

Umm... I don't think about it that much. The AutoConnect HTML/CSS templates is large. Initially, I gave up on changing the order of menus with the dynamic generation of the HTML/CSS and wanted to reduce the consumption heap size of the library. Processes implemented for dynamic change of menu order will cause templates enlargement. So even now the order of the menu item is fixed.

Hieromon, 2019-02-10:
@ageurtse Also about custom fields data type, I thought that it would suffice to convert data type with a user sketch until pointed out from you and @pequalsvtimesi at #37. For example:

With the embedded AutoConnectElement
```cpp
ACInput inputAGE("", "AGE");
ACInput inputIP("", "IP Address");
...
int age = inputAGE.value.toInt();
IPAddress ip;
ip.fromString(inputIP.value);
```
With JSON
```json
{
...ommision
  "element": [
    {
      "name": "inputAGE",
      "type": "ACText",
      "label": "AGE"
    },
    {
      "name": "inputAGE",
      "type": "ACText",
      "label": "IP Address"
    }
  ]
...ommision
}
```
```cpp
...
portal.load(aux);
AutoConnectInput& inputAGE = aux.getElement<AutoConnectInput>("inputAGE");
AutoConnectInput& inputIP = aux.getElement<AutoConnectInput>("inputIP");
...
int age = inputAge.value.toInt();
IPAddress ip;
ip.fromString(inputIPAddress.value);
```

However, thanks to being pointed out, I understood the need for an example like as above and will add some example sketches, an explanation for the typical data type conversion.
Another thing, it will need to check the data format of the input value with a sketch, so the sketches need its convenient utility. That's why it's a necessity for the input pattern specification with regular expressions and the invalid highlighting in CSS, I think.

Hieromon, 2019-02-10:
@ageurtse 
> If i need to read a pre made manual, i'm glad to do so. just let me know.

Thank you so much! I deployed the document I'm writing on the [development branch](https://github.com/Hieromon/AutoConnect/tree/enhance/AutoConnectAux). You can download and read at /docs.

ageurtse, 2019-02-12:
if i look at mkdocs is that the same as docs ?

Hieromon, 2019-02-13:
Sometimes not, I'm committing after completed part so that you can easily review my English draft. Currently, mkdocs not synchronize with docs.

Hieromon, 2019-02-14:
@ageurtse I staged the latest draft to the develop branch. It is not complete but currently /docs synchronize with /mkdocs.

ageurtse, 2019-02-14:
I hope i have tonight time to look into it.
Keep you informed.

closed 2019-03-01 by Hieromon