Hieromon/AutoConnect #24: Request: in AP mode, change SSID name

ageurtse, 2018-12-20:
Hello, 

I prever to have a uniq SSID when in softAP mode. 
So i alter in file AutoConnect.h at line 117 
    apid(String(AUTOCONNECT_APID),
into 
    apid(String(AUTOCONNECT_APID)+"_"+String(ESP.getChipId(),HEX)),

maybee this could be implemented in the next release. 
With this small modification, you dont get any trouble when there are more then one devices in softAP mode. 

next thing realy would like to see is the possibility to alter the device name, now when i look in my router it says ESP_[deviceID], i realy would have a own device name, that makes sense to me. 
maybe as a stored variable from within the setup page.

ageurtse, 2018-12-21:
Great to here about accepting the hostname config. 

but when trying out the code from the 1ste point is get  a watchdog reset. 
don't know what causes this problem. 

this is the error log:
``` 
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0) [AC] DHCP client(STARTED)
[AC] WiFi.begin()
[AC] Connecting.......[AC] established IP:192.168.1.207
[AC] http server started
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0) [AC] DHCP client(STARTED)
[AC] WiFi.begin()
[AC] Connecting.[AC] established IP:192.168.1.207
[AC] http server started
WiFi connected: 192.168.1.207

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0) [AC] DHCP client(STARTED)
[AC] WiFi.begin()
[AC] Connecting.[AC] established IP:192.168.1.207
[AC] http server started
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0) [AC] DHCP client(STARTED)
[AC] WiFi.begin()
[AC] Connecting.[AC] established IP:192.168.1.207
[AC] http server started
WiFi connected: 192.168.1.207
```

and this is my main file:
```/*
  Simple.ino, Example for the AutoConnect library.
  Copyright (c) 2018, Hieromon Ikasamo
  https://github.com/Hieromon/AutoConnect

  This software is released under the MIT License.
  https://opensource.org/licenses/MIT
*/

#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <AutoConnect.h>
#include <ArduinoOTA.h>
#include <ESP8266HTTPUpdateServer.h>
#include <ESP8266mDNS.h>

ESP8266WebServer Server;
AutoConnect      Portal(Server);
ESP8266HTTPUpdateServer UpdateServer;
AutoConnectConfig config;

void rootPage() {
  String  content =
    "<html>"
    "<head>"
    "<meta name=\"viewport\" content=\"width=device-width, initial-scale=1\">"
    "</head>"
    "<body>"
    "<h2 align=\"center\" style=\"color:blue;margin:20px;\">Hello, world</h2>"
    "<p style=\"padding-top:10px;text-align:center\">" AUTOCONNECT_LINK(COG_32) "</p>"
    "</body>"
    "</html>";
  Server.send(200, "text/html", content);
}

void StartOTAServer (){
	//Starting Update Server
	     ArduinoOTA.onStart([]() {
	       String type;
	       if (ArduinoOTA.getCommand() == U_FLASH) {
	         type = "sketch";
	       } else { // U_SPIFFS
	         type = "filesystem";
	       }

	       // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
	       Serial.println("Start updating " + type);
	     });
	     ArduinoOTA.onEnd([]() {
	       Serial.println("\nEnd");
	     });
	     ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
	       Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
	     });
	     ArduinoOTA.onError([](ota_error_t error) {
	       Serial.printf("Error[%u]: ", error);
	       if (error == OTA_AUTH_ERROR) {
	         Serial.println("Auth Failed");
	       } else if (error == OTA_BEGIN_ERROR) {
	         Serial.println("Begin Failed");
	       } else if (error == OTA_CONNECT_ERROR) {
	         Serial.println("Connect Failed");
	       } else if (error == OTA_RECEIVE_ERROR) {
	         Serial.println("Receive Failed");
	       } else if (error == OTA_END_ERROR) {
	         Serial.println("End Failed");
	       }
	     });
	     ArduinoOTA.begin();

	     Serial.println("OTA server started.");
}

void setup() {
  delay(1000);
  Serial.begin(115200);
  Serial.println();

 // config.apid = String(AUTOCONNECT_APID)+"______"+String(ESP.getChipId(),HEX);

  UpdateServer.setup(&Server);

  config.apid = String(AUTOCONNECT_APID)+"_"+String(ESP.getChipId(),HEX);
  Portal.config(config);
  Portal.begin();

  Server.on("/", rootPage);
  //Portal.config(config);
  if (Portal.begin()) {
    Serial.println("WiFi connected: " + WiFi.localIP().toString());
    StartOTAServer();

    //Added by AJG, the device will be acsabible from [name].local
  //  char __dataFileName[sizeof(Portal.AP_hostname)];
  //  Portal.AP_hostname.toCharArray(__dataFileName, sizeof(__dataFileName));
  //  MDNS.begin(__dataFileName );

  }
}

void loop() {
  Portal.handleClient();
	ArduinoOTA.handle();
}
```

ageurtse, 2018-12-21:
maybe as thirth point, you could add mdns to autoconnect, this way you could use the given device name to connect to it.

ageurtse, 2018-12-22:
a little update, when ussing AutoConnect-enhance-AutoConnectAux lib. the sketch is working fine.

ageurtse, 2018-12-22:
are there more things that i can try with this lib ?

Hieromon, 2018-12-22:
> Are there more new things in this lib that i can try.

It is the [development branch](https://github.com/Hieromon/AutoConnect/tree/enhance/AutoConnectAux), I updated the branch with a version you can evaluate the development situation. However, some rework is necessary and the documentation is not yet.
You can try the custom parameters handling by the [mqttRSSI example](https://github.com/Hieromon/AutoConnect/tree/enhance/AutoConnectAux/examples/mqttRSSI).
This version is ArduinoJson required. If you don't have it, please install the [ArduinoJson library v5](https://arduinojson.org/v5/doc/installation/).

closed 2019-03-01 by Hieromon