Hieromon/AutoConnect #27: change of labels

ageurtse, 2018-12-23:
Now you have 2 labels, Reset... and Disconnect.

one is for ereasing the device and one for rebooting. Why not name what they do.

Erasing device...
Rebooting device...

Hieromon, 2018-12-23:
Your "Erasing device ..." is about the behavior of the API, the SDK has such a way to disconnect from the AP, and the STA mode is maintained because AutoConnect disconnects from the current AP, it is "Disconnect".
Also, "Reset..." calls ESP.reset() is a hard reset and can leave some of the registers in the old state, ESP.restart() tells the SDK to reboot, so its a more clean reboot. Both behaviors are different. 
So, what's this issue? only label description?

ageurtse, 2018-12-23:
yes, it is more a label description, but that is my opnion :)

Hieromon, 2018-12-23:
Basically, I agree with your opinion. It is important for UI design. However, I would like to prioritize differences in behavior according to the specification of SDK for this label.

ageurtse, 2018-12-23:
maybe at a later state, you could make the label as a variabel so one could change like other config item's

for now, let's work on a stable new release 👍

ageurtse, 2019-03-05:
Maybew the labels could be placed in a separate file, so people can localize the labels

Hieromon, 2019-03-06:
So, the label will be determined by the separate file like as define macros contained in the user sketch folder at build timing. And If the file does not exist, the label will assume the defaults. right?
I basically adopt your idea, but it may be difficult to implement due to Arduino IDE build limitations.
Well, I start the study. You can confirm the status with the [project]( https://github.com/Hieromon/AutoConnect/projects).

Hieromon, 2019-03-06:
@ageurtse We will have no choice but to lower the goal standard, I think.
Initially, I considered overwriting the menu string by individually putting only the label definition file in each sketch folder and aimed to adopt the AutoConnect defaults if there is no separate label definition file.
It is rationally to change the label string at build time because scenes that necessity rewriting labels during sketch executing are very rare.
However, this strategy is unlikely to be realized. I reviewed the c++ grammar to avoid conflicts between a separate file and AutoConnect initial values, but I can not find a way to work. It is possible if I intervene in Arduino's build mechanism, but that would be impossible. (Of course, lack of my knowledge might be  factors)
So I would like to propose to you the followings:

1. The label definition file is only one in AutoConnect library
It is the simplest solution. The definition file is named `AutoConnectLabels.h` and will placed in the `Arduio/library/AutoConnect/src` folder. You can change the label string with the AutoConnectLabels.h but it will influence all sketches. If you want to change the label for each sketch, you need to change the distribution file of the AutoConnect library every time you build.

2. Enhance AutoConnectConfig to replace the labels
Make the labels specified by AutoConnectConfig. The labels specified with AutoConnectConfig overrides the default at sketch execution time.
I don't like this solution so much.

How about your opinion?

ageurtse, 2019-03-06:
didn't know i could change it in that file. 
I didn't look in there, that is for me the right location. 
Thanks, for me it could be closed.

Hieromon, 2019-03-07:
AutoConnectLabels.h not separated yet. It will be for v0.9.8.
If you agree with the solution by AutoConnectLabels.h, it will be the implementation with v0.9.8. OK?

Hieromon, 2019-03-11:
Hello @ageurtse, I staged up the supports AutoConnectLabels.h to the [development branch](https://github.com/Hieromon/AutoConnect/tree/enhance/v098) for v 0.9.8.
You can try now. If some problem you find, please let me know.

Hieromon, 2019-04-14:
Merged #57

closed 2019-04-14 by Hieromon