Hieromon/AutoConnect #32: Feature Request / Example Addition: Captive Portal, AutoRefres and SPIFFS

SandeepanSengupta, 2019-02-01:
Is it possible to

- Launch a **Captive Portal**
- **AutoRefresh** date/time in [this](https://github.com/Hieromon/AutoConnect/blob/master/examples/Simple/Simple.ino) example
- Implement **Wear Leveling** on SPI Flash

Hieromon, 2019-02-03:
How does your request contribute to improving the AutoConnect library? Or is it that you want me to make an application sketch of NTP based real time clock?

SandeepanSengupta, 2019-02-04:
Greetings @Hieromon,

Sorry for delayed response.

First of all, I find your project brilliant and well documented and sorry that I may had skipped a few lines while reading the documentation. Please clarify if

- AutoConnect library already have *Wear Leveling* or equivalent feature

> How does your request contribute to improving the AutoConnect library?

A **_WiFi Manager_** like approach like [this](https://github.com/zhouhan0126/WIFIMANAGER-ESP32) or [this](https://github.com/tonyp7/esp32-wifi-manager) of launching a *Captive Portal* will make it easier to use.

> Or is it that you want me to make an application sketch of NTP based real time clock?
I noticed that

1. The date/time **DO NOT** auto-refresh in [**this**](https://github.com/Hieromon/AutoConnect/blob/master/examples/Simple/Simple.ino) example titled _**'simple'**_ (_**unless** the page is refreshed_)
Is there any way to do so (like by using **AJAX** or anything similar) ? If so, please make an application sketch.
2. **Feature Request**: There is **no function** mentioned in the sketch/library to easily change **TIMEZONE** or **Date/Time Format** (as in YYYY/MM/DD to DD/MM/YYYY), is it possible to add so ?

**P.S.**: I am really sorry for asking too many things at a time but I just started with ESP32 and really need the community help.

Regards,
@SandeepanSengupta
India

Hieromon, 2019-02-05:
> A **_WiFi Manager_** like approach like...

AutoConnect uses the Captive portal already. I cannot understand your point but if you say about the custom parameters, you can try the next version at the [branch](https://github.com/Hieromon/AutoConnect/tree/enhance/AutoConnectAux).

> 1. Is there any way to do so (like by using **AJAX** or anything similar) ? If so, please make an application sketch.

You can use NTP and JavaScript. But there is no plan to make it at the moment for an example sketch for AutoConnect. It does not relate the feature for AutoConnect library. Because such a real-time clock should be a module condensed into one using ESP8266 as a kernel, and it makes no sense to make it function on a browser basis. ESP8266 is a module for the embedding. If it is a module to query from the browser via HTTP,  it is enough to reload the page without requiring a real-time display.

> 2. **Feature Request**: There is **no function** mentioned in the sketch/library to easily change **TIMEZONE**

You can try [v0.9.7](https://github.com/Hieromon/AutoConnect/tree/enhance/AutoConnectAux) branch, it contains the Simple.ino improved for TIMEZONE.

closed 2019-02-18 by SandeepanSengupta