Hieromon/AutoConnect #33: reboot after connect

Globalskipper, 2019-02-02:
I'm testing AutoConnect with the SimpleExample and always have the problem that immediately after the connection to pc, iPad or mobile is established the ESP2866 is rebooting:

[AC] WiFi.config(IP=(IP unset), Gateway=(IP unset), Subnetmask=(IP unset), DNS1=(IP unset), DNS2=(IP unset)) [AC] DHCP client(STARTED)
[AC] WiFi.begin()
[AC] Connecting....................................................................................................[AC] time out IP:(IP unset)
[AC] http server started
[AC] SoftAP esp8266ap/12345678 CH(1) H(0) IP:192.168.244.1
[AC] DNS server started
[AC] /connecttest.txt
[AC] Page[] allocated
[AC] www.msftconnecttest.com/bag
[AC] Page[] allocated
[AC] init-p01st.push.apple.com/connecttest.txt
[AC] Page[] allocated
[AC] www.msftconnecttest.com/redirect
[AC] Page[] allocated
[AC] www.msftconnecttest.com/_ac
[AC] Page[/_ac] allocated
[AC] 192.168.244.1/_ac

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

I tested it with NodeMcu v1 and Lolin V3. Same behaviour.
I have two sensors and an oled connected. Which is running fine.
Even if I disconnect everything everything it's not working. 

Thanks for your help.

Hieromon, 2019-02-03:
Which version of arduino core-esp8266 are you using?

Globalskipper, 2019-02-03:
I was using 2.5.0-beta2
Because of our question I went back to 2.4.2 with complete new installation of Arduino IDE (Windows 10 App). Only installed libs are AutoConect (0.9.5) and PageBuilder (1.3.0).
Same behavior.

Hieromon, 2019-02-05:
I merged [PageBuilder 1.3.1](https://github.com/Hieromon/PageBuilder/tree/master), please you try AutoConnect 0.9.6 with the PageBuilder 1.3.1 again.

Globalskipper, 2019-02-05:
I tried Page BuilderB 1.3.1 with AutoConnect 0.9.6 (only 0.6.5 can be installed via Arduino IDE). 
No rebooting any more. 
Only the 192.168.244.1/_ac page is not loaded automatically.
Some progress ++++

Hieromon, 2019-02-05:
> only 0.6.5 can be installed via Arduino IDE

I agree. v0.96 has been lost from Arduino Library manager indexer. I have released v0.9.6 again, and the library indexer will fetch in a few hours.

> the 192.168.244.1/_ac page is not loaded automatically

AutoConnect v0.9.6 combined PageBuilder v1.3.1 with esp8266 core 2.42 works on Windows 10 and Chrome browser. I confirmed it.
Can you access 192.168.244.1/_ac from the browser manually? (Which browser do you use?)
If you can access such the uri, please enable the PageBuilder log and esp8266 core debug, then build > rerun. Then post the output log.
- To enable the PageBuiler log, Uncomment this line in PageBuilder.h :
https://github.com/Hieromon/PageBuilder/blob/6a73c1bdba9fd59508f266a0a4c4127812a6a5e2/src/PageBuilder.h#L34
- To enable the esp8266 core debug log.
Arduino IDE > Tool > Debug Port > Serial & Debug Level > CORE, WIFI, HTTP_SERVER

If cannot access 192.168.244.1/_ac manually, your router did not assign a dynamic IP address with DHCP after connection established maybe. The cause will unknown at now, but we may find something in the esp 8266 core debug log. Please try it.

Hieromon, 2019-02-05:
Thank you for the three parts.
About PC, Perhaps the cause is incomplete sending chunks of `/_ac` context. it's fixed with v0.9.7 already.
What is strange is Android as Samsung. `/_ac` as pop-up content sends successfully, but the screen does not appear.
I don't have an android device to test unfortunately. Does Samsung device have a customization function for suppression of captive portal by security or detection mode of WiFi spot on setting screen?
It seems that automatic popup suppression on the device side. But if you can connect some WiFi hot-spot by captive portal (e.q. Login screen) pop-up automatically with the Samsong, the problem lies elsewhere. How about?

Hieromon, 2019-03-02:
Resolving the reboot problem.
Android's WiFi connection Apps does not auto pop up Problems dependent on device configuration.
So close.

closed 2019-03-02 by Hieromon