Hieromon/AutoConnect #36: Simple HTTP GET Request doesn't work

GeorgeFlorian, 2019-02-08:
Hardware:

Board: ESP32 DEVKIT1

Core Installation version: 1.0.1-git this one

IDE name: Arduino IDE

Flash Frequency: 80Mhz

PSRAM enabled: no

Upload Speed: 115200

Computer OS: Linux Mint 19.1 Mate

### Description:

I'm trying to integrate a simple GET Request inside a simple sketch containing AutoConnect and it doesn't work.

### Sketch: 
 
```
#include <WiFi.h>
#include <WebServer.h>
#include <AutoConnect.h>
#include <HTTPClient.h>

WebServer Server;
AutoConnect Portal(Server);

void rootPage() {
    char content[] = "Hello, world !";
    Server.send(200, "text/plain", content);
  }

void setup () {
    delay(1000);
    Serial.begin(115200);
    Serial.println();
  
    Server.on("/", rootPage);
    if (Portal.begin()) {
        Serial.println("WiFi Connected: " + WiFi.localIP().toString());
      }
  }

void loop() {
    
    HTTPClient http;
    
    http.begin("http://jsonplaceholder.typicode.com/comments?id=10"); //Specify the URL
    int httpCode = http.GET();
  
    if(httpCode > 0) {
        String payload = http.getString();
        Serial.println(httpCode);
        Serial.println(payload);
        
      } else {
          Serial.println("Error on HTTP request");
        }
     http.end();

     Portal.handleClient();     

     delay(10000);
  }
```

### Debug Messages:
```
14:00:02.188 -> ets Jun  8 2016 00:22:57
14:00:02.188 -> 
14:00:02.188 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
14:00:02.188 -> configsip: 0, SPIWP:0xee
14:00:02.188 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
14:00:02.188 -> mode:DIO, clock div:1
14:00:02.188 -> load:0x3fff0018,len:4
14:00:02.188 -> load:0x3fff001c,len:1100
14:00:02.188 -> load:0x40078000,len:10088
14:00:02.188 -> load:0x40080400,len:6380
14:00:02.188 -> entry 0x400806a4
14:00:03.505 -> 
14:00:03.505 -> [W][WiFiGeneric.cpp:570] setSleep(): STA has not been started
14:00:03.704 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
14:00:03.704 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
14:00:03.803 -> [AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0) [AC] WiFi.begin()
14:00:03.803 -> [AC] Connecting...[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 4 - STA_CONNECTED
14:00:04.699 -> .[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 7 - STA_GOT_IP
14:00:04.732 -> [D][WiFiGeneric.cpp:385] _eventCallback(): STA IP: 192.168.1.101, MASK: 255.255.255.0, GW: 192.168.1.1
14:00:04.997 -> [AC] established IP:192.168.1.101
14:00:04.997 -> [AC] http server started
14:00:05.031 -> WiFi Connected: 192.168.1.101
14:00:05.031 -> [D][HTTPClient.cpp:265] beginInternal(): host: jsonplaceholder.typicode.com port: 80 url: /comments?id=10
14:00:05.064 -> [D][HTTPClient.cpp:977] connect():  connected to jsonplaceholder.typicode.com:80
14:00:05.130 -> [D][HTTPClient.cpp:1102] handleHeaderResponse(): code: 200
14:00:05.130 -> [D][HTTPClient.cpp:1105] handleHeaderResponse(): size: 322
14:00:05.130 -> [D][WiFiClient.cpp:463] connected(): Disconnected: RES: 0, ERR: 128
14:00:05.163 -> [D][HTTPClient.cpp:1239] writeToStreamDataBlock(): connection closed or file end (written: 322).
14:00:05.163 -> [D][HTTPClient.cpp:369] disconnect(): tcp is closed
14:00:05.163 -> 
14:00:05.163 ->  [D][HTTPClient.cpp:369] disconnect(): tcp is closed
14:00:05.163 -> 
14:00:15.146 -> [D][HTTPClient.cpp:265] beginInternal(): host: jsonplaceholder.typicode.com port: 80 url: /comments?id=10
14:00:15.179 -> [D][HTTPClient.cpp:977] connect():  connected to jsonplaceholder.typicode.com:80
14:00:15.312 -> [D][HTTPClient.cpp:1102] handleHeaderResponse(): code: 200
14:00:15.312 -> [D][HTTPClient.cpp:1105] handleHeaderResponse(): size: 322
14:00:15.312 -> [D][WiFiClient.cpp:463] connected(): Disconnected: RES: 0, ERR: 128
14:00:15.312 -> [D][HTTPClient.cpp:1239] writeToStreamDataBlock(): connection closed or file end (written: 322).
14:00:15.345 -> [D][HTTPClient.cpp:369] disconnect(): tcp is closed
14:00:15.345 -> 
14:00:15.345 -> is nobis reprehenderit ipsum amet nulla\nquia quas dolores velit et non\naut quia necessitatibus\nnostrum quaerat nulla et accusamus nisi facilis"
14:00:15.345 ->   }
14:00:15.345 -> ]
14:00:15.345 -> [D][HTTPClient.cpp:369] disconnect(): tcp is closed
14:00:15.345 -> 

And so on. 
```

Hieromon, 2019-02-09:
The sketch is wrong. Handle handleClient outside the timer loop. That's the way to use the WebServer library.

closed 2019-02-09 by Hieromon

GeorgeFlorian, 2019-02-11:
> The sketch is wrong. Handle handleClient outside the timer loop. That's the way to use the WebServer library.

In all of your examples you put `handleClient` or `handleRequest` inside `void loop() {}` .

If you know how to fix this sketch please help me. I wouldn't have bothered you, but I couldn't find any tutorial or guide.

Hieromon, 2019-02-11:
ESP32 is multicore, but AutoConnect hosts the WebServer instances in the same core, not multitasking. AutoConnect::handleClient internally calls WebServer::handleClient, but it resides in the same thread. Therefore it is the same as ESP8266WebServer::handleClient even if running on ESP32. See the [documentation](https://hieromon.github.io/AutoConnect/basicusage/index.html).

> In all of your examples you put `handleClient` or `handleRequest` inside `void loop() {}` .

If your sketch intends to respond to another HTTP request as a web server at the same time issuing a GET request every 10 seconds, `delay(10000);` stalls WebServer process because it runs in the same thread. So WebServer cannot respond to new HTTP requests. It will cause a timeout error on the client (usually a web browser). The outside of the timer loop means that handleClient must be performed under the `loop()` immediately avoid `delay(10000);`. Because it is not multitasking. I encourage you to first read all the necessary documentation for ESP32, ESP8266. AutoConnect assumes an understanding of the basic usage of the ESP8266WebServer library.

I don't know your sketch intention, but the sketch below will be helpful. It excludes AutoConnect from your sketch and made it just using the WebServer library. This is not work properly.
So, it is not a problem with AutoConnect. Make sure that you first the confirmation on the WebServer library only and post it.

```cpp
#include <WiFi.h>
#include <WebServer.h>
#include <HTTPClient.h>

WebServer Server;

void rootPage() {
  char content[] = "Hello, world !";
  Server.send(200, "text/plain", content);
}

void setup () {
  delay(1000);
  Serial.begin(115200);
  Serial.println();

  Server.on("/", rootPage);
  WiFi.mode(WIFI_STA);
  WiFi.begin("YOUR_SSID", "YOUR_PASSWORD");
  while (WiFi.status() != WL_CONNECTED) {
    delay(300);
  }
  Serial.println("WiFi Connected: " + WiFi.localIP().toString());
  Server.begin();
}

void loop() {
  HTTPClient http;

  http.begin("http://jsonplaceholder.typicode.com/comments?id=10"); //Specify the URL
  int httpCode = http.GET();
  
  if (httpCode > 0) {
    String payload = http.getString();
    Serial.println(httpCode);
    Serial.println(payload);
  } else {
    Serial.println("Error on HTTP request");
  }
  http.end();

  Server.handleClient();     

  delay(10000);
}
```