Hieromon/AutoConnect #37: Example for data types entry (integer, IP Address, string values) and retrieval

pequalsvtimesi, 2019-02-08:
Please consider crafting an example that includes a simple data entry page, possibly using pagebuilder.

Lacking practical experience with HTML, my attempts to extend the existing autoconnect, or pagebuilder examples to include a simple data entry page (i..e integers, ip address, string values) has been difficult for all the moving parts in learning autoconnect and pagebuilder, and HTML.

Hieromon, 2019-02-09:
You can try AutoConnectAux that is staged on the [development branch](https://github.com/Hieromon/AutoConnect/tree/enhance/AutoConnectAux).
Keep in mind that this branch under developing. #23

ageurtse, 2019-02-09:
Hieromon, 

i can't find a good documentation on how to use the data type, yes there is an example. but not for a dropdownbox. and not for integers. 

It would be nice if you release this version there is a help page with all the datatypes and how to use them.

closed 2019-03-01 by Hieromon