Hieromon/AutoConnect #39: Error Compiling

alexp1975, 2019-02-14:
Hi trying for days to compile the basic code on arduino for auto connect and keep getting the same error in the log. I've checked against the source code for AutoConnectPage.cpp and the code looks identical?, Im probably doing something wrong but for the life in me I can't figure it out. i'm trying to run on to a esp32 dev board with the following..

#include <WiFi.h>
#include <WebServer.h>
#include <AutoConnect.h>
WebServer Server;
AutoConnect      Portal(Server);
void rootPage() {
  char content[] = "Hello, world";
  Server.send(200, "text/plain", content);
}
void setup() {
  delay(1000);
  Serial.begin(115200);
  Serial.println();
  Server.on("/", rootPage);
  if (Portal.begin()) {
    Serial.println("HTTP server:" + WiFi.localIP().toString());
  }
}
void loop() {
  Portal.handleClient();
}

And keep getting the same on the console log.....

/Documents/Arduino/libraries/AutoConnect/src/AutoConnectPage.cpp: In member function 'String AutoConnect::_token_STATION_STATUS(PageArgument&)':
/Users/Documents/Arduino/libraries/AutoConnect/src/AutoConnectPage.cpp:892:57: error: 'wlStatusSymbol' may be used uninitialized in this function [-Werror=maybe-uninitialized]
   return "(" + String(st) + ") " + String(wlStatusSymbol);
                                                                                                                          ^

alexp1975, 2019-02-14:
Issue closed managed to fix the issue, no idea how but its now working correctly !!!

Hieromon, 2019-02-15:
Yes, it is due to a warning level of compilation option. It reports cause there is a logical path where wlStatusSymbol is referenced with uninitialized. But that state does not exist, and the actual problem will not occur.
However, it is not preferable. So I will merge the fix.

closed 2019-02-15 by Hieromon