Hieromon/AutoConnect #40: issue with sample ino : HandlePortalEx.ino

pfeno, 2019-02-24:
Hi,

just discover your lib and I try a sample to test the ac menu.
First, I thing a error appear in line 41
`/head`
`head` seem more appropriate.

After, could you explain me this line : 
`static const char PROGMEM mold_page[] = R"*lit(` and particulary `R"lit` expression ?

I compile the file and wen I try to connect to the ESP32 (core v1.0.1), I have the following trace :

[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0) 
[AC] WiFi.begin()
[AC] Connecting.....................................................................................................
[AC] time out IP:0.0.0.0
[AC] http server started
[AC] SoftAP esp32ap/12345678 CH(1) H(0) IP:192.168.11.1
Captive portal started, SoftAP IP:192.168.11.1
[AC] DNS server started
E (389186) event: mismatch or invalid event, id=63
E (389187) event: default event handler failed!
dhcps: send_offer>>udp_sendto result 0
[AC] /generate_204
[AC] Page[] allocated

but I have nothing on my smartphone web browser ? And I can't know what?
Could you help me ?
Best regards
Nomi

PS : I use the library v0.9.6 with PageBuilder v1.3.1 and ArduinoJSon v5.13.4

Hieromon, 2019-02-25:
> `head` seem more appropriate.

You are right, that's my mistake. Thank you for pointing out. I will fix with v0.9.7.

> but I have nothing on my smartphone web browser ? And I can't know what?

WiFi connection seems to be done correctly caused your Android smartphone is requesting generate_ 204. Perhaps the PageBuilder is not properly closing the TCP connection.
Can you try the least version of PageBuilder?
https://github.com/Hieromon/PageBuilder/tree/enhance/AutoConnectAux

pfeno, 2019-02-25:
Hi

thanks for your answer.
I test with the new PageBuilder and just below the trace from serial
monitor.
When I connect to IP  (192.168.11.1), I have an 404 error.
For access to web page, i need to add "_ac" to have the menu. Is it correct
?
Best regards

[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0,
DNS1=0.0.0.0, DNS2=0.0.0.0) [AC] WiFi.begin()
[AC]
Connecting.....................................................................................................[AC]
time out IP:0.0.0.0
[AC] http server started
[AC] SoftAP esp32ap/12345678 CH(1) H(0) IP:192.168.11.1
Captive portal started, SoftAP IP:192.168.11.1
[AC] DNS server started
E (45062) event: mismatch or invalid event, id=63
E (45063) event: default event handler failed!
dhcps: send_offer>>udp_sendto result 0
[AC] /generate_204
[AC] Page[] allocated
[AC] connectivitycheck.gstatic.com/generate_204
[AC] Page[] allocated
[AC] connectivitycheck.gstatic.com/
[AC] Page[] allocated
[AC] 192.168.11.1/generate_204
[AC] Page[] allocated
[AC] clients3.google.com/chat
[AC] Page[] allocated
[AC] c.whatsapp.net/gslb/
[AC] Page[] allocated
[AC] resolver.msg.global.xiaomi.net/generate_204
[AC] Page[] allocated
[AC] connect.rom.miui.com/generate_204
[AC] Page[] allocated
[AC] clients3.google.com/generate_204
[AC] Page[] allocated
[AC] connect.rom.miui.com/_ac
[AC] Page[/_ac] allocated
[AC] 192.168.11.1/_ac
[AC] 192.168.11.1/_ac
[AC] 192.168.11.1/_ac
[AC] 192.168.11.1/chat
[AC] Page[] allocated
[AC] c.whatsapp.net/favicon.ico
[AC] Page[] allocated
[AC] 192.168.11.1/generate_204
[AC] Page[] allocated
[AC] connect.rom.miui.com/generate_204
[AC] Page[] allocated
[AC] connectivitycheck.gstatic.com/connect
[AC] Page[] allocated
[AC] check.googlezip.net/connect
[AC] Page[] allocated
[AC] check.googlezip.net/
[AC] Page[] allocated
[AC] 192.168.11.1/favicon.ico
[AC] Page[] allocated
[AC] 192.168.11.1/e2e_probe
[AC] Page[] allocated
[AC] check.googlezip.net/_ac
[AC] Page[/_ac] allocated
[AC] 192.168.11.1/_ac
[AC] 192.168.11.1/e2e_probe
[AC] Page[] allocated
[AC] check.googlezip.net/_ac
[AC] Page[/_ac] allocated
[AC] 192.168.11.1/_ac
[AC] 192.168.11.1/
[AC] Page[] allocated
[AC] gicdqotgtse/
[AC] Page[] allocated
[AC] diaqdswjtxja/
[AC] Page[] allocated
[AC] popqiqgmnph/_ac
[AC] Page[/_ac] allocated
[AC] 192.168.11.1/_ac
[AC] 192.168.11.1/_ac
[AC] 192.168.11.1/_ac

[AC] 192.168.11.1/_ac
[AC] 192.168.11.1/_ac
[AC] 192.168.11.1/generate_204
[AC] Page[] allocated
[AC] connect.rom.miui.com/chat
[AC] Page[] allocated
[AC] c.whatsapp.net/chat
[AC] Page[] allocated
[AC] c.whatsapp.net/generate_204
[AC] Page[] allocated
[AC] connect.rom.miui.com/chat
[AC] Page[] allocated
[AC] c.whatsapp.net/chat
[AC] Page[] allocated
[AC] c.whatsapp.net/generate_204
[AC] Page[] allocated
[AC] connect.rom.miui.com/e2e_probe
[AC] Page[] allocated
[AC] check.googlezip.net/_ac
[AC] Page[/_ac] allocated
[AC] 192.168.11.1/_ac
[AC] 192.168.11.1http://check.googlezip.net/e2e_probe
[AC] Page[] allocated
[AC] check.googlezip.net/_ac
[AC] Page[/_ac] allocated
[AC] 192.168.11.1/_ac
[AC] 192.168.11.1http://check.googlezip.net/e2e_probe
[AC] Page[] allocated
[AC] check.googlezip.net/_ac
[AC] Page[/_ac] allocated
[AC] 192.168.11.1/_ac
[AC] 192.168.11.1/chat
[AC] Page[] allocated
[AC] c.whatsapp.net/chat
[AC] Page[] allocated
[AC] c.whatsapp.net/generate_204
[AC] Page[] allocated
[AC] clients3.google.com/generate_204
[AC] Page[] allocated
[AC] connect.rom.miui.comhttp://check.googlezip.net/e2e_probe
[AC] Page[] allocated
[AC] check.googlezip.net/_ac
[AC] Page[/_ac] allocated
[AC] 192.168.11.1/_ac
[AC] 192.168.11.1/chat
[AC] Page[] allocated
[AC] c.whatsapp.net/chat
[AC] Page[] allocated
[AC] c.whatsapp.net/chat
[AC] Page[] allocated
[AC] c.whatsapp.net/chat
[AC] Page[] allocated
[AC] c.whatsapp.net/gslb/
[AC] Page[] allocated
[AC] resolver.msg.global.xiaomi.net/_ac
[AC] Page[/_ac] allocated
[AC] 192.168.11.1/_ac
[AC] 192.168.11.1/generate_204
[AC] Page[] allocated

Hieromon, 2019-02-25:
`/_ac` is the default URI for AutoConnect. If your sketch does not have a `/` handler, that access will be 404.
If you want to respond to the AutoConnect menu by accessing the `/` URI, you need to redirect to `/_ac` with the `/` handler by `WebServer::on` function or specifies AUTOCONNECT_URI. Refer to the [documentation](https://hieromon.github.io/AutoConnect/advancedusage/index.html#relocate-the-autoconnect-home-path).

pfeno, 2019-02-25:
Well, I just test **your** sample sketch !
`HandleClient`
`HandlePortal`
`HandlePortalEx`
...

Hieromon, 2019-02-25:
> I just test **your** sample sketch !

So, `/chat` access will be a captive portal probe for android and it keeps requesting `/generate_204`. It seems that the captive portal detection of Android OS is continuing. 
`192.168.11.1` is AP of ESP32, not your accesspoint. Did you finish connecting to your access point?

Hieromon, 2019-03-02:
Closing due to lack of response.

closed 2019-03-02 by Hieromon