Hieromon/AutoConnect #46: send value to custom config page as initial values.

ageurtse, 2019-03-06:


ageurtse, 2019-03-06:
seems that i have found a way.
```
  AutoConnectInput& Dimvalue1 = aux.getElement<AutoConnectInput>("Dimvalue1");
  // Set initial value for the input box in a custom Web page.
  Dimvalue1.value = "25";
  // Nothing appendix for a generated HTML.
  return String();
```

now i which to know is there a one line solution to set the value ?

ageurtse, 2019-03-07:
by the way, when in github on the readme clicking on a link to the documentation sometimes i get a 404 error, it looks like there is somthing wrong with the links.

Hieromon, 2019-03-10:
@ageurtse 
> now i which to know is there a one line solution to set the value ?

Basically, your method is correct for setting the initial value.
Alternatively, you can use the [setElementValue](https://hieromon.github.io/AutoConnect/apiaux.html#setelementvalue) function.

```cpp
String settingsOn(AutoConnectAux& aux, PageArgument& args) {
  aux.setElementValue("Dimvalue1", "25");
  return String();
}
```

Hieromon, 2019-03-10:
> sometimes i get a 404 error, it looks like there is somthing wrong with the links.

Thank you!, I missed it and fixed it.

closed 2019-03-11 by ageurtse