Hieromon/AutoConnect #47: Doesn't work with latest version of arduinoJSON library!

nilava, 2019-03-07:
Please note that it doesn't work with the latest version of arduinojson library i.e. 6.9.1 ,mention that in the readme to not use the latest library !

ageurtse, 2019-03-07:
Nilava, 

This is mentioned in the installation manual https://hieromon.github.io/AutoConnect/index.html#installation.

@Hieromon, Maybe a good idea to upgrade the software to the latest JSON library.

Hieromon, 2019-03-08:
@ageurtse Yup. This thread tracks in the project as a v0.9.8.

Hieromon, 2019-03-08:
I set an initial goal for support ArduinoJson V6 as a below.

1. The user sketch can be executed with either ArduinoJson V5 or V6.
Depending on the version of ArduinoJson installed, the AutoConnect library builds correctly.

2. Supports the PSRAM for ESP32.
JsonBuffer can allocate to the PSRAM. Related the topics:
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/mem_alloc.html#external-spi-memory
https://github.com/bblanchon/ArduinoJson/issues/876

The goal may change with the progress of the task.

nilava, 2019-03-08:
@ageurtse Damn i missed that totally! Anyway works perfect now! @Hieromon will be waiting for new release! Its an awesome library! Thanks for making this!

cujomalainey, 2019-03-26:
It would be awesome if you would lock the arduinojson library to that version in library.json so people don't have to discover this the hard way when using platformio

Hieromon, 2019-04-05:
Same as Issue #52

Hieromon, 2019-04-05:
I staged ArudonoJson V6 support version as v0.9.8 to the[ development branch](https://github.com/Hieromon/AutoConnect/tree/enhance/v098). This version can evaluate the support status of ArduinoJson V6 and requires [PageBuilder of development branch](https://github.com/Hieromon/PageBuilder/tree/Enhance/AutoConnect_v098) to try.

**An early documentation for V0.9.8 related ArduinoJson V6 support**:

1. Supports ArduinoJson 6.9.1 or later.
2. Supports allocating JsonDocument buffer to PSRAM on ESP32 with ArduinoJson 6.10.0.
3. Supports **operator []** as a shortcut for **AutoConnectAux::getElement** function.
4. Supports **AutoConnectElement::as** function to easily coding for conversion from an AutoConnectElement to an actual type.
    It can be described as follows in combination with the operator [].

```cpp
AutoConnectAux& aux = *portal.aux("/custompage");  // Should be taken as a reference with operator*, not a pointer.
AutoConnectInput& input = aux["input_field"].as<AutoConnectInput>();
String inputValue = input.value;
```
Also, only the value member does not need conversion by AutoConnectElement::as.
```cpp
String inputValue = aux["input_field"].value;
```
However, the current development version does not yet implement direct value assignment. Currently, the following assignment to custom fields is not possible yet.
```cpp
AutoConnectAux& aux = *portal.aux("/custompage");
aux["input_field"].value = "initial_value";    // It's not possible yet. Not an error but the code is fragile.
AutoConnectText& caption = aux["caption"].as<AutoConnectText>();
caption.value = "Hello, world";    // It's possible.
caption.style = "color:pink";      // It's possible.
```
Now, open for the bug reports!

Hieromon, 2019-04-14:
Merged #57

closed 2019-04-14 by Hieromon