Hieromon/AutoConnect #48: How to get values stored previously in Custom Web Page w Json?

nilava, 2019-03-08:
Hey! I am trying to use your library to connect to blynk! I can't get to fetch the custom value after device restarts in the code. How to fetch it ?

ageurtse, 2019-03-08:
This is writen out in the manual, also look at the mqtt exames. 

Manual: https://hieromon.github.io/AutoConnect/achandling.html

Look at the paragraf “Where to pick up the values” there is it written out.

nilava, 2019-03-08:
Hey! I already looked into it, however the issue is that it would return a value only when the page is accessed, what i am trying to do is let the device turn on and connect automatically to the server with desired credentials without me having to navigate to that page and proceed, would that be possible?

ageurtse, 2019-03-08:
The first time you have to store the values to spiffs. 
otherwise you have to hard code the data, but that is not where the custom pages are for. 

Look at my github, where i'm in devolping of a Wifi Dimmer, in here i store the settings into a runtime created setup file in spiffs. 
https://github.com/ageurtse/Wifi-Dimmer/tree/master/dimmer_3.0

In the sub file filehandeling one can see how i generate the file and save settings. 
i hope the manual, the example files here and my personel project could help you.

nilava, 2019-03-09:
Thank you soo much! :smile: It helped me use it connect to blynk server automatically now, life is so much easy now !

nilava, 2019-03-09:
@Hieromon @ageurtse  i have made an example for using autoconnect with blynk server! You could include it in examples if you want and then i could share this library in the blynk community for more people to use it! Thanks! Love the library a lot! 
[src.zip](https://github.com/Hieromon/AutoConnect/files/2948330/src.zip)

closed 2019-03-09 by nilava