Hieromon/AutoConnect #51: Configure new AP blank at times

nilava, 2019-03-09:
Hey the page "Configure new AP" renders blank at times, after multiple refresh it loads, its usually happening when more than 7+ ssid are found. Is it a bug or is there any workaround for that or is it some issue with my code?

My Code - [Gdrive Link](https://drive.google.com/open?id=1Rv3uhxafs2aK7ndMPLRKxMsHG2qmedu2)

ageurtse, 2019-03-09:
Reset wifi credentials, can be done by reflashing and clearing all flash storage.

As far as i know there is no simple reset command to erease all data.

@hieromon, i need to load the wifi pag 2 times to see all networks. 
Thought it was an issue from my side.

nilava, 2019-03-10:
@ageurtse so i am not the only one with the issue it seems. Could you confirm with a simple sketch if the issue is still there? Obviously test when there's a lot of ssid!

nilava, 2019-03-10:
I did further digging and found out the issue is there only when we load custom web page otherwise its absolutely fine handling lots of SSID.

ageurtse, 2019-03-10:
I go try it tonight when back at home.
@nilava do you have some test files which i can use.
Vor non custom webpage i could use the basic example from the programm folder.
But what did you use for the custom webpage.
So if i have the same problem, hieromon could maybee reproduce the problem and solve it.

nilava, 2019-03-10:
@ageurtse Look at your code for Wifi-Dimmer, just remove [this](https://github.com/ageurtse/Wifi-Dimmer/blob/bd38f33fe0e31edefa83553e20d582e0733a3cbe/dimmer_3.0/dimmer_3.0.ino#L57) line and try, you will see it works perfectly! From my guess there is no memory left after loading so many config and then after returning a huge string to be served.

In the library you can see that with lots of SSID the string "ssidList" can become really large and thus might lead to blank render of the config page.
https://github.com/Hieromon/AutoConnect/blob/2fee92ca54c2a2bdd3280001b3f3b4fbaa0a2059/src/AutoConnectPage.cpp#L1106-L1126

@Hieromon i hope you look into this.

Hieromon, 2019-03-10:
@nilava I have not tested for reproducing with your uploaded code yet, but probably It will be insufficient memory as you guessed. The root cause is not the generation of the SSID list. It caused by the [PageBuilder](https://github.com/Hieromon/PageBuilder). PageBuilder concatenates the page contents by String. 

https://github.com/Hieromon/PageBuilder/blob/3a6c5b27998ff2ddf601e27d50aa8cc2029fe13d/src/PageBuilder.cpp#L204-L227

So, the String class of ESP8266 reserves the heap by the realloc on concatenation. If realloc fails, String will be lost. The function is **String::changeBuffer**. This function has been updated with the upstream of ESP8266 core. https://github.com/esp8266/Arduino/pull/5690
But, this PR does not seem to be included in stable release 2.5.0.

However, there is still the possibility that the heap will run out. AutoConnect generated HTML is long. If there are seven SSIDs, it will be near the 10K bytes. It means that a heap of about 20K bytes will be required for concatenation by PageBuilder. It is limit of an available heap that will remain after the ESP8266WebServer + AutoConnect + ArduinoJson resides excepts Blynk & ESP8266HTTPUpdateServer.
You can enable PageBuilder's debug option to actually check for out of memory. If the heap is run out, you will see the Serial message of the result of getFreeHeap.

PageBuilder.h L34
```cpp
#define PB_DEBUG
```
https://github.com/Hieromon/PageBuilder/blob/3a6c5b27998ff2ddf601e27d50aa8cc2029fe13d/src/PageBuilder.h#L34

And I prepared a workaround for PageBuilder. Its workaround is to reserve the string buffer by the length of the content in advance and will suppress fragmentation of the heap that occurs each time a string is concatenated. I have tested it over and over. However, since its effectiveness can not be expected so much, I have not implemented the AutoConnect side yet. 

https://github.com/Hieromon/PageBuilder/blob/3a6c5b27998ff2ddf601e27d50aa8cc2029fe13d/src/PageBuilder.cpp#L207-L214

Can you try with the debug option enabled on PageBuilder?
I will consider using AutoConnect's content string buffer reservation implementation with your code.

nilava, 2019-03-10:
@Hieromon you were right, anyway i am happy with refreshing multiple times for now until there's a proper workaround for that.

Serial Output: 
`[AC] 5 network(s) found
[PB] at leaving build: 18632 free
[PB] Content lost, length:9211
[PB] Free heap:20536
[PB] Res:200, Chunked:1
[PB] Free heap:29760, content len.:0`

r3na, 2019-03-25:
Hey @Hieromon I tried the 0.98 branch and the bug is still there (in my case, here it is showing 30~40 networks ). Does this fix is already there?

Hieromon, 2019-03-26:
@r3na 
> Does this fix is already committed there?

Sorry, not yet. You can try the buffer reservation option as an interim workaround. This option reserves a buffer for content building and will suppress fragmentation. 
https://github.com/Hieromon/AutoConnect/blob/04c236fb0a8ece12748ff7b5df91dac3db5d1d8f/src/AutoConnectDefs.h#L127
However, if there are as many as 30 access points found, improvement can not be expected. HTML content length of the SSID list is about 250 bytes by each. The content length for displaying 30 SSIDs will exceed 15K bytes in total length, and it needs within an unfragmented area.

> Would not be possible also to limit the number of access points so to avoid this problem? 

It is possible. It equips with the Next Page button, and I will implement experimentally on the v.098 branch to reduce the displayed SSID to 5 at a time.
Thank you for your advice.

Hieromon, 2019-03-26:
Hello @nilava, @r3na. I staged a trial version of paging the SSID list to the development branch. Please try this version in an environment where the esp8266 can find many SSIDs if you can.
The current configuration has 5 SSIDs for display per a page. You can change this with the **AutoConnectDefs.h** macro.
https://github.com/Hieromon/AutoConnect/blob/0fd3171fb5ba610c16980476b5376265857d6391/src/AutoConnectDefs.h#L132

nilava, 2019-03-26:
Negative, didn't solve the issue, went down as low as 1 SSID to be displayed per page, didn't work still.
Thanks for your effort though!

Hieromon, 2019-03-27:
@nilava, Thank you for testing. Hmmm, Yes. the operation is not stable. Sometimes appear, sometimes gone.
However, I will leave the option for paging display of the SSID list as a build option.
I think another way to resolve, sorry to keep you waiting.

r3na, 2019-03-28:
@Hieromon just saw now the last updates. Let me know if you need any test from my side

Hieromon, 2019-03-29:
Hello @nilava @r3na , I have changed the page generation and destruction scheme to avoid heap fragmentation and tried to chunk transfer some pages. The latest commit of the development branch absorbed these fixes.
Please try it from both the [PageBuilder ](https://github.com/Hieromon/PageBuilder/tree/Enhance/AutoConnect_v098) and [AutoConnect](https://github.com/Hieromon/AutoConnect/tree/enhance/v098) development branches.

nilava, 2019-03-29:
Amazing!!! :heart_eyes: Works Great! At times however the input form disappears, which is not a huge issue though, but as for blank page, it never comes anymore!

nilava, 2019-03-29:
[image] [image]

The minor bug i mentioned in the previous comment

Hieromon, 2019-03-30:
@nilava Thank you for testing and report. It seems that the changed scheme is still insufficient. I will consider more to stabilize the behavior of the Config New operation.

Hieromon, 2019-03-30:
@nilava I improved heap fragmentation in page building and staged the fix to the develop branch [enhance/v098](https://github.com/Hieromon/AutoConnect/tree/enhance/v098). However, I have not tested enough because I can not prepare an environment with 6 SSIDs on the same channel.
If the problem recurs, turn on debug logging with PageBuilder.h ```#define PB_DEBUG``` and check PageBuilder logs which say Failed building. If you find the failure log, please try increasing the following value.

AutoConnectDefs.h#L127

https://github.com/Hieromon/AutoConnect/blob/2c0207fb405ad1dd99708d2b4dde307d69ee693d/src/AutoConnectDefs.h#L127

This value will reserve the content buffer at the time of the chunked transfer. PageBuilder cannot reserve the buffer in the first place if it is too large, and if it is too small, PageBuilder cannot create content. Maybe correct size scope is from **12 * 1024** up to **15 * 1024**, I think.

nilava, 2019-03-31:
` #define AUTOCONNECT_CONTENTBUFFER_SIZE  (13 * 1024) `

This ensured maximum stability with as many as 12 SSID's! The issue is finally solved completely! Thanks! @Hieromon

Hieromon, 2019-03-31:
@nilava Thank you for testing. The default value is `15 *1024' based on your test results, which will be released in v098.

closed 2019-03-31 by Hieromon