Hieromon/AutoConnect #52: error compiling for board generic esp8266

jmartens2, 2019-03-10:
hi,
im new to your library and i tried uploading simple and hello world example and  neither would compile. ive been working with generic esp8266 for a bit and i believe i have the arduino 1.8.8 IDE setup correctly. is ther a change that i must make in the sketch or in src folders? i just downloaded the library today so im hoping its the latest issue. thanks for your hard work

Hieromon, 2019-03-11:
@jmartens2 Hello, I'm developing with Arduino1.8.8 under Windows 7 and Windows 10 environment, but I can not reproduce the compile error with Generic esp8266 module, also Arduino1.8.6.
Please provide me the following information. I'll diagnose the problem based on the information which will be provided by you.
- Your computer (host) OS and version.
- ESP8266 arduino core version which you are using. (If you used the upstream of the master branch, the commit id is needed.)
- Actual esp8266 board.
- Whole build options you specify with the Arduino-IDE tool menu when you choose Generic esp8266  module.
- Compiler message log.

jmartens2, 2019-03-11:
I’m sorry but I’m working thru that problem I’ve downgraded to arduino json 5. I’m making progress. I’m going back thru the instructions now to find more of my errors. Thank you for the greT response!!

jmartens2, 2019-03-11:
I must explain a little better. I must admit I did not expect such a quick reply to my question. It’s hard to find such great support! Please excuse my newness, I did not read the instructions well enough before I reached out. I must apologize. But thank you for your hard work and time.

Hieromon, 2019-03-12:
It's no problem @jmartens2. Thank you for having interest in AutoConnect.
Please close this topic if your problem is solved.

closed 2019-03-12 by jmartens2

Hieromon, 2019-04-05:
@jmartens2 I staged ArudonoJson V6 support version as v0.9.8 to the [development branch](https://github.com/Hieromon/AutoConnect/tree/enhance/v098). This version can evaluate the support status of ArduinoJson V6. You also need [PageBuilder 1.3.3](https://github.com/Hieromon/PageBuilder/tree/Enhance/AutoConnect_v098) branch to evaluate this version.