Hieromon/AutoConnect #53: Automatic open browser after connect to AP to load index.html saved on the SPIFFS

williamesp2015, 2019-03-30:
I really having difficulty to have a automatic loading index.html saved on the SPIFFS. I need to automatically open index.html without user require open browser and type the IP.

server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
Serial.println("/");
Serial.println("redirecting to /index.html");
request->redirect("/index.html");
});
#if ESP8266
if(!SPIFFS.begin()){
Serial.println("SPIFFS Mount Failed");
}
#endif
#if ESP32
if(!SPIFFS.begin(true)){
Serial.println("SPIFFS Mount Failed on ..");
}
#endif

listDir(SPIFFS, "/", 0);
server.serveStatic("/", SPIFFS, "/").setCacheControl("max-age=600").setDefaultFile("index.html");
server.begin();
Thank you

Hieromon, 2019-03-30:
I can not understand your issue. What's the problem and what you want to do?

williamesp2015, 2019-03-30:
Thanks @Hieromon .  Non of AutoConnect Examples is able to automatically load index.html once connected to SoftAP so I am looking for help and I can ignore AsyncWebserver  and use WebServer. I want simplify procedure for the user to just open SSID and automatically index.html pops like Airport webpage without typing the IP and load index.html file from SPIFFS.

Hieromon, 2019-03-30:
I still cannot understand your request well.
> I want simplify procedure for the user to just open SSID and automatically index.html pops
It is just using to the captive portal and You can mimic the DNS example of ESP8266.
Or do you want to pop up after ESP8266/ESP32 connecting to some AP as a station? Even so, you can still use the client device-side Capty Portal Detection, which has nothing to do with AutoConnect.

Hieromon, 2019-03-31:
There is no response and the topic has no relation with AutoConnect. Close.

closed 2019-03-31 by Hieromon