Hieromon/AutoConnect #54: How do you load parameters?

gcormier, 2019-03-31:
I am following the mqttRSSI example for custom parameters.

I have it working - I can set 2 values, and they are saved to flash.

How do I load them when the ESP32 is rebooted? Currently, while they are saved to flash, the values of my variables in my code are not set unless they are set via the web console.

I don't see where they are loaded in mqttRSSI on power-on.

Hieromon, 2019-04-01:
The mattRESS loads parameter only when accessing /mqtt_setting as a custom Web page. It is probably different from the needs of yours.
The easiest way for parameter loading with rebooting the ESP module is loading parameters immediately after loading the custom web page. It is made possible by calling loadParams at the beginning part of the setup(). Insert the following codes immediately after L357 of mqtt RSSI. This process allows you to access the parameter values even ​​after ESP32 boot.

https://github.com/Hieromon/AutoConnect/blob/342bdf3336db5d6d7875567e331bb3a34b7375c7/examples/mqttRSSI/mqttRSSI.ino#L357-L358

Insert below lines to between L357 and L358.
```cpp
    PageArgument dummyArg;  // Declare to align the interface.

    // Load parameter after boot
    loadParams(*mqtt_setting, dummyArg);

    // Retrieve each element
    AutoConnectInput& acMqttServer = mqtt_setting->getElement<AutoConnectInput>("mqttserver");
    AutoConnectInput& acChannelId = mqtt_setting->getElement<AutoConnectInput>("channelid");
    AutoConnectInput& acUserKey = mqtt_setting->getElement<AutoConnectInput>("userkey");
    AutoConnectInput& acApiKey = mqtt_setting->getElement<AutoConnectInput>("apikey");

    // Preset the parameters from reloaded elements
    serverName = acMqttServer.value;
    channelId = acChannelId.value;
    userKey = acUserKey.value;
    apiKey = acApiKey.value;

    // Refer preloaded parameters
    Serial.println("Parameter loaded");
    Serial.println("Server name:" + serverName);
    Serial.println("Channel ID:" + channelId);
    Serial.println("User key: " + userKey);
    Serial.println("API key: " + apiKey);
```

gcormier, 2019-04-04:
Thanks - just want to let you know I haven't had a chance to try this out but hopefully on the weekend and I will report back!

ageurtse, 2019-04-08:
No spamming,i’m loading settings from spiffs. The settings file is build at runtime.
Justlook at my git page.

Hieromon, 2019-05-29:
Since there was no response even after one month passed, closed this issue.
When some reports are posted, reopen it.

closed 2019-05-29 by Hieromon