Hieromon/AutoConnect #55: Sketch crashes

bobcroft, 2019-04-01:
I have Arduino 1.8.9, ESP8266 v2.5, ESP32 v1.1 and ArduinoJson v6.10 all loaded on a frsh Arduino install.  I get the following error message;
Error while detecting libraries included by C:\Arduino189\portable\sketchbook\libraries\AutoConnect\src\AutoConnectAux.cpp
and then I get a huge list of errors mentioning JSON.  The lis is so long I can't paste it here
Do you have any idea what is wrong? please.  I have tried loading the library from Library manager and manually from the zip download.

Hieromon, 2019-04-02:
Same as Issue #47, This is already mentioned in the installation documentation. AutoConnect requires ArduinoJson V5.
https://hieromon.github.io/AutoConnect/#installation
ArduinoJson V6 support will be released v0.9.8.

closed 2019-04-02 by Hieromon

bobcroft, 2019-04-02:
Hi thanks for the reply, when will version 0.9.8 be released please.

 

Thanks

 

Bob

 

https://hieromon.github.io/AutoConnect/#installation