Hieromon/AutoConnect #56: Very slow to populate the list of wifinetworks - possibly needs user feedback

geeksville, 2019-04-04:
If used in an area where many wifi networks are visible it may take a number of seconds (10? 20?) to populate the list of wifi access points.  It would be great if there was some sort of feedback given to the user that this is happening - to avoid the user thinking the device has hung.

If no one is working on this I'll try to send in a PR someday - but I wanted to record it.

Downstream issue here: https://github.com/geeksville/ezdevice-esp32/issues/1

Hieromon, 2019-04-05:
You can try the develop branch as v098. It supported page mode with the menu with Config new. If it's still slow, the essence of the problem will be WiFi.scanNetworks.
It is necessary to separate the causes.

Hieromon, 2019-04-07:
Similar to issue #51, Improved with the release of v0.9.8

closed 2019-04-07 by Hieromon

geeksville, 2019-04-09:
@Hieromon Thanks for the reply!  I'll pull in 0.9.8 when I return to the land of computers (currently hiking till May 9th).

Hieromon, 2019-04-10:
@geeksville Ok. After your evaluation, if you find any problem please post it. I will make efforts to solve as quickly as possible and open the issue again.