Hieromon/AutoConnect #58: Error Compiling AutoConnectUploadImpl

petreanv, 2019-04-15:
Hi trying for days to compile the basic code on arduino for auto connect and keep getting the same error in the log. I've checked against the source code for AutoConnectUploadImpl.h and the code looks identical?, Im probably doing something wrong but for the life in me I can't figure it out. i'm trying to run on to a esp8266 dev board with the following..

#include <AutoConnect.h>

#include <ESP8266WiFi.h>          // Replace with WiFi.h for ESP32
#include <ESP8266WebServer.h>     // Replace with WebServer.h for ESP32

ESP8266WebServer Server;          // Replace with WebServer for ESP32
AutoConnect      Portal(Server);

void rootPage() {
  char content[] = "Hello, world";
  Server.send(200, "text/plain", content);
}

void setup() {
  delay(1000);
  Serial.begin(115200);
  Serial.println();

  Server.on("/", rootPage);
  if (Portal.begin()) {
    Serial.println("WiFi connected: " + WiFi.localIP().toString());
  }
}

My error: Arduino: 1.8.8 (Windows 10), Board: "NodeMCU 0.9 (ESP-12 Module), 80 MHz, Flash, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from C:\Users\vladp\Documents\Arduino\libraries\AutoConnect-master\src\AutoConnectAux.cpp:13:0:

C:\Users\vladp\Documents\Arduino\libraries\AutoConnect-master\src\AutoConnectUploadImpl.h: In member function 'virtual void AutoConnectUploadSD::_close()':

C:\Users\vladp\Documents\Arduino\libraries\AutoConnect-master\src\AutoConnectUploadImpl.h:135:13: error: 'SDClassT' has no member named 'end'

     _media->end();

             ^

exit status 1
Error compiling for board NodeMCU 0.9 (ESP-12 Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
 
Thanks,
Vlad

Hieromon, 2019-04-16:
Place the include directive `#include <AutoConnect.h>`  behined both `#include <ESP8266WiFi.h>` and `#include <ESP8266WebServer.h>`.

petreanv, 2019-04-16:
The same error...

Thanks,
Vlad

Hieromon, 2019-04-16:
I could not find a mistake from the information you provided. The following sketch can be compiled with Arduino IDE 1.8.9 in my Windows10 environment.
And I was developing with 1.8.8 until just before the release.

Can you compile any of the examples? Or is the SD library mixed in your private library?

```cpp
#include <ESP8266WiFi.h> // Replace with WiFi.h for ESP32
#include <ESP8266WebServer.h> // Replace with WebServer.h for ESP32
#include <AutoConnect.h>

ESP8266WebServer Server; // Replace with WebServer for ESP32
AutoConnect Portal(Server);

void rootPage() {
  char content[] = "Hello, world";
  Server.send(200, "text/plain", content);
}

void setup() {
  delay(1000);
  Serial.begin(115200);
  Serial.println();

  Server.on("/", rootPage);
  if (Portal.begin()) {
    Serial.println("WiFi connected: " + WiFi.localIP().toString());
  }
}

void loop() {
}
```

petreanv, 2019-04-17:
The following sketch can be compiled with Arduino IDE 1.8.9 in my Windows 10 environment, but the same thing, the same error. I can compile the SD examples or SDWebServer and they work.

Hieromon, 2019-04-18:
Please, you post compilation messages with formatted them in **a code block** with the three back quotation. It may be long, but I expect to get clues to the cause.

petreanv, 2019-04-18:
`Arduino: 1.8.9 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from C:\Users\vladp\Documents\Arduino\libraries\AutoConnect-master\src\AutoConnectAux.cpp:13:0:

C:\Users\vladp\Documents\Arduino\libraries\AutoConnect-master\src\AutoConnectUploadImpl.h: In member function 'virtual void AutoConnectUploadSD::_close()':

C:\Users\vladp\Documents\Arduino\libraries\AutoConnect-master\src\AutoConnectUploadImpl.h:135:13: error: 'SDClassT' has no member named 'end'

     _media->end();

             ^

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

petreanv, 2019-04-18:
```
Arduino: 1.8.9 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from C:\Users\vladp\Documents\Arduino\libraries\AutoConnect-master\src\AutoConnectAux.cpp:13:0:

C:\Users\vladp\Documents\Arduino\libraries\AutoConnect-master\src\AutoConnectUploadImpl.h: In member function 'virtual void AutoConnectUploadSD::_close()':

C:\Users\vladp\Documents\Arduino\libraries\AutoConnect-master\src\AutoConnectUploadImpl.h:135:13: error: 'SDClassT' has no member named 'end'

     _media->end();

             ^

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

Hieromon, 2019-04-19:
Is this the compiler's whole messages?
I feel you didn’t fully understand what I meant. The cause of the problem is not AutoConnectUploadImpl.h The AutoConnect library can be built successfully with [Travis](https://travis-ci.org/Hieromon/AutoConnect). Therefore the cause is in your compilation environment.

The SDClassT in error is a redefinition of SDClass. SDClass is a class defined by the ESP8266 core SD library. If the Arduino environment is correct, its default location is:
`C:\Users\[username]\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\SD`

This error indicates that the compiler included another SD.h and the cause is that there are mixed libraries other than the ESP8266 core SD library, Or conflicts with other libraries in your environment. Depending on the cause, I may need to place the AutoConnectUploadHandler class into a namespace. But I still do not know.
To find out, I need to know how the compiler interpreted the include in your environment. The information you provided is missing to do that. So, I need whole messages outputted from the compiler.

nealmartini, 2019-04-21:
**I am getting the same error using the sketch below on my windows 10 computer**

#include <ESP8266WiFi.h>          // Replace with WiFi.h for ESP32
#include <ESP8266WebServer.h>     // Replace with WebServer.h for ESP32
#include <AutoConnect.h>

ESP8266WebServer Server;          // Replace with WebServer for ESP32
AutoConnect      Portal(Server);

void rootPage() {
	char content[] = "Hello, world";
	Server.send(200, "text/plain", content);
}
void setup() {
	delay(1000);
	Serial.begin(115200);
	Serial.println();

	Server.on("/", rootPage);
	if (Portal.begin()) {
		Serial.println("WiFi connected: " + WiFi.localIP().toString());
	}
}
void loop() {
	Portal.handleClient();
      }**_

 **My error message is** 

AutoConnectAux.cpp:13: In file included from
AutoConnectUploadImpl.h: In member function virtual void AutoConnectUploadSD::_close()
 
AutoConnectUploadImpl.h: 135:13: error: 'SDClassT' has no member named 'end
   _media->end()
Error compiling libraries
Build failed for project 'AutoConnect_1'

nealmartini, 2019-04-21:
Update----Compilation error went away when I updated the ESP8266 board support to version 2.5

petreanv, 2019-04-21:
> Update----Compilation error went away when I updated the ESP8266 board support to version 2.5

Work after update. 

Thanks,
Vlad

nealmartini, 2019-04-21:
Now I have another problem. My esp8266 does not show up in the settings->wifi->choose a network step as shown in the "Getting Started" wrtie up for Auto Connect.  I plan to add code to put the esp8266 into AP mode and see if that works. Petreanv---does your esp8266 show up in the wifi list on your smartphone???

nealmartini, 2019-04-22:
Portal.begin() is returning false. Sure why. I am using Windows 10, Node MCU 1.0 (ESP-12E Module). I have been able to get the module in AP mode with other programs, but not the above AutoConnect example. Done for today. Any help would be appreciated.

Hieromon, 2019-04-22:
@nealmartini 
> I have been able to get the module in AP mode with other programs

Please post a sketch of that "other programs".

nealmartini, 2019-04-22:
Here is a sketch that properly creates an access point that appears on my iPhone and can be connected to:

#include <ESP8266WiFi.h>
const char * APssid = "ESP8266";     // SSID used when in Access Point mode
WiFiServer server(80);

void setup() {
	Serial.begin(115200);
	WiFi.softAP(APssid);
	IPAddress myIP = WiFi.softAPIP();
	Serial.print("AP IP address: ");
	Serial.println(myIP);
	server.begin();
	Serial.println("Server started");
}

void loop() {
	handleWebClient();
}

void handleWebClient() {
	WiFiClient client = server.available();
	if (!client) {
		return;
	}
	Serial.println("new client");
	while (!client.available()) {
		delay(1);
	}
	delay(1);
	Serial.println("Client disonnected");
}

Also there were two ArduinoJson libraries to choose from when loading. Do you use the one by Benoit Blachon or the on by Arduino? I am using the one by Blanchon?

Thanks
Neal

Hieromon, 2019-04-23:
@nealmartini Thank you post. I confirmed your code. Your code is starting SoftAP from the beginning. AutoConnect does not launch SoftAP immediately by default. Please check that there no connection established with the first WiFi.begin(). Refer to the [AutoConnect::begin logic sequence](https://hieromon.github.io/AutoConnect/lsbegin.html).
If you want to start the AP from the beginning, you can use [an immediate start](https://hieromon.github.io/AutoConnect/apiconfig.html#immediatestart) by AutoConnectConfig.

Next, make sure there is no device with the IP of **192.168.244.1** in your local network. Its IP address is the default IP of AutoConnect SoftAP. If it in conflict with another device you can change it with AutoConnectConfig.
If there is no conflict with the IP address, please reproduce the problem with one of the examples and make sure to enable the AutoConnect debugging option and collect logs for the reproduction experiment. Also, enable the ESP8266 core debug too via Arduino IDE. After that, please post it and I'll investigate it. I may determine the immediate reason why AutoConnect::begin() returns false. 
Refer to [FAQ](https://hieromon.github.io/AutoConnect/faq.html#3-turn-on-the-debug-log-options) for details.

If not solve, here is the code that extracted from AutoConnect::begin() only the interface part with arduino esp8266 core and it excludes the AutoConnect particular process. Its meaning is to isolate the problem cause. Please try this code and collect logs both ESP8266 core and AutoConnect.

```cpp
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <DNSServer.h>

ESP8266WebServer WebServer;
DNSServer DNSServer;

wl_status_t waitForConnect(unsigned long timeout) {
  wl_status_t wifiStatus;

  Serial.print("Connecting");
  unsigned long st = millis();
  while ((wifiStatus = WiFi.status()) != WL_CONNECTED) {
    if (timeout) {
      if (millis() - st > timeout)
        break;
    }
    Serial.print('.');
    delay(300);
  }
  Serial.printf("%s IP:%s\n", wifiStatus == WL_CONNECTED ? "established" : "time out", WiFi.localIP().toString().c_str());
  return wifiStatus;
}

bool isIP(String ipStr) {
  for (uint8_t i = 0; i < ipStr.length(); i++) {
    char c = ipStr.charAt(i);
    if (c != '.' && (c < '0' || c > '9'))
      return false;
  }
  return true;
}

void setup() {
  delay(1000);
  Serial.begin(115200);
  Serial.println();

  WebServer.on("/", [] (void) {
    WebServer.send(200, "text/plain", "Hello, world");
  });

  WebServer.onNotFound([] (void) {
    String hostHeader = WebServer.hostHeader();
    if (!isIP(hostHeader) && hostHeader != WiFi.localIP().toString())
      WebServer.send(200, "text/plain", "This is the captive portal.");
    else
      WebServer.send(404, "text/plain", "Page not found");
  });

  WiFi.softAPdisconnect(true);
  WiFi.mode(WIFI_STA);
  delay(100);

  if (!WiFi.config(IPAddress(0U), IPAddress(0U), IPAddress(0U), IPAddress(0U), IPAddress(0U))) {
    Serial.println("WiFi.config failed, the sketch hung. Actually, AutoConnect::begin returns with false.");
    while (1)
      yield();
  }

  WiFi.begin();
  bool cs = waitForConnect(30000) == WL_CONNECTED;
  if (!cs) {
    WiFi.setAutoConnect(false);
    WiFi.disconnect(false);
    while (WiFi.status() == WL_CONNECTED) {
      delay(1);
      yield();
    }
    WiFi.mode(WIFI_AP_STA);
    while (WiFi.getMode() != WIFI_AP_STA) {
      delay(1);
      yield();
    }
    WiFi.softAPConfig(IPAddress(192, 168, 244, 1), IPAddress(192, 168, 244, 1), IPAddress(255, 255, 255, 0));
    WiFi.softAP("esp8266ap", "12345678", 1, false);
    do {
      delay(100);
      yield();
    } while (WiFi.softAPIP() != IPAddress(192, 168, 244, 1));
    IPAddress currentHostIP = WiFi.softAPIP();
    Serial.printf("SoftAP start %s\n", currentHostIP.toString().c_str());

    WebServer.begin();
    DNSServer.setErrorReplyCode(DNSReplyCode::NoError);
    DNSServer.start(53, "*", WiFi.softAPIP());

    while (WiFi.status() != WL_CONNECTED) {
      DNSServer.processNextRequest();
      WebServer.handleClient();
      yield();
    }
    DNSServer.stop();
  }
  else
    WebServer.begin();
}

void loop() {
  WebServer.handleClient();
}
```

Hieromon, 2019-04-23:
@nealmartini 

>Also there were two ArduinoJson libraries to choose from when loading. Do you use the one by Benoit Blachon or the on by Arduino? I am using the one by Blanchon?

This is mentioned in the [documentation](https://hieromon.github.io/AutoConnect/index.html#installation), section Additional library (Optional). It's https://github.com/bblanchon/ArduinoJson

nealmartini, 2019-04-24:
Thank you Hieromon for your comment "Please check that there no connection established with the first WiFi.begin()" It led me to a solution to my problem. I used WiFi.disconnect() with WIFI_STA mode to clear the last SSID memorized by a previous ESP8266 successful connection. This is also described in the FAQ called "Does not appear esp8266ap in smartphone". 

My successful sketch looks like this:

#include <ESP8266WiFi.h>          // Replace with WiFi.h for ESP32
#include <ESP8266WebServer.h>     // Replace with WebServer.h for ESP32
#include <AutoConnect.h>

ESP8266WebServer Server;          // Replace with WebServer for ESP32
AutoConnect      Portal(Server);

void rootPage() {
	char content[] = "Hello, world";
	Server.send(200, "text/plain", content);
}
void setup() {
	delay(1000);
	Serial.begin(115200);
	Serial.println();

	WiFi.mode(WIFI_STA);        ////////////////////addition to clear SSID//////   
	delay(100);                                                                                              //                      
	WiFi.begin();                                                                                           //                   
	if (WiFi.waitForConnectResult() == WL_CONNECTED) {                        // 
		WiFi.disconnect();                                                                          // 
		while (WiFi.status() == WL_CONNECTED) delay(100);                  //
	        }                                                                                                     //
	Serial.println("WiFi disconnected."); ///////////////////////////////////////
	
	Server.on("/", rootPage);
	Portal.begin();
	Serial.println("WiFi connected: " + WiFi.localIP().toString());
}

void loop() {
	Portal.handleClient();
}

I believe this issue can now be closed. Thanks again. AutoConnect is terrific!!

closed 2019-04-24 by petreanv