Hieromon/AutoConnect #59: How to turn on captive portal when the ESP8266 already connected to WiFi Connection

lukiusnatanael, 2019-04-19:
Hi, I just found this great library, and I think it's very useful and fits my needs. When I try to configure my ESP8266 WiFi through my phone, it works great and normal. But, I have problem when I want to change my ESP8266 WiFi connection while it's already connected to another WiFi connection that i already set before. I can't find my ESP8266 AP WiFi Connection and the only way to change the WiFi connection is by connect to the same router and find out the ESP8266 IP Address. Is there any other way to connect to the captive portal through ESP8266 AP?
Anyone have this same problem?

Hieromon, 2019-04-20:
I can not understand your request well, but you can not do that an ESP8266 connected to some AP connects to another AP via a captive portal again. It is not an ESP8266 problem or an AutoConnect problem.

> I can't find my ESP8266 AP WiFi Connection

What does this mean?

lukiusnatanael, 2019-04-20:
Sorry for my bad language. Let me explain what I mean. For example I have 2 WiFi connection called Wifi-A and Wifi-B. First I connect my ESP8266 to Wifi-A through my phone via captive portal. But the next day I want to change my ESP8266 connection to Wifi-B. How can I do this when my ESP8266 already connected to Wifi-A?

ageurtse, 2019-04-20:
Go to the ipadres of this module when in wifi-a network and add /_ac to the ip adress

For a device in my network it looks like this
http://192.168.1.185/_ac

lukiusnatanael, 2019-04-20:
> Go to the ipadres of this module when in wifi-a network and add /_ac to the ip adress
> 
> For a device in my network it looks like this
> http://192.168.1.185/_ac

I understand what you mean. But to do that I need to connect to Wifi-A and know the IP Address first and it's hard to make it portable. What I'm looking for is how to configure it via my ESP8266 AP without need to connect to Wifi-A first as the first time I set my ESP8266. Is there any other way to keep my ESP8266 act as AP and STA even it's already connect to a WiFi Connection so I can connect to the AP anytime I want?

ageurtse, 2019-04-20:
i don't think this is possible, maybe you could temporarly disable wifi-a, reboot your device. 
it can't connect so it will go to AP mode, now you can connect to wifi-B

or you need to make a ap-mode button on your device. when hitting this button the device goes to AP mode, from there you could connect to its wifi and alter the wifi settings.

lukiusnatanael, 2019-04-20:
> i don't think this is possible, maybe you could temporarly disable wifi-a, reboot your device.
> it can't connect so it will go to AP mode, now you can connect to wifi-B
>
> or you need to make a ap-mode [...]

So the only way to set my ESP8266 to AP-mode is by let it timeout when trying to connecting to saved WiFi? Is that what u mean?

ageurtse, 2019-04-20:
yes, that is what i mean, i don't think there is an easy different way.

lukiusnatanael, 2019-04-20:
> yes, that is what i mean, i don't think there is an easy different way.

Okay, thank you very much for your response and help. I'll hope I can find another option for this case.

Hieromon, 2019-04-20:
I think the method shown by @ageurtse is effective basically. I feel like what @lukiusnatanael wants is some gateway facility or the router. It needs to cross different networks, right?

lukiusnatanael, 2019-04-20:
> I think the method shown by @ageurtse is effective basically. I feel like what @lukiusnatanael wants is some gateway facility or the router. It needs to cross different networks, right?

Yeah, maybe more like I'm trying to set the ESP8266 as AP and STA at the same time always. So it's still can connected to a router but it's also always act as AP so I can reconfigure it everytime I want.

lukiusnatanael, 2019-04-20:
Finally I find the solution about how to turn the AP and STA at the same time so I can still reconfigure my ESP8266 anytime I want. 

In AutoConnect.cpp :
`bool AutoConnect::begin(const char* ssid, const char* passphrase, unsigned long timeout)`
line 99, I just change the WiFi.mode from WIFI_STA to WIFI_AP_STA and add this line below.
`WiFi.softAP(_apConfig.apid.c_str(), _apConfig.psk.c_str(), _apConfig.channel, _apConfig.hidden);`

Hieromon, 2019-04-20:
Indeed, the method you have shown will expose the ESP8266's SSID at AP_STA. However, if your request is that ESP 8266 will always stay at AP_STA, the following steps may be more appropriate:

1. Comment out the below line.
https://github.com/Hieromon/AutoConnect/blob/c2eeafe759be409a5b80b1fc1cf00c9344df6990/src/AutoConnect.cpp#L127

2. Configure the immediateStart & the autoReconnect.

3. Performs AutoConnect::config with the above option.

4. Performs AutoConnect::begin.

Please try this. If some problems occur, we can consult again.

lukiusnatanael, 2019-04-22:
> Indeed, the method you have shown will expose the ESP8266's SSID at AP_STA. However, if your request is that ESP8266 will always stay at AP_STA, the following steps may be more appropriate:
> 
> 1. Comment out the below line.
>    
>      
>        
>          [AutoConnect/src/AutoConnect.cpp](https://github.com/Hieromon/AutoConnect/blob/c2eeafe759be409a5b80b1fc1cf00c9344df6990/src/AutoConnect.cpp#L127)
>        
>        
>             Line 127
>          in
>          [c2eeafe](/Hieromon/AutoConnect/commit/c2eeafe759be409a5b80b1fc1cf00c9344df6990)
>        
>        
>        
>        
>    
>            
>              
>               _apConfig.autoReconnect = false;
> 2. Configure the immediateStart & the autoReconnect.
> 3. Performs AutoConnect::config with the above options.
> 4. Performs AutoConnect::begin.
> 
> This method may have some little difference with your request because it cannot still span different networks. But you can try this. If some problems occur, we can consult again.

Okay I'll try it, I'll tell you the result later.
Thankyou very much for your help.

scropion86, 2019-04-23:
`wifi.disconnect()   to dissconnect and clear the saved WiFi credentials (By SDK not the library)`

then

`portal.begin() to start AutoConnect again and join another wifi SSID `

Hieromon, 2019-04-23:
@scropion86 
That's right. It's simple.

Hieromon, 2019-05-29:
Since there was no response even after one month passed, closed this issue.
When some reports are posted, reopen it.

closed 2019-05-29 by Hieromon

antoniocgg, 2019-07-14:
Had the same issue.
Wanted to access my esp32 anytime, was it connected or not. The solution from [lukiusnatanael ](https://github.com/lukiusnatanael) Apr 20, worked.