Hieromon/AutoConnect #62: Notice for the update feature planned in v0.99

Hieromon, 2019-04-23:
To all contributors and Users,

Thank you for your interest in AutoConnect. I prepared for the prototype of the update server and application sketch which for sketch binary updating via OTA. I will implementation for v0.99 based on this prototype and will also add automatic updates by version level according to each device. 
Please post a comment if you have any requirements that I need to consider for implementation.

[image] [image] [image]

Hieromon, 2019-05-21:
Defer this feature enhancement and give priority to esp8266 core 2.5.2 support.
The release of the update feature is expected to be v0.9.10 or later.

huexpub, 2019-06-06:
maybe  esp32 ota feature ？ i add with autoconnect it crash.....

Hieromon, 2019-06-06:
The commit section of the development branch is unstable. It does not always work.
It should work with resources staged to the development branch currently and has already implemented most of the above update features.
You can try with the Update.ino example and updateserver.py script included in the resources of the development branch

Hieromon, 2019-07-18:
The update feature will be released next update turn as v1.0.0.
This topic title changed.
Development branch continues to use [Enhance/v099](https://github.com/Hieromon/AutoConnect/tree/Enhance/v099).

closed 2019-09-07 by Hieromon