Hieromon/AutoConnect #64: How to save and reload autoconnectselect and autoconnectradio? Not working for me.

smiley561, 2019-04-26:
Not working for me:

Save:
 SPIFFS.begin();
  File param = SPIFFS.open("/param", "w");
  if (!param) {
      Serial.println("file open failed");
  }  
  auxPage->saveElement(param, {"period"}); 
  param.close();
  SPIFFS.end();

Load:
SPIFFS.begin();
  File param = SPIFFS.open("/param", "r");
  if (param){
    auxPage->loadElement(param, "period");
  }
    else {
     Serial.println("Param nicht gefunden");
  }
  param.close();
  SPIFFS.end();

Pls. help!

Hieromon, 2019-04-26:
What kind of phenomenon is it? For example, the saved element appears in a custom web page, but the value is incorrect?
Please post your sketch code with formatted by the code block tag. (Reduce to the reproducible minimum code for the problem), and add the following information:
- Hardware module.
- Arduino core version.
- Your PC operating system.
- ArduinoJson version.

Hieromon, 2019-04-27:
Loading for the selected option of the AutoConnectSelect element was missing.
Enabled saving and restoring the selected state of the option by adding the selected attribute in AutoConnectSelect.
A patch for the fix was committed to the master. Documentation will be available with v0.9.9.

However, another trap was found during the investigation. Even if you access the page after executing loadElement, the selected option value may not be set. This may be a malfunction.  will be under investigation until the situation becomes clear.

smiley561, 2019-04-29:
Please do it for AutoConnectRadio too! It's the same behavior.

Hieromon, 2019-04-29:
I don't think so. As far as my diagnostic is concerned, that's not the problem with AutoConnectRadio. AutoConnectRadio has the `checked` attribute already.
The problem will closely be allied with storing the value of AutoConnectElements during page transitions. It occurs in the following cases:
- A custom web page containing any AutoConnectElement that allows parameter input.
- Its page allows loading parameters using AutoConnectSubmit.
- Execute loadElement in AutoConnectSubmit's uri handler for loading. It is for the elements of the first page.
- It will transition to the first page after loading. Or call the page from the menu.
- The loaded parameter is not reflected in elements.

If your case is different from the above, it is a different problem. I also need to find out about it.
Please post your sketch that I can reproduce, the partial code as loading and saving you provided is not sufficient. Refer to [FAQ](https://hieromon.github.io/AutoConnect/faq.html#4-reports-the-issue-to-autoconnect-github-repository).

Hieromon, 2019-04-29:
I committed the fix to the master. Please confirm.
If the phenomenon recurs in your sketch, the cause is another. Please post your sketch to reproduce the issue. (Reduce to the reproducible minimum code for the problem)

smiley561, 2019-04-30:
Sorry but it's not working:
```
AutoConnectAux* auxPage;
. . .
 SPIFFS.begin();
  File param = SPIFFS.open("/param", "r");

  if (param){
    // portal.where().loadElement(param);
    auxPage->loadElement(param, "rgAuswahl");
    auxPage->loadElement(param, "period");
    AutoConnectInput& wahl= auxPage->getElement<AutoConnectInput>("rgAuswahl");
    AutoConnectRadio& peri = auxPage->getElement<AutoConnectRadio>("period");

    Serial.print("Wahl: ");
    Serial.println(fahr.value); //-----------> O.K.
    Serial.print("Periode: ");
    Serial.println(String(peri.checked)); //------------------> Wrong!

  }
    else {
     Serial.println("/param nicht gefunden");
  }
 
  param.close();
  SPIFFS.end();
```
The value for "checked" in filesystem (/param in SPIFFS) is O.K.!!! I read it out with an other sketch.

Thank you for your efforts!

smiley561, 2019-04-30:
The same for AutoconnectSelect and ".selected"!

By the way: auxPage->saveElement(param) without fieldlist doesn't save anything.

Hieromon, 2019-05-01:
In this case, the problem was complicated, but I think it was solved. There were three problems that you found. Each phenomenon and cause are as follows:

1. Loaded element values ​​are not restored to the custom web page.
    The cause is that the process of saving data from the post body to AutoConnectElements during the page transition is wrong. This problem does not depend on the element type. Fixed.

2. AutoConnectRaio or AutoConnectSelect value does not load.
    It was by two causes.

    2-1. Only the first element of the JSON document is loaded.
        It is a bug of the AutoConnect. It happens even if it is not AutoConnectRadio or AutoConnectSelect. Only the saved top element was loaded. Coincidentally, it was AutoConnectInput. Fixed.

    2-2. Your sketch is wrong.
        Your sketch provides the file stream to the loadElement. The loadElement passes the stream as is to the ArduinoJson parser, which parses the stream to the end and constructs a JSON object. After that, the loadElement loads only the specified element from the array of JSON objects. As of its moment, the stream has already reached the end. So you're passing the stream that reached the end with the second loadElement. To read the stream again, you need to rewind the file position.

    ```cpp
    File param = SPIFFS.open("/param", "r);
    aux->loadElement(param, "rgAuswahl");
    param.seek(0, SeekSet);	// -------------> rewind
    aux->loadElement(param, "period");
    param.close();
    ```

3. The saveElement without field list doesn't save elements.
    It occurs with ArduinoJson V6. It works fine with ArduinoJson V5, but this is only lucky.
    Calculation of JSON document buffer size was incorrect. ArduinoJson V5 dynamically increases the JSON buffer size, but V6 does not. AutoConnect under-computed buffer size. Fixed.

4. New supports
    The following features have been added along with this measure.

    4-1. The selected attribute for the AutoConnectSelect.
        You can save and load AutoConnectSelect selection values.

    4-2. Load multiple elements.
        You can specify multiple elements with the AutoConnectAux::loadElement function.

    ```cpp
    File param = SPIFFS.open("/param", "r);
    aux->loadElement(param, { "rgAuswahl", "period" });
    param.close();
    ```

I was able to find serious malfunction of AutoConnect by your report. Thank you for your contribution.
Please confirm the revised version at the master and let me know the result. If it is good, I will re-release it as v0.9.8.

smiley561, 2019-05-02:
Everything is fine now. Great work!!!

closed 2019-05-02 by smiley561