Hieromon/AutoConnect #67: Issue with Library thinger.io. Can not open Portal.

Keyj1n, 2019-04-28:
Hello there,

i have the problem that when I use AutoConnect with ThingerIO Library i can not open the Portal at the uri "/ac".
Here is my code in the loop method

`void loop()
{
  thing.handle();
  Portal.handleClient();
}`

Have you any Idea what could cause this Issue?
If you need more information please let me know.

Hieromon, 2019-04-29:
It is insufficient memory. 
How much heap does the Thinger.io library consume? I examined the library description of thinger.io but could not find it. You can check the free size heap at runtime with the Thinger.io by ESP.getFreeHeap excluding AutoConnect.
AutoConnect requires a free heap of at least 12K bytes for HTML generation. Refer to [FAQ](https://hieromon.github.io/AutoConnect/faq.html#how-much-memory-does-autoconnect-consume) and related issue #51.
I don't know the heap size consumed by the Thinger.io library. But The remaining 5.6K bytes in _ac HTML generation will due to the many memory consumptions of Theinger.io, I think.

Keyj1n, 2019-04-29:
Thanks for the Information!
It's like 20096 - 20600 
Any Idea how I could solve this problem? Or is there nothing that i can do about it.
Sry im pretty nooby in the microcontroller thing.

Hieromon, 2019-04-29:
> It's like 20096 - 20600

Thinger.io consumes huge memory. I don't think there is an easy way to increase available memory with ESP8266. It is the limitation of Thinger.io, right? Also, Thinger.io itself should equip some kind of connection utility if consumption much memory, I think.
With AutoConnect, in the simpler sketch below, the heap remaining capacity immediately after setup is about **46 Kbytes** for the NodeMCU 1.0 module. 
```cpp
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <AutoConnect.h>

ESP8266WebServer server;
AutoConnect portal(server);
static bool onetime = true;

void setup() {
  delay(1000);
  Serial.begin(115200);
  Serial.println();
  portal.begin();
  Serial.println("connected");
  server.begin();
}

void loop() {
  portal.handleClient();
  if (onetime) {
    Serial.println(ESP.getFreeHeap());
    onetime = false;
  }
}
```
ESP32 has more memory.

Keyj1n, 2019-04-30:
@Hieromon thank you very much for this detailed explanation. I switched to Thingsboard which is not consuming so much ram.
Works now fine.

closed 2019-04-30 by Keyj1n