Hieromon/AutoConnect #71: How to implement websocket in simple example.

ajinkyaw, 2019-05-05:
I am trying to use WebSocketsServer library, and this is how i am declaring and initiating the instances.

       WebServer server;
       WebSocketsServer    webSocket = WebSocketsServer(81);
       AutoConnect       Portal(server);

      void setup(){

         Portal.join({ Timezone });        // Register aux. page
         webSocket.begin();
         webSocket.onEvent(webSocketEvent);
                
         server.on("/", rootPage);
         server.on("/start", startPage);   // Set NTP server trigger handler
         server.on("/chk", checkConnect);
               
         if (Portal.begin()) {
            Serial.println("WiFi connected: " + WiFi.localIP().toString());
         }

      }
 
but i am not be able to get the websocket initiated.
can you guide please.

Hieromon, 2019-05-06:
The `WebSocketsServer::begin` function provided by Links2004's WebSockets library internally invokes `WiFiServer::begin`. Therefore, an ESP module must already have a WiFi connection at this point and must be transparent for TCP.
AutoConnect executes WiFi.begin inside [the AutoConnect::begin logic](https://hieromon.github.io/AutoConnect/lsbegin.html) and returns true if the connection establish. So, `WebSocketsServer::begin` must be called after AutoConnect::begin succeeded. That's reversed in your sketch.

Here is an example that provides a logic scheme using the WebSocketsServer with AutoConnect.

```cpp
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <WebSocketsServer.h>
#include <Hash.h>
#include <AutoConnect.h>

ESP8266WebServer server(80);
WebSocketsServer webSocket(81);
AutoConnect      portal(server);
AutoConnectAux   AUX;

void setup() {
  delay(1000);
  Serial.begin(115200);
  Serial.println();

  portal.join({ AUX }); // Join the AutoConnectAux
  if (portal.begin()) { // <--- Should be first.
    webSocket.begin();  // <--- After AutoConnect::begin
    webSocket.onEvent([] (uint8_t num, WStype_t type, uint8_t * payload, size_t length) {
      switch(type) {
      case WStype_DISCONNECTED:
        DEBUG_ESP_PORT.printf("[%u] Disconnected!\n", num);
        break;
      case WStype_CONNECTED: {
        IPAddress ip = webSocket.remoteIP(num);
        DEBUG_ESP_PORT.printf("[%u] Connected from %s\n", num, ip.toString().c_str());
        webSocket.sendTXT(num, "Connected");
        }
        break;
      case WStype_TEXT:
        DEBUG_ESP_PORT.printf("[%u] get Text: %s\n", num, payload);
        // send some response to the client
        // webSocket.sendTXT(num, "message here");
        break;
      }
    });
  }
}

void loop() {
  webSocket.loop();
  portal.handleClient();
}
```

And if you run the following HTML in a browser that can reach the ESP module, you will be able to see the arrived message displayed on the serial monitor.

```
[WS-Server] Server Started.
...
[WS][0][handleWebsocket] text: Hello, world
```

```html
<!DOCTYPE html>
<html>
<body>
<h2>WebSocket client</h2>
<button type="button" onclick="SendText();">Click to send</button>
<script type="text/javascript">
function SendText() {
  var ws = new WebSocket("ws://YOUR_WebSocketsServer_IP_ADDRESS:81");
  ws.onopen = function(event) {
    ws.send('Hello, world');
  };
}
</script>
</body>
</html> 
```

By the way, Is your sketch like this one?
Get the current time from one of the NTP servers with ESP8266 (or ESP32?) and display it on a browser with the AutoConnectAux page equipped with a client JavaScript that gets it with WebSocket.

Hieromon, 2019-06-03:
Since there was no response even after one month passed, closed this issue.
When some reports are posted, reopen it.

closed 2019-06-03 by Hieromon

playground, 2022-11-07:
Hi @Hieromon,  getting
```WebServer.cpp649] _handleRequest(): request handler not found``` when tried to connect to websocket server.

Hieromon, 2022-11-09:
@playground, ```request handler not found``` messages can be output from the ESP32 Arduino Core WebServer even in the typical case. It is not a bug. And it has no relation with WebSocketsServer; I already know and have verified that AutoConnect can coexist with the [WebSocket](https://github.com/Links2004/arduinoWebSockets) library.

If your post points to a possible problem with AutoConnect, please provide me with more information. If you can provide me with the appropriate information, I will have the opportunity to investigate your encountered phenomenon in more detail.

See [here](https://hieromon.github.io/AutoConnect/faq.html#4-reports-the-issue-to-autoconnect-github-repository).