Hieromon/AutoConnect #73: main menu colors

smiley561, 2019-05-09:
Is it possible to change text/background colors of the main menu?

Hieromon, 2019-05-10:
I made it possible to customize menu text/background color statically. You can update your AutoConnect library with the latest commit on the master to use color customization.
The menu color scheme has been separated to [`AutoConnectLabels.h`](https://github.com/Hieromon/AutoConnect/blob/master/src/AutoConnectLabels.h). You can change the color scheme of the menu with the following three color codes. The color code also accepts CSS standard color names.

- #define AUTOCONNECT_MENUCOLOR_TEXT
  Defines the menu text color.
https://github.com/Hieromon/AutoConnect/blob/779bd164223ca8b4c20b3a65453c27c822d7329d/src/AutoConnectLabels.h#L74

- #define AUTOCONNECT_MENUCOLOR_BACKGROUND
  Defines the menu background color.
https://github.com/Hieromon/AutoConnect/blob/779bd164223ca8b4c20b3a65453c27c822d7329d/src/AutoConnectLabels.h#L80

- #define AUTOCONNECT_MENUCOLOR_ACTIVE
  Defines the active menu item background color.
https://github.com/Hieromon/AutoConnect/blob/779bd164223ca8b4c20b3a65453c27c822d7329d/src/AutoConnectLabels.h#L86

I'll pick up some color schemes.

### Indigo
```cpp
#define AUTOCONNECT_MENUCOLOR_TEXT        "#ffa500"
#define AUTOCONNECT_MENUCOLOR_BACKGROUND  "#1a237e"
#define AUTOCONNECT_MENUCOLOR_ACTIVE      "#283593"
```
[image]

### Dimgray
```cpp
#define AUTOCONNECT_MENUCOLOR_TEXT        "#fffacd"
#define AUTOCONNECT_MENUCOLOR_BACKGROUND  "#696969"
#define AUTOCONNECT_MENUCOLOR_ACTIVE      "#808080"
```
[image]

### Brown
```cpp
#define AUTOCONNECT_MENUCOLOR_TEXT        "#e6e6fa"
#define AUTOCONNECT_MENUCOLOR_BACKGROUND  "#3e2723"
#define AUTOCONNECT_MENUCOLOR_ACTIVE      "#4e342e"
```
[image]

smiley561, 2019-05-11:
[image]

Nice thing but not all items have changed their color.

Hieromon, 2019-05-11:
I cannot reproduce. iOS Safari is Ok, Chrome is Ok, MS Edge is Ok.
Are some extensions affect to this phenomenon?

smiley561, 2019-05-12:
Sorry, it was an addon for changing colors of visited pages.

closed 2019-05-12 by smiley561