Hieromon/AutoConnect #75: Unable compile Simple example

stuffshit, 2019-05-18:
Getting an error while trying to compile "Simple" example. Any help would be appreciated

```
Arduino: 1.8.9 (Mac OS X), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (1M SPIFFS), v2 Lower Memory, Serial, HTTP_SERVER, Only Sketch, 921600"

In file included from /Users/<user>/Documents/Arduino/libraries/AutoConnect/src/AutoConnect.h:31:0,
                 from /Users/<user>/Documents/Arduino/libraries/AutoConnect/src/AutoConnectAux.cpp:10:
/Users/<user>/Documents/Arduino/libraries/AutoConnect/src/AutoConnectDefs.h:138:33: error: 'SD_CHIP_SELECT_PIN' was not declared in this scope
 #define AUTOCONNECT_SD_CS       SD_CHIP_SELECT_PIN
                                 ^
/Users/<user>/Documents/Arduino/libraries/AutoConnect/src/AutoConnectUploadImpl.h:108:68: note: in expansion of macro 'AUTOCONNECT_SD_CS'
   explicit AutoConnectUploadSD(SDClassT& media, const uint8_t cs = AUTOCONNECT_SD_CS, const uint32_t speed = AUTOCONNECT_SD_SPEED) : _media(&media), _cs(cs), _speed(speed) {}
                                                                    ^
In file included from /Users/<user>/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/libraries/ESP8266SdFat/src/SdCard/SdSpiCard.h:33:0,
                 from /Users/<user>/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/libraries/ESP8266SdFat/src/BlockDriver.h:32,
                 from /Users/<user>/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/libraries/ESP8266SdFat/src/SdFat.h:32,
                 from /Users/<user>/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/libraries/SDFS/src/SDFS.h:36,
                 from /Users/<user>/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/libraries/SD/src/SD.h:25,
                 from /Users/<user>/Documents/Arduino/libraries/AutoConnect/src/AutoConnectUploadImpl.h:21,
                 from /Users/<user>/Documents/Arduino/libraries/AutoConnect/src/AutoConnectAux.cpp:13:
/Users/<user>/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/libraries/ESP8266SdFat/src/SdCard/SdInfo.h:107:70: error: could not convert 'SPISettings(4000000u, 1, 0)' from 'SPISettings' to 'uint32_t {aka unsigned int}'
 #define SD_SCK_HZ(maxSpeed) SPISettings(maxSpeed, MSBFIRST, SPI_MODE0)
                                                                      ^
/Users/<user>/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/libraries/ESP8266SdFat/src/SdCard/SdInfo.h:128:30: note: in expansion of macro 'SD_SCK_HZ'
 #define SPI_HALF_SPEED       SD_SCK_HZ(4000000)
                              ^
/Users/<user>/Documents/Arduino/libraries/AutoConnect/src/AutoConnectDefs.h:147:33: note: in expansion of macro 'SPI_HALF_SPEED'
 #define AUTOCONNECT_SD_SPEED    SPI_HALF_SPEED
                                 ^
/Users/<user>/Documents/Arduino/libraries/AutoConnect/src/AutoConnectUploadImpl.h:108:110: note: in expansion of macro 'AUTOCONNECT_SD_SPEED'
   explicit AutoConnectUploadSD(SDClassT& media, const uint8_t cs = AUTOCONNECT_SD_CS, const uint32_t speed = AUTOCONNECT_SD_SPEED) : _media(&media), _cs(cs), _speed(speed) {}
                                                                                                              ^
In file included from /Users/<user>/Documents/Arduino/libraries/AutoConnect/src/AutoConnectAux.cpp:13:0:
/Users/<user>/Documents/Arduino/libraries/AutoConnect/src/AutoConnectUploadImpl.h: In member function 'virtual bool AutoConnectUploadSD::_open(const char*, const char*)':
/Users/<user>/Documents/Arduino/libraries/AutoConnect/src/AutoConnectUploadImpl.h:114:34: error: no matching function for call to 'SDClass::begin(uint8_t&, uint8_t&)'
     if (_media->begin(_cs, _speed)) {
                                  ^
/Users/<user>/Documents/Arduino/libraries/AutoConnect/src/AutoConnectUploadImpl.h:114:34: note: candidate is:
In file included from /Users/<user>/Documents/Arduino/libraries/AutoConnect/src/AutoConnectUploadImpl.h:21:0,
                 from /Users/<user>/Documents/Arduino/libraries/AutoConnect/src/AutoConnectAux.cpp:13:
/Users/<user>/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/libraries/SD/src/SD.h:34:13: note: boolean SDClass::begin(uint8_t, SPISettings)
     boolean begin(uint8_t csPin, SPISettings cfg = SPI_HALF_SPEED) {
             ^
/Users/<user>/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/libraries/SD/src/SD.h:34:13: note:   no known conversion for argument 2 from 'uint8_t {aka unsigned char}' to 'SPISettings'
In file included from /Users/<user>/Documents/Arduino/libraries/AutoConnect/src/AutoConnectAux.cpp:14:0:
/Users/<user>/Documents/Arduino/libraries/AutoConnect/src/AutoConnectElementBasisImpl.h: In member function 'bool AutoConnectFileBasis::attach(ACFile_t)':
/Users/<user>/Documents/Arduino/libraries/AutoConnect/src/AutoConnectElementBasisImpl.h:88:43: error: call to 'AutoConnectUploadSD::AutoConnectUploadSD(SDClassT&, uint8_t, uint32_t)' uses the default argument for parameter 2, which is not yet defined
     handlerSD = new AutoConnectUploadSD(SD);
                                           ^
/Users/<user>/Documents/Arduino/libraries/AutoConnect/src/AutoConnectElementBasisImpl.h:88:43: error: cannot convert 'SPISettings' to 'uint32_t {aka unsigned int}' for argument '3' to 'AutoConnectUploadSD::AutoConnectUploadSD(SDClassT&, uint8_t, uint32_t)'
Multiple libraries were found for "SD.h"
 Used: /Users/<user>/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/libraries/SD
 Not used: /Applications/Arduino.app/Contents/Java/libraries/SD
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
```

jumpjack, 2019-05-18:
Same issue here.
Windows 10
Arduino IDE 1.8.9 Windows Store

jumpjack, 2019-05-18:
Is it all right here?

`#if defined(ARDUINO_ARCH_ESP8266)
#include <core_version.h>
#include <ESP8266WiFi.h>
#elif defined(ARDUINO_ARCH_ESP32)
#include <WiFi.h>
#include <SPIFFS.h>
#endif
#include <SPI.h>
#include <SD.h>
#define FS_NO_GLOBALS
#include <FS.h>`

(from AutoConnectUploadImpl.h )

Hieromon, 2019-05-18:
The SD file system has changed significantly with ESP8266 core 2.5.1. https://github.com/esp8266/Arduino/pull/5525
There are also many more patches applied and I am still evaluating the ESP8266 core 2.5.1. AutoConnect will support core 2.5.1 with v0.9.9.

jumpjack, 2019-05-18:
Is this file needed for basic sketches? I mean, I am not uploading anything, I am just using GET to receive some files.

Anyway I tried downgrading ESP8266 libraries to 2.5.0: these errors disappear... but other appears.
I'll keep downgrading until I'll get something working. :-)

jumpjack, 2019-05-18:
Errors with ESP8266 core 2.5.0:

```

In file included from F:\documenti\Arduino\libraries\AutoConnect-master\src/AutoConnect.h:16:0,

                 from F:\documenti\Arduino\autoconnect-test\autoconnect-test.ino:1:

F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0\libraries\DNSServer\src/DNSServer.h:67:13: error: cannot declare field 'DNSServer::_udp' to be of abstract type 'WiFiUDP'

     WiFiUDP _udp;

             ^

In file included from F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0\libraries\DNSServer\src/DNSServer.h:3:0,

                 from F:\documenti\Arduino\libraries\AutoConnect-master\src/AutoConnect.h:16,

                 from F:\documenti\Arduino\autoconnect-test\autoconnect-test.ino:1:

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\libraries\WiFi\src/WiFiUdp.h:27:7: note:   because the following virtual functions are pure within 'WiFiUDP':

 class WiFiUDP : public UDP {

       ^

In file included from C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\libraries\WiFi\src/WiFiUdp.h:23:0,

                 from F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0\libraries\DNSServer\src/DNSServer.h:3,

                 from F:\documenti\Arduino\libraries\AutoConnect-master\src/AutoConnect.h:16,

                 from F:\documenti\Arduino\autoconnect-test\autoconnect-test.ino:1:

F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/Udp.h:82:27: note: 	virtual IPAddress UDP::remoteIP() const

         virtual IPAddress remoteIP() const =0;

                           ^

F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/Udp.h:84:26: note: 	virtual uint16_t UDP::remotePort() const

         virtual uint16_t remotePort() const =0;

                          ^

Più di una libreria trovata per "WiFiUdp.h"
Usata: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\libraries\WiFi
Non usata: F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266WiFi
Più di una libreria trovata per "ESP8266WebServer.h"
Usata: F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266WebServer
Non usata: F:\documenti\Arduino\libraries\WebServer_tng-master
Più di una libreria trovata per "SD.h"
Usata: F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0\libraries\SD
Non usata: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\libraries\SD
exit status 1
Errore durante la compilazione per la scheda Generic ESP8266 Module.

```

Hieromon, 2019-05-18:
Can you compile an example that uses ESP8266WebServer and DNS without AutoConnect in your environment?

jumpjack, 2019-05-19:
Errors for core 2.4.0 with an empty skecth (only "include <AutoConnect.h>"):
```

In file included from F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src/WiFiServerSecureBearSSL.h:24:0,

                 from F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src/WiFiServerSecure.h:21,

                 from F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src/ESP8266WiFi.h:41,

                 from F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WebServer\src\ESP8266WebServer.h:29,

                 from F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WebServer\src\ESP8266WebServer.cpp:28:

F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src/WiFiClientSecureBearSSL.h:191:36: error: expected ')' before '*' token

     WiFiClientSecure(ClientContext *client, const BearSSLX509List *chain, unsigned cert_issuer_key_type,

                                    ^

F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src/WiFiClientSecureBearSSL.h:193:35: error: expected ')' before '*' token

     WiFiClientSecure(ClientContext* client, const BearSSLX509List *chain, const BearSSLPrivateKey *sk,

                                   ^

In file included from F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src/WiFiServerSecureBearSSL.h:24:0,

                 from F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src/WiFiServerSecure.h:21,

                 from F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src/ESP8266WiFi.h:41,

                 from F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WebServer\src\ESP8266WebServer.h:29,

                 from F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WebServer\src\ESP8266WebServer.cpp:28:

F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src/WiFiClientSecureBearSSL.h:36:5: error: 'BearSSL::WiFiClientSecure::~WiFiClientSecure()' marked override, but does not override

     ~WiFiClientSecure() override;

     ^

F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src/WiFiClientSecureBearSSL.h:39:9: error: 'int BearSSL::WiFiClientSecure::connect(String, uint16_t)' marked override, but does not override

     int connect(const String host, uint16_t port) override;

         ^

F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src/WiFiClientSecureBearSSL.h:44:12: error: 'size_t BearSSL::WiFiClientSecure::write_P(const char*, size_t)' marked override, but does not override

     size_t write_P(PGM_P buf, size_t size) override;

            ^

F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src/WiFiClientSecureBearSSL.h:56:12: error: 'size_t BearSSL::WiFiClientSecure::peekBytes(uint8_t*, size_t)' marked override, but does not override

     size_t peekBytes(uint8_t *buffer, size_t length) override;

            ^

In file included from F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src/WiFiClientSecure.h:23:0,

                 from F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src/ESP8266WiFi.h:42,

                 from F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WebServer\src\ESP8266WebServer.h:29,

                 from F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WebServer\src\ESP8266WebServer.cpp:28:

F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src/WiFiClientSecureAxTLS.h:88:33: error: expected ')' before '*' token

   WiFiClientSecure(ClientContext* client, bool usePMEM, const uint8_t *rsakey, int rsakeyLen, const uint8_t *cert, int certLen);
[... 445 lines cut ...]
          ^

In file included from F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WebServer\src\ESP8266WebServerSecureBearSSL.h:26:0,

                 from F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WebServer\src\ESP8266WebServerSecureBearSSL.cpp:28:

F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h: In member function 'virtual size_t ESP8266WebServer::_currentClientWrite_P(const char*, size_t)':

F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:140:83: error: 'class WiFiClient' has no member named 'write_P'

   virtual size_t _currentClientWrite_P(PGM_P b, size_t l) { return _currentClient.write_P( b, l ); }

                                                                                   ^

F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WebServer\src\ESP8266WebServerSecureBearSSL.cpp: In member function 'virtual void BearSSL::ESP8266WebServerSecure::close()':

F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WebServer\src\ESP8266WebServerSecureBearSSL.cpp:152:17: error: 'class BearSSL::WiFiServerSecure' has no member named 'close'

   _serverSecure.close();

                 ^

Errore durante la compilazione per la scheda Generic ESP8266 Module.

```

jumpjack, 2019-05-19:
2.4.0 errors:

```
In file included from C:\Users\luca\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:41:0,

                 from C:\Users\luca\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WebServer\src\ESP8266WebServer.h:29,

                 from C:\Users\luca\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WebServer\src\Parsing.cpp:25:

C:\Users\luca\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi\src/WiFiClientSecure.h:34:3: error: 'WiFiClientSecure::~WiFiClientSecure()' marked override, but does not override

   ~WiFiClientSecure() override;

   ^

C:\Users\luca\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi\src/WiFiClientSecure.h:39:7: error: 'int WiFiClientSecure::connect(String, uint16_t)' marked override, but does not override

   int connect(const String host, uint16_t port) override;

       ^

C:\Users\luca\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi\src/WiFiClientSecure.h:51:10: error: 'size_t WiFiClientSecure::peekBytes(uint8_t*, size_t)' marked override, but does not override

   size_t peekBytes(uint8_t *buffer, size_t length) override;

          ^

In file included from C:\Users\luca\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:41:0,

                 from C:\Users\luca\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WebServer\src\ESP8266WebServer.h:29,

                 from C:\Users\luca\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WebServer\src\ESP8266WebServer.cpp:28:

C:\Users\luca\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi\src/WiFiClientSecure.h:34:3: error: 'WiFiClientSecure::~WiFiClientSecure()' marked override, but does not override

   ~WiFiClientSecure() override;

   ^

C:\Users\luca\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi\src/WiFiClientSecure.h:39:7: error: 'int WiFiClientSecure::connect(String, uint16_t)' marked override, but does not override

   int connect(const String host, uint16_t port) override;

       ^

C:\Users\luca\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi\src/WiFiClientSecure.h:51:10: error: 'size_t WiFiClientSecure::peekBytes(uint8_t*, size_t)' marked override, but does not override

   size_t peekBytes(uint8_t *buffer, size_t length) override;

          ^

C:\Users\luca\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WebServer\src\ESP8266WebServer.cpp: In constructor 'ESP8266WebServer::ESP8266WebServer(IPAddress, int)':

C:\Users\luca\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WebServer\src\ESP8266WebServer.cpp:55:17: error: no matching function for call to 'WiFiServer::WiFiServer(IPAddress&, int&)'

 , _chunked(false)

                 ^

C:\Users\luca\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WebServer\src\ESP8266WebServer.cpp:55:17: note: candidates are:

In file included from C:\Users\luca\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WebServer\src\ESP8266WebServer.cpp:26:0:

C:\Program Files (x86)\Arduino\libraries\WiFi\src/WiFiServer.h:36:3: note: WiFiServer::WiFiServer(uint16_t)

   WiFiServer(uint16_t);

   ^

C:\Program Files (x86)\Arduino\libraries\WiFi\src/WiFiServer.h:36:3: note:   candidate expects 1 argument, 2 provided

C:\Program Files (x86)\Arduino\libraries\WiFi\src/WiFiServer.h:31:7: note: constexpr WiFiServer::WiFiServer(const WiFiServer&)

 class WiFiServer : public Server {

       ^

C:\Program Files (x86)\Arduino\libraries\WiFi\src/WiFiServer.h:31:7: note:   candidate expects 1 argument, 2 provided

C:\Program Files (x86)\Arduino\libraries\WiFi\src/WiFiServer.h:31:7: note: constexpr WiFiServer::WiFiServer(WiFiServer&&)

C:\Program Files (x86)\Arduino\libraries\WiFi\src/WiFiServer.h:31:7: note:   candidate expects 1 argument, 2 provided

C:\Users\luca\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WebServer\src\ESP8266WebServer.cpp: In member function 'void ESP8266WebServer::close()':

C:\Users\luca\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WebServer\src\ESP8266WebServer.cpp:328:11: error: 'class WiFiServer' has no member named 'close'

   _server.close();

           ^

C:\Users\luca\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WebServer\src\ESP8266WebServer.cpp: In member function 'void ESP8266WebServer::sendContent_P(const char*, size_t)':

C:\Users\luca\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WebServer\src\ESP8266WebServer.cpp:458:18: error: 'class WiFiClient' has no member named 'write_P'

   _currentClient.write_P(content, size);

                  ^

Più di una libreria trovata per "SD.h"
Usata: C:\Users\luca\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\SD
Non usata: C:\Program Files (x86)\Arduino\libraries\SD
Più di una libreria trovata per "WiFiUdp.h"
Usata: C:\Program Files (x86)\Arduino\libraries\WiFi
Non usata: C:\Users\luca\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi
Più di una libreria trovata per "ESP8266WebServer.h"
Usata: C:\Users\luca\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WebServer
Non usata: F:\documenti\Arduino\libraries\WebServer_tng-master
exit status 1
Errore durante la compilazione per la scheda Generic ESP8266 Module.

```

jumpjack, 2019-05-19:
Core 2.5.1 not compiling.
Core 2.5.0 ok

Errors in 2.5.1:
```

In file included from F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.5.1\libraries\ESP8266SdFat\src/SdCard/SdSpiCard.h:33:0,

                 from F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.5.1\libraries\ESP8266SdFat\src/BlockDriver.h:32,

                 from F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.5.1\libraries\ESP8266SdFat\src/SdFat.h:32,

                 from F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.5.1\libraries\SDFS\src/SDFS.h:36,

                 from F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.5.1\libraries\SD\src/SD.h:25,

                 from F:\documenti\Arduino\libraries\AutoConnect-master\src\AutoConnectUploadImpl.h:22,

                 from F:\documenti\Arduino\libraries\AutoConnect-master\src\AutoConnectAux.cpp:13:

F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.5.1\libraries\ESP8266SdFat\src/SdCard/SdInfo.h:107:70: error: could not convert 'SPISettings(4000000u, 1, 0)' from 'SPISettings' to 'uint32_t {aka unsigned int}'

 #define SD_SCK_HZ(maxSpeed) SPISettings(maxSpeed, MSBFIRST, SPI_MODE0)

                                                                      ^

F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.5.1\libraries\ESP8266SdFat\src/SdCard/SdInfo.h:128:30: note: in expansion of macro 'SD_SCK_HZ'

 #define SPI_HALF_SPEED       SD_SCK_HZ(4000000)

                              ^

F:\documenti\Arduino\libraries\AutoConnect-master\src\AutoConnectDefs.h:151:33: note: in expansion of macro 'SPI_HALF_SPEED'

 #define AUTOCONNECT_SD_SPEED    SPI_HALF_SPEED

                                 ^

F:\documenti\Arduino\libraries\AutoConnect-master\src\AutoConnectUploadImpl.h:109:110: note: in expansion of macro 'AUTOCONNECT_SD_SPEED'

   explicit AutoConnectUploadSD(SDClassT& media, const uint8_t cs = AUTOCONNECT_SD_CS, const uint32_t speed = AUTOCONNECT_SD_SPEED) : _media(&media), _cs(cs), _speed(speed) {}

                                                                                                              ^

In file included from F:\documenti\Arduino\libraries\AutoConnect-master\src\AutoConnectAux.cpp:13:0:

F:\documenti\Arduino\libraries\AutoConnect-master\src\AutoConnectUploadImpl.h: In member function 'virtual bool AutoConnectUploadSD::_open(const char*, const char*)':

F:\documenti\Arduino\libraries\AutoConnect-master\src\AutoConnectUploadImpl.h:115:34: error: no matching function for call to 'SDClass::begin(uint8_t&, uint8_t&)'

     if (_media->begin(_cs, _speed)) {

                                  ^

F:\documenti\Arduino\libraries\AutoConnect-master\src\AutoConnectUploadImpl.h:115:34: note: candidate is:

In file included from F:\documenti\Arduino\libraries\AutoConnect-master\src\AutoConnectUploadImpl.h:22:0,

                 from F:\documenti\Arduino\libraries\AutoConnect-master\src\AutoConnectAux.cpp:13:

F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.5.1\libraries\SD\src/SD.h:34:13: note: boolean SDClass::begin(uint8_t, SPISettings)

     boolean begin(uint8_t csPin, SPISettings cfg = SPI_HALF_SPEED) {

             ^

F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.5.1\libraries\SD\src/SD.h:34:13: note:   no known conversion for argument 2 from 'uint8_t {aka unsigned char}' to 'SPISettings'

In file included from F:\documenti\Arduino\libraries\AutoConnect-master\src\AutoConnectAux.cpp:14:0:

F:\documenti\Arduino\libraries\AutoConnect-master\src\AutoConnectElementBasisImpl.h: In member function 'bool AutoConnectFileBasis::attach(ACFile_t)':

F:\documenti\Arduino\libraries\AutoConnect-master\src\AutoConnectElementBasisImpl.h:88:43: error: cannot convert 'SPISettings' to 'uint32_t {aka unsigned int}' for argument '3' to 'AutoConnectUploadSD::AutoConnectUploadSD(SDClassT&, uint8_t, uint32_t)'

     handlerSD = new AutoConnectUploadSD(SD);

                                           ^

Più di una libreria trovata per "ESP8266WebServer.h"
Usata: F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.5.1\libraries\ESP8266WebServer
Non usata: F:\documenti\Arduino\libraries\WebServer_tng-master
Più di una libreria trovata per "SD.h"
Usata: F:\documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.5.1\libraries\SD
Non usata: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\libraries\SD
exit status 1
Errore durante la compilazione per la scheda Generic ESP8266 Module.

```

jumpjack, 2019-05-19:
My projects compiles fine with core 2.5.0.

Hieromon, 2019-05-20:
OK. Please wait for the v0.9.9 release.

franknh23, 2019-05-20:
@Hieromon Geat. When we can expect for the new release?

stuffshit, 2019-05-20:
@Hieromon Geat. When we can expect for the new release?

Hieromon, 2019-05-20:
@stuffshit The main title of AutoConnect v0. 9. 9 is the implementation of the update feature but still needs some time.
I am tracking issues for core 2.5.1, on the board 2.5.2 release is foreseen in a short time. What is the reason you want to use core 2.5.1? 
Depending on the request, it may be worthwhile to prioritize for 2.5.1 supports. But as for me, I may forego support for 2.5.1. Many bugs are expected to be absorbed in 2.5.2.

stuffshit, 2019-05-21:
The reason is I always try to keep my libraries updated.
Good to know your plans regarding the library. Thanks for sharing!

Hieromon, 2019-05-21:
@stuffshit esp8266 core 2.5.2 has been released as expected. I started the evaluation such a version. 
The original SDFAT seems to be re-integrated but it needs validation. https://github.com/esp8266/Arduino/pull/6092
But I think compilation errors still occur. It is also worthwhile to support SPIFFS and SD unified in 2.5.1.
I decided to temporarily stop enhancing the update feature and give priority to 2.5.2 support. I expect to finish the work within a few weeks.
I'm sorry to keep you waiting.

jumpjack, 2019-05-22:
Maybe we could temprarily override this incompatibility by not including  AutoConnectUploadImpl.h in our projects unless strictly needed? Do I need it if I don't need to upload anything?

closed 2019-05-29 by Hieromon