Hieromon/AutoConnect #77: ESP32 no custom pages

ageurtse, 2019-05-18:
When i run the example where custom pages should be loaded, i don't get the custom pages. 
When i run the code on a ESP8266 everything is fine. 

Could this be the problem ?
[AC] Deserialize:NoMemory
[AC] Deserialize error:NoMemory

If so, how to solve it. 
These are my flash setting within Arduino.

[image]

Hieromon, 2019-05-18:
> [AC] Deserialize:NoMemory
> [AC] Deserialize error:NoMemory

The cause of this error is that ArduinoJson cannot allocate the JSON document buffer.
If the ESP32 module has [PSRAM](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/external-ram.html), AutoConnect orders ArudonoJson to allocate a JSON document buffer in PSRAM. The determination of whether "PSRAM is enabled" depends on the compilation option, not on whether it actually exists.
https://github.com/Hieromon/AutoConnect/blob/779bd164223ca8b4c20b3a65453c27c822d7329d/src/AutoConnectElementJson.h#L75
You specified the "PSRAM Enabled", but does your ESP32 module have PSRAM?

...I found such a topic, now.
https://github.com/espressif/arduino-esp32/issues/1613

You can check the psRAM existence by like this:
```cpp
if (!psramFound()){
  Serial.println("PSRAM not equipped");
  // ....
}
```
Also, I need to improve the library to use the psramFound function.

ageurtse, 2019-05-19:
After checking for PSRAM, there was nothing, so disabled the option in the compiler and now it is running. 
that was it, changed the setting for PSRAM Enabled to Disabled and now it works. 
Thanks a lot.

closed 2019-05-19 by ageurtse