Hieromon/AutoConnect #78: How to connect to free Wifi with access control?

jumpjack, 2019-05-18:
Some hotspots allow free connection without password... but then they ask for a password in a web page.
Can Autoconnect handle with this?

Hieromon, 2019-05-20:
It has nothing to do with AutoConnect.
Most free WiFi hotspots are implemented using the captive portal. Its principle is the same as AutoConnect. AutoConnect is not a library that interfaces with a specific web page with an HTML UI.

closed 2019-05-20 by Hieromon

jumpjack, 2019-05-20:
it's an "autoconnection-related issue", so I hoped... ;-)
I guess there would be need for cookie management to save login data, quite hard to implement. :-(

Hieromon, 2019-05-20:
Absolutely no, the login page responded by the hotspot is generated by the hotspot router or gateway. They will release IP addresses after a login process successful with fair order. Just because you don't need a password does not mean that they are not doing anything.
It is the role of the application layer but different implementations for each hotspot.
Also, cookies may not always be available under captive portal detection.

jumpjack, 2019-05-21:
I've seen that autoconnect "eavesdrops" traffic between smartphone and router, I'll see if I can add some instructions to read login data coming from smartphone to save them somewhere.

Hieromon, 2019-05-21:
Sorry, I cannot understand your point.
I’m willing to consider it, if what you're looking for fits with the AutoConnect library's mechanism.
However, AutoConnect is itself the [captive portal](https://en.wikipedia.org/wiki/Captive_portal) which implementation using DNS hijacking. The AutoConnect web page pops up when you access the ESP module AP from your smartphone because the smartphone OS internally performs an Internet connection detection process using http after WiFi connection established. Its mechanism implemented in the OS for the purpose of user protection and is opened in an ad hoc incognito mode window separate from the normal browser. By being open in incognito mode, there is no using the normal browser session or cookie information.
In other words, the ESP module running AutoConnect is in the same behavior as the hotspot.
The role of AutoConnect is to establish a data link with the WiFi access point, connect IP with the router and enable TCP transparency. In other words, the lower hierarchy from the transport layer is the scope of the library function.
The process after the IP address is assigned should be handled at the application layer. AutoConnect provides a custom web page, but its purpose is only to help the user sketch, not to care what the user sketch does.

> then they ask for a password in a web page.
>  I can add some instructions to read login data coming from smartphone to save them somewhere.

The quote you mentioned above is the web page that the hotspot is sending out, right?
If you want what is some automatic login facility to the hotspot, it needs to reproduce the hotspot's required handshaking procedure with http. It's like a script engine that executes some logic sequence descriptions and is truly a sketch application deployed outside the scope of the AutoConnect feature.
Of course, you can implement it the `on` handler of ESP8266WebServerClass, if you can handle everything you claim "traffic". I have no idea. Is not what you are referring to is a problem known as captive portal avoidance?

> I've seen that autoconnect "eavesdrops" traffic between smartphone and router,

I also cannot understand this well. What is the protocol of that traffic? Which session are the router and a smartphone? (Which network interface task?)
I don't think that what you have seen makes a lot of sense in implementing automatic login to hotspots. The reason I mentioned cookies is that hotspots often create their own TLS sessions (such as HTTPS, IMAPS, etc.) during the login sequence, and attempting to interrupt them results in security violations.

If you want to continue this discussion, I can reopen this topic, but please explain a solution which you can run a login session from your device running a captive portal to a higher captive portal without causing a security violation.
If the idea is that the ESP module will proxy in the http session the login sequence that the hotspot requires, it is the role of another library like curl, not AutoConnect.