Hieromon/AutoConnect #79: AutoConnect w/ ESP32 does automatically join saved wireless network all the times

r3na, 2019-05-24:
Using version 0.98 I am able to connect to a wireless network but after restarting the equipment, it does not join the network automatically all the times. From the logs/serial output I can get "networks found -2" or sometimes "-1". Then it creates the hotspot where I am able to join and force it to connect to the saved SSID. Using the same code (w/ minor changes) on a ESP8266 it automatically connects to the wireless network ALL the times. 
ps: I am the one with dozens of networks available (probably more than 30 or 40 in my area).

Hieromon, 2019-05-25:
Where there are a lot of WiFi radio signals mixed, the sequence up to [first WiFi.begin which not accompanied by SSID and password](https://hieromon.github.io/AutoConnect/lsbegin.html)  of the current AutoConnect may be insufficient.
ESP32 temporarily will disconnect the WiFi by switching channels. AutoConnect's default channel is number-1, but channel switching will occur when the access point's channel is different. If many WiFi channels are mixed at this time, ESP32 may not be able to be reconnection well. Such issues sometimes occur and are tracked on ESP32 repository.

Can you do the following tests to pursue this issue? None of them use AutoConnect. It verifies the behavior of WiFi.begin in a large number of WiFi radio environments.

```cpp
#include <WiFi.h>

wl_status_t waitForConnect(unsigned long timeout) {
  wl_status_t wifiStatus;

  Serial.print("Connecting");
  unsigned long st = millis();
  while ((wifiStatus = WiFi.status()) != WL_CONNECTED) {
    if (timeout) {
      if (millis() - st > timeout)
        break;
    }
    Serial.print('.');
    delay(300);
  }
  Serial.printf("%s\nIP:%s CH(%d)\n", wifiStatus == WL_CONNECTED ? "established" : "time out", WiFi.localIP().toString().c_str(), WiFi.channel());
  return wifiStatus;
}

void setup(){
  Serial.begin(115200);
  Serial.println();
  WiFi.softAPdisconnect(true);
  WiFi.mode(WIFI_STA);
  delay(100);
  if (!WiFi.config(IPAddress(0U), IPAddress(0U), IPAddress(0U), IPAddress(0U), IPAddress(0U)))
    Serial.println("WiFi config failed");
  WiFi.begin();
  waitForConnect(30000);
}

void loop() {
}
```

The above code is an excerpt part of AutoConnect::begin for first WiFi.begin. Please try running the above code with `Core Debug Level:Debug`. And check the occurrence of channel switching.
The most suspicious is the leading WiFi.config. It may be worthwhile to omit this.
Even if channel switching occurs, the core will dump the following log if the IP assignment is successful finally.
Could you post that log?
With that as a clue, I will start to study sequences that can be connected even in a crowded WiFi environment.
```
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 7 - STA_GOT_IP
```
Thank you for your cooperation.

r3na, 2019-06-04:
Hi, sorry for the delay. I did the test you asked.
At the logs you can see that randomly connects. When It was taking too long to connect I pressed the "restart" button and then randomly connects as you can see at the logs bellow:

`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x40080400,len:5868
entry 0x4008069c

[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
Connecting....[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:379] _eventCallback(): STA IP: 192.168.188.50, MASK: 255.255.255.0, GW: 192.168.188.1
established
IP:192.168.188.50 CH(1)
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x40080400,len:5868
entry 0x4008069c

[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
Connecting..........................................................................................................................................................................................................................................................ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x40080400,len:5868
entry 0x4008069c

[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
Connecting....[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:379] _eventCallback(): STA IP: 192.168.188.50, MASK: 255.255.255.0, GW: 192.168.188.1
established
IP:192.168.188.50 CH(1)
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x40080400,len:5868
entry 0x4008069c

[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
Connecting...............................................................................................................................................................................................................................ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x40080400,len:5868
entry 0x4008069c

[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
Connecting....[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:379] _eventCallback(): STA IP: 192.168.188.50, MASK: 255.255.255.0, GW: 192.168.188.1
established
IP:192.168.188.50 CH(1)
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x40080400,len:5868
entry 0x4008069c

[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
Connecting...........................................................................................................ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x40080400,len:5868
entry 0x4008069c

[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
Connecting....[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:379] _eventCallback(): STA IP: 192.168.188.50, MASK: 255.255.255.0, GW: 192.168.188.1
established
IP:192.168.188.50 CH(1)
`

Hieromon, 2019-06-05:
Thank you for your testing and the result reports.
It is still unclear, but the channel switching has not occurred and the cause of the problem is not due to it.
A similar phenomenon was posted on ESP32's repo. https://github.com/espressif/arduino-esp32/issues/234
https://github.com/espressif/esp-idf/issues/341 And unfortunately, the same phenomenon occurs with the esp-idf. https://github.com/espressif/esp-idf/issues/341

Here, I have one suggestion.
First, check the soldering of your ESP32 module. Especially the GND line stable is required.
https://github.com/espressif/arduino-esp32/issues/234#issuecomment-302447789
Then insert a delay. The place is as follows:
```cpp
void setup(){
  Serial.begin(115200);
  Serial.println();
  WiFi.softAPdisconnect(true);
  delay(1000);  // <-- here, insert additional delay
  WiFi.mode(WIFI_STA);
  delay(100);
  if (!WiFi.config(IPAddress(0U), IPAddress(0U), IPAddress(0U), IPAddress(0U), IPAddress(0U)))
    Serial.println("WiFi config failed");
  WiFi.begin();
  waitForConnect(30000);
}
```
However, there is no basis for this workaround. Even if it works well, I think it is not different from a magic spell.

Another possibility is router DHCP. The router always assigns 192.168.188.50. This means suggests that the connection with the MAC address of ESP32 is completed, but ESP32 may not capture the lease time of DHCP correctly. I recommend assigning a fixed IP to identify the source of the problem.
The problem is considered to be DHCP on the router side or DHCP client on ESP32 if the connection is always successful by the following modification:
```cpp
IPAddress local_IP(192, 168, 188, 50);
IPAddress gateway(192, 168, 188, 1);
IPAddress subnet(255, 255, 255, 0);
IPAddress primaryDNS(8, 8, 8, 8); //optional
IPAddress secondaryDNS(8, 8, 4, 4); //optional
void setup(){
  Serial.begin(115200);
  Serial.println();
  WiFi.softAPdisconnect(true);
  WiFi.mode(WIFI_STA);
  delay(100);
  if (!WiFi.config(local_IP, gateway, subnet, primaryDNS, secondaryDNS))
    Serial.println("WiFi config failed");
  WiFi.begin();
  waitForConnect(30000);
}
```
Can you try the above? The second test report from you will provide what we should do next. I will carry out a little more investigation.

Hieromon, 2019-09-07:
Since there was no response even after one month passed, closed this issue.

closed 2019-09-07 by Hieromon