Hieromon/AutoConnect #80: AutoConnect::begin with timeout but no SIID and no password?

LorenzoCerda, 2019-05-26:
Hello,

    First of all, thanks for such a nice work. I am trying to change the timeout of the initial WiFi.begin (which is performed by AutoConnect::begin, if I have understood correctly), but without using any SSID nor password in the AutoConnect::begin call. My case is that the esp32 turns on at the same time as the WLAN, so sometimes the 30 seconds (which is defined in AUTOCONNECT_TIMEOUT 30000) aren't enough for the WLAN to come active. 

  I also tried to redefine the macro value using:
#undef AUTOCONNECT_TIMEOUT            // un-define it
#define AUTOCONNECT_TIMEOUT 60000     // redefine it with the new value

but that didn't work either (it uses 30 seconds anyways). 

Just an extra info: for my application, it is ok to wait for a network (stored) until it finds it and, if not, enter in Portal "mode". The thing is that it can happen that the whole system resets (for ex: power failure) and when it comes back, I need to be sure that the esp32 will try to connect to the stored wifi networks. But, there are other situations where after some time, I need the Portal to be active (for example,add/erase a wifi network). 

  So, how should I proceed? Thank you in advance, best regards.

Hieromon, 2019-05-27:
@LorenzoCerda Thank you for your interest in AutoConnect.
In the problems you presented, about a timeout that is not effective.
```cpp
#undef AUTOCONNECT_TIMEOUT // un-define it
#define AUTOCONNECT_TIMEOUT 60000 // redefine it with the new value
```
In which source file did you write the above macro directive? The AUTOCONNECT_TIMEOUT expands in **AutoConnect.cpp**. Therefore, there is no effect unless it is included in the compilation of AutoConnect.cpp by **AutoConnectDefs.h**.

```cpp
// Time-out limitation when AutoConnect::begin [ms]
#ifndef AUTOCONNECT_TIMEOUT
// #define AUTOCONNECT_TIMEOUT     30000
#define AUTOCONNECT_TIMEOUT     60000
#endif // !AUTOCONNECT_TIMEOUT
```

Or you can specify the following in the sketch:

```cpp
AutoConnect portal;
portal.begin(nullptr, nullptr, 60000);
```

Hieromon, 2019-05-27:
@LorenzoCerda Here's my advice on the second question.
I see. I think that condition is a little complicated to maintain the connection.
So how about trying the following sequence?
```cpp
AutoConnect portal;
AutoConnectConfig  config;

void setup() {
  config.autoReconnect = true;
  portal.config(config);
  WiFi.disconnect(true); // <-- here is a trick.
  portal.begin(nullptr, nullptr, 60000);
}
```
1. Reconnection does not depend on the history stored inside esp32. It is in the role of AutoConnect.
2. Erase the history of esp32 side using WiFi.disconnect for the above behavior.
3. The first WiFi.begin will always fail, but it will be recovered with the autoReconnect option.
4. If automatic reconnection fails due to moving the access point, etc., the captive portal will start.

I think the portalTime and the retainPortal are not appropriate for the current situation. These two options are intended for running sketches even in offline situations.

LorenzoCerda, 2019-05-31:
Hi again,

Sorry for the delay, and thank you for your answers! Concerning them:

This: 
```
    AutoConnect portal;
    portal.begin(nullptr, nullptr, 60000);
```

  worked perfectly (using Config.autoReconnect = true also). 

Second advice:

  Oh.. I didn't know this trick (WiFi.disconnect(true);) and I haven't time to try it, but I will definitely will (and I will post here how did it go). 

  Thank you again for your help.

  Best regards

Hieromon, 2019-11-06:
Closed due to posting no activity reports.
Reopen with a new post.

closed 2019-11-06 by Hieromon