Hieromon/AutoConnect #82: error while compiling

igorlab, 2019-05-29:
Hello, I can't compile your example due to the error:

In file included from D:\WORK\Dropbox\Ardunio\!Programs\libraries\AutoConnect-master\src/AutoConnect.h:34:0,

                 from D:\WORK\Dropbox\Ardunio\!Programs\libraries\AutoConnect-master\examples\HelloWorld\HelloWorld.ino:26:

D:\WORK\Dropbox\Ardunio\!Programs\libraries\AutoConnect-master\src/AutoConnectAux.h:66:31: error: 'PageBuilder::UploadFuncT' has not been declared

   void  onUpload(PageBuilder::UploadFuncT uploadFunc) override { _uploadHandler = uploadFunc; }

                               ^

D:\WORK\Dropbox\Ardunio\!Programs\libraries\AutoConnect-master\src/AutoConnectAux.h:146:16: error: 'UploadFuncT' in 'class PageBuilder' does not name a type

   PageBuilder::UploadFuncT    _uploadHandler; /**< The AutoConnectFile corresponding to current upload */

                ^

D:\WORK\Dropbox\Ardunio\!Programs\libraries\AutoConnect-master\src/AutoConnectAux.h:66:9: error: 'void AutoConnectAux::onUpload(int)' marked 'override', but does not override

   void  onUpload(PageBuilder::UploadFuncT uploadFunc) override { _uploadHandler = uploadFunc; }

         ^

D:\WORK\Dropbox\Ardunio\!Programs\libraries\AutoConnect-master\src/AutoConnectAux.h: In constructor 'AutoConnectAux::AutoConnectAux(const String&, const String&, bool, AutoConnectElementVT)':

D:\WORK\Dropbox\Ardunio\!Programs\libraries\AutoConnect-master\src/AutoConnectAux.h:51:116: error: class 'AutoConnectAux' does not have any field named '_uploadHandler'

     _title(title), _menu(menu), _uriStr(String(uri)), _addonElm(addons), _handler(nullptr), _order(AC_EXIT_AHEAD), _uploadHandler(nullptr) { _uri = _uriStr.c_str(); _next.release(); _ac.release(); }

                                                                                                                    ^

D:\WORK\Dropbox\Ardunio\!Programs\libraries\AutoConnect-master\src/AutoConnectAux.h: In member function 'void AutoConnectAux::onUpload(int)':

D:\WORK\Dropbox\Ardunio\!Programs\libraries\AutoConnect-master\src/AutoConnectAux.h:66:66: error: '_uploadHandler' was not declared in this scope

   void  onUpload(PageBuilder::UploadFuncT uploadFunc) override { _uploadHandler = uploadFunc; }

                                                                  ^

D:\WORK\Dropbox\Ardunio\!Programs\libraries\AutoConnect-master\src/AutoConnectAux.h: In member function 'void AutoConnectAux::onUpload(T&)':

D:\WORK\Dropbox\Ardunio\!Programs\libraries\AutoConnect-master\src/AutoConnectAux.h:70:5: error: '_uploadHandler' was not declared in this scope

     _uploadHandler = std::bind(&T::upload, &uploadClass, std::placeholders::_1, std::placeholders::_2);

     ^nnectAux.h:146:16: error: 'UploadFuncT' in 'class PageBuilder' does not name a type

Thx

Hieromon, 2019-05-29:
The version of PageBuilder you are using is not consistent. Probably the index for PageBuilder of Arduino-IDE Library Manager has been lost. The consistent version of PageBuilder is 1.3.3.
https://github.com/Hieromon/PageBuilder/blob/2eeb8e01c75f34bc2def7da49c740061260bf304/src/PageBuilder.h#L175
Please install PageBuilder manually from the [repository](https://github.com/Hieromon/PageBuilder/tree/master) again without using Arduino IDE's Library Manager until it is restored.

Hieromon, 2019-06-11:
This topic is the same as the resolved issue #89 and closes along with it.

closed 2019-06-11 by Hieromon