Hieromon/AutoConnect #85: ESP32 No Captive Portal?

bazfp, 2019-06-01:
I'm running some simple code and I cannot get the captive portal to work.

It is accessible at `http://192.168.244.1/_ac` but I am not prompted on any device to access this page? I don't think the DNS is working correctly for captive portal?

Device: TTGO T-Beam ESP32

```
#include <WebServer.h>
#include <AutoConnect.h>

WebServer Server;
AutoConnect Portal(Server);

void setup() 
{
  delay(1000);
  Serial.begin(115200);
  Portal.begin();
  Serial.println("Starting web server");
}

void loop() 
{
  Portal.handleClient();
  // put your main code here, to run repeatedly:
}

```
Log:

```
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[E][WiFiSTA.cpp:211] begin(): connect failed!
[AC] WiFi.begin()
[AC] Connecting.....................................................................................................time out IP:0.0.0.0
[AC] SoftAP configure 192.168.244.1, 192.168.244.1, 255.255.255.0
[AC] SoftAP esp32ap/12345678 Ch(1) IP:192.168.244.1
[AC] http server started
[AC] DNS server started
[AC] Host:,URI:/,ignored
[E][WebServer.cpp:602] _handleRequest(): request handler not found
[PB] at leaving build: 266748 free
[AC] Host:192.168.244.1,URI:/,ignored
[E][WebServer.cpp:602] _handleRequest(): request handler not found
[AC] Detected application, ypmypnbwo, 0.0.0.0
[AC] Host:ypmypnbwo,URI:/,ignored
[E][WebServer.cpp:602] _handleRequest(): request handler not found
[AC] Detected application, jrowdoirylk, 0.0.0.0
[AC] Host:jrowdoirylk,URI:/,ignored
[E][WebServer.cpp:602] _handleRequest(): request handler not found
[AC] Detected application, imuicdkpddurvyh, 0.0.0.0
[AC] Host:imuicdkpddurvyh,URI:/_ac,generated:/_ac, allocated
[AC] Host:192.168.244.1,URI:/_ac,already allocated
[PB] at leaving build: 250248 free
[PB] Free heap:253256, content len.:6520
[PB] Res:200, Chunked:1
[E][WiFiClient.cpp:365] write(): fail on fd 60, errno: 104, "Connection reset by peer"
[AC] Host:192.168.244.1,URI:/_ac,already allocated
[AC] Host:192.168.244.1,URI:/_ac,already allocated
[PB] at leaving build: 252004 free
[PB] Free heap:255016, content len.:6520
[PB] Res:200, Chunked:1
[E][WiFiClient.cpp:365] write(): fail on fd 61, errno: 104, "Connection reset by peer"
[AC] Host:192.168.244.1,URI:/_ac,already allocated
[AC] Host:192.168.244.1,URI:/_ac,already allocated
[PB] at leaving build: 254092 free
[PB] Free heap:257108, content len.:6520
[PB] Res:200, Chunked:1
[E][WiFiClient.cpp:365] write(): fail on fd 62, errno: 104, "Connection reset by peer"
[AC] Host:192.168.244.1,URI:/_ac,already allocated
[AC] Host:192.168.244.1,URI:/_ac,already allocated
[PB] at leaving build: 253592 free
[PB] Free heap:256576, content len.:6520
[PB] Res:200, Chunked:1
```

bazfp, 2019-06-01:
It works for http requests but it would be nicer if I can get a platform specific captive portal detection. Is it somehow related to end point resolution?

https://socifi-doc.atlassian.net/wiki/spaces/SC/pages/94371841/DNS+Workaround+to+keep+Android+Splash+Page+and+the+Captive+Portal+Notification+active

bazfp, 2019-06-02:
I've solved it! I now get captive portal notifications and automatic forwarding to the captive portal on my Android devices (Samsung Galaxy S8) when I connect to the network. The solution was unintuitive but works.

Set the SoftAP apIP and apGatway to a WAN IP instead of default LAN 192.168.244.1. It probably works on other platforms too.

Recommend this should be the default (172.168.244.1 instead of 192.168.244.1)

AutoConnectDefs.h
```
#ifndef AUTOCONNECT_AP_IP
#define AUTOCONNECT_AP_IP 0x01F4A8AC  //*< 172.168.244.1 */
#endif // !AUTOCONNECT_AP_IP
#ifndef AUTOCONNECT_AP_GW
#define AUTOCONNECT_AP_GW 0x01F4A8AC  //*< 172.168.244.1 */
#endif // !AUTOCONNECT_AP_GW
```

[image]

Hieromon, 2019-06-03:
@bazfp Thank you for your information. This phenomenon is already discussed in arduino-esp32 core repositories. https://github.com/espressif/arduino-esp32/issues/1037
As per your suggestion, PR has been requested to change the SoftAP's default with the esp8266 core. https://github.com/esp8266/Arduino/pull/5529
But the PR is not accepted for now. In my view, I think d-v-a's claim to the PR is right and 172.168.xxx.xxx is not a private range. It is an issue that depends on the implementation of captive portal detection in a specific version of AndroidOS.
I cannot confirm the following option because I do not have any Android devices that experience this symptom, but have you tried the following options on your device? (Related issue #33)
Its option seems to be related to the address range for the captive portal detection.
https://github.com/tzapu/WiFiManager/issues/553#issuecomment-410577039

However, I will keep it open for users with a similar problem, as it may be useful as a temporary workaround for a particular mobile OS.

Hieromon, 2019-08-18:
@bazfp, I decided to solve this problem by taking your solution that according to [the PR](https://github.com/esp8266/Arduino/commit/adf2b14a6a1493b324d8e176a1519af76bd7e5fb). The following quote is causing.

> in order to get a captive notification to show or the popup to open, the DNS server must resolve to a public IP. It will not work with a private one (e.g. 192.168.4.1).
>
> On Android, a notification ("Register with Network") is displayed in the top left notification bar.
> On IOS, the login popup is displayed.

Default softAPIP and gateway IP for the captive portal activation have been changed to **172.217.28.1**. IP-range 172.217.0.0/16 has assigned to Google LLC so I used the range for new SoftAP IP to support newer Android.
Please try a new release as v0.9.12.

Hieromon, 2019-09-07:
Closes until a new phenomenon occurs caused by the SoftAP default IP has changed.
It will reopen if the issue recurs.

closed 2019-09-07 by Hieromon