Hieromon/AutoConnect #86: Soft_AP

Grutsch, 2019-06-05:
I have a sketch were the customer can decide if he joins network or not. I did implement this question with the AutoElements. If he answers yes then it will use the new AP credentials. If the answer is no it should start in SoftAP Mode. Is this possible?

Alex

Hieromon, 2019-06-05:
I would like to make your requirements more clearly.
Does it mean that the connection can establish by the current credential but it not actually connect yet?
Can we know if WiFi.begin () will succeed before the  WiFi.begin? Is it synonymous with that?

Grutsch, 2019-06-05:
Like I mentioned, I want two possibilities:
1. Join existing network with credentials. This is already working with autoconnect.
2. If I set a parameter in my sketch e.g. standalone the esp is working as AP and autoconnect is just providing Information and custom webpages

Hieromon, 2019-06-05:
It is still vague, but I think it is possible.

> If I set a parameter in my sketch e.g. standalone the esp is working as AP

What kind of behavior do you want? Is it acting as the simple web server featuring custom web pages, not as a captive portal?
If so you can use the [immediateStart](https://hieromon.github.io/AutoConnect/apiconfig.html#immediateStart) option accompanied by the [onDetect exit handler](https://hieromon.github.io/AutoConnect/advancedusage.html#captive-portal-start-detection). If the **onDetect** handler returns the false, start SoftAP by suppressing the captive portal. Additionally, depending on your request you may need to force the erasure of previous credentials stored by the ESP module via WiFi.disconnect(true). But I do not know your situation, so I have no idea if it is necessary. It will probably be something like as following:
```cpp
AutoConnect portal;
void setup() {
  portal.onDetect([](IPAddress ip) {
    return false;
  });
  config.immediateStart = true;
  portal.config(config);
  WiFi.disconnect(true);  // <-- if necessary
  delay(1000);  // Wait for disconnect
  portal.begin();
}
```

However, I cannot understand how it consistent with the quotation below you said:
> If the answer is no it should start in SoftAP Mode.

Because the ESP module is already in SoftAP mode to prompt the custom Web page for the dialog as you said:

> I did implement this question with the AutoElements.

Grutsch, 2019-06-06:
Thanks for your quick respond. The sketch is controlling a heater with build in led. All is controlled via a webserver (webpage) with websocket connection for quick response. I want to be able to have the ESP32 to work on its own (AP Mode) or to join a network. Anyhow I will try what you suggested.
Alex

computergeek1507, 2019-06-16:
I have a similar requirement. When the captive portal is active I would like my custom webpages to be active and also allow the loop method to preform an action.

Hieromon, 2019-06-16:
@computergeek1507, thank you for your suggestion.
You want the ESP module to act as a simple access point and optionally make a WiFi connection request (from the AutoConnect menu), right? And during that time, custom Web pages are also valid.
Is the PR #91 an implementation of your requirement?, If so, how is this PR different from the suppression method of the captive portal by non-start of spoofed DNS that returns false in the onDetect handler I showed?

jssahoo, 2019-08-26:
When captive portal is activated(i.e no wifi credential is saved) no other calling functions in loop() does not work. Why??

Hieromon, 2019-08-27:
AutoConnect does not return control to the sketch until WiFi connected or times out. See the [Inside AutoConnect::begin](https://hieromon.github.io/AutoConnect/lsbegin.html) chapter in the documentation.
If you call AutoConnect::begin in the loop() and do WiFi connection operation, you need to catch the transition of status to WL_DISCONNECTED.
Similar issue https://github.com/Hieromon/AutoConnect/issues/93

closed 2019-08-27 by Hieromon