Hieromon/AutoConnect #88: Does it work with Arduino + ESP8266 combo through software serial?

hk89, 2019-06-08:
Title pretty much says it! Does it?

Hieromon, 2019-06-10:
What for software serial? Do you want to control the ESP8266 using AT commands with NonOS_SDK via software serial from other Arduino boards?
If so, you can't.

willfaulds, 2019-06-18:
I think this question is about whether you can use this library with an esp8266 based board interfaced with a classic Arduino board e.g. UNO. The idea being run main code on the UNO and the gui/iot aspect on the esp8266. Use software serial to interface the 2 boards. @hk89 that what you’re asking?

Hieromon, 2019-08-19:
Since there was no response even after one month passed, closed this issue.
When some reports are posted, reopen it.

closed 2019-08-19 by Hieromon