Hieromon/AutoConnect #89: Error compiling for Wemos Mini

Andreadj, 2019-06-10:
Sorry, I'M new to ESP8266 and compiling and programming, maybe it's my fault but I've got error in compiling simple sketch:

Arduino:1.8.9 (Windows Store 1.8.21.0) (Windows 10), Scheda:"LOLIN(WEMOS) D1 R2 & mini, 160 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, All Flash Contents, 115200"

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware -hardware E:\Privato\Documenti\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools E:\Privato\Documenti\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\libraries -libraries E:\Privato\Documenti\Arduino\libraries -fqbn=esp8266:esp8266:d1_mini:xtal=160,vt=flash,exception=disabled,ssl=all,eesz=4M1M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=all,baud=115200 -ide-version=10809 -build-path C:\Users\Ufficio\AppData\Local\Temp\arduino_build_295849 -warnings=none -build-cache C:\Users\Ufficio\AppData\Local\Temp\arduino_cache_938154 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=E:\Privato\Documenti\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-3-20ed2b9.path=E:\Privato\Documenti\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.python.path=E:\Privato\Documenti\ArduinoData\packages\esp8266\tools\python\3.7.2-post1 -prefs=runtime.tools.python-3.7.2-post1.path=E:\Privato\Documenti\ArduinoData\packages\esp8266\tools\python\3.7.2-post1 -prefs=runtime.tools.mkspiffs.path=E:\Privato\Documenti\ArduinoData\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs-2.5.0-3-20ed2b9.path=E:\Privato\Documenti\ArduinoData\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -verbose E:\Privato\Documenti\Progetto Led\Nuova cartella\AutoConnect-master\examples\Simple\Simple.ino
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware -hardware E:\Privato\Documenti\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools E:\Privato\Documenti\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\libraries -libraries E:\Privato\Documenti\Arduino\libraries -fqbn=esp8266:esp8266:d1_mini:xtal=160,vt=flash,exception=disabled,ssl=all,eesz=4M1M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=all,baud=115200 -ide-version=10809 -build-path C:\Users\Ufficio\AppData\Local\Temp\arduino_build_295849 -warnings=none -build-cache C:\Users\Ufficio\AppData\Local\Temp\arduino_cache_938154 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=E:\Privato\Documenti\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-3-20ed2b9.path=E:\Privato\Documenti\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.python.path=E:\Privato\Documenti\ArduinoData\packages\esp8266\tools\python\3.7.2-post1 -prefs=runtime.tools.python-3.7.2-post1.path=E:\Privato\Documenti\ArduinoData\packages\esp8266\tools\python\3.7.2-post1 -prefs=runtime.tools.mkspiffs.path=E:\Privato\Documenti\ArduinoData\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs-2.5.0-3-20ed2b9.path=E:\Privato\Documenti\ArduinoData\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -verbose E:\Privato\Documenti\Progetto Led\Nuova cartella\AutoConnect-master\examples\Simple\Simple.ino
Using board 'd1_mini' from platform in folder: E:\Privato\Documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2
Using core 'esp8266' from platform in folder: E:\Privato\Documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2
Detecting libraries used...
"E:\\Privato\\Documenti\\ArduinoData\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IE:\\Privato\\Documenti\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/include" "-IE:\\Privato\\Documenti\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/lwip2/include" "-IE:\\Privato\\Documenti\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\Ufficio\\AppData\\Local\\Temp\\arduino_build_295849/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK221=1 -DF_CPU=160000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10809 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_WEMOS_D1MINI\"" -DFLASHMODE_DIO -DESP8266 "-IE:\\Privato\\Documenti\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.2\\cores\\esp8266" "-IE:\\Privato\\Documenti\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.2\\variants\\d1_mini" "C:\\Users\\Ufficio\\AppData\\Local\\Temp\\arduino_build_295849\\sketch\\Simple.ino.cpp" -o nul
"E:\\Privato\\Documenti\\ArduinoData\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IE:\\Privato\\Documenti\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/include" "-IE:\\Privato\\Documenti\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/lwip2/include" "-IE:\\Privato\\Documenti\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\Ufficio\\AppData\\Local\\Temp\\arduino_build_295849/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK221=1 -DF_CPU=160000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10809 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_WEMOS_D1MINI\"" -DFLASHMODE_DIO -DESP8266 "-IE:\\Privato\\Documenti\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.2\\cores\\esp8266" "-IE:\\Privato\\Documenti\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.2\\variants\\d1_mini" "-IE:\\Privato\\Documenti\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.2\\libraries\\ESP8266WiFi\\src" "C:\\Users\\Ufficio\\AppData\\Local\\Temp\\arduino_build_295849\\sketch\\Simple.ino.cpp" -o nul
"E:\\Privato\\Documenti\\ArduinoData\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IE:\\Privato\\Documenti\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/include" "-IE:\\Privato\\Documenti\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/lwip2/include" "-IE:\\Privato\\Documenti\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\Ufficio\\AppData\\Local\\Temp\\arduino_build_295849/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK221=1 -DF_CPU=160000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10809 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_WEMOS_D1MINI\"" -DFLASHMODE_DIO -DESP8266 "-IE:\\Privato\\Documenti\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.2\\cores\\esp8266" "-IE:\\Privato\\Documenti\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.2\\variants\\d1_mini" "-IE:\\Privato\\Documenti\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.2\\libraries\\ESP8266WiFi\\src" "-IE:\\Privato\\Documenti\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.2\\libraries\\ESP8266 [... cut ...]
Uso la libreria SD alla versione 2.0.0 nella cartella: E:\Privato\Documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\libraries\SD 
Uso la libreria SDFS alla versione 0.1.0 nella cartella: E:\Privato\Documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\libraries\SDFS 
Uso la libreria ESP8266SdFat alla versione 1.0.16 nella cartella: E:\Privato\Documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266SdFat 
exit status 1
Errore durante la compilazione per la scheda LOLIN(WEMOS) D1 R2 & mini.

Hieromon, 2019-06-10:
Same as #82 
Please re-install latest [PageBuilder 1.3.3](https://github.com/Hieromon/PageBuilder/releases/latest)

Andreadj, 2019-06-11:
Already did it, nothung changed. I'll retry...

Hieromon, 2019-06-11:
Please verify the message output from the compiler. You used the PageBuilder 1.3.2, not 1.3.3 allowed the compiler message as follows:
```
Uso la libreria PageBuilder alla versione 1.3.2 nella cartella: E:\Privato\Documenti\Arduino\libraries\PageBuilder
```
PageBuilder can be installed without using the Arduino-IDE Library Manager. You can download .ZIP from the [PageBuilder GitHub repository](https://github.com/Hieromon/PageBuilder) and can import it to Arduino IDE immediately. After downloaded, the PageBuilder-master.zip file will be saved in your download folder. Then in the Arduino IDE, navigate to "Sketch > Include Library". At the top of the drop-down list, select the option to "Add .ZIP Library...". Details for [Arduino official page](https://www.arduino.cc/en/Guide/Libraries#toc4).

Andreadj, 2019-06-11:
Hi Hieromon,
sorry you were right I had 3 different libraries, erased the old ones it Worked!
Thanks a lot, very kind!!!

closed 2019-06-11 by Andreadj