Hieromon/AutoConnect #93: Question: Always start AP and connect to last one

SimonKlausLudwig, 2019-06-17:
Is it possible to always start an AP but at the same time connect to the last known AP ?

Hieromon, 2019-06-18:
@SimonKlausLudwig, Thank you for the intention of your requirement.
Recently, there has been an increasing demand for the coexistence of station operation while using the ESP module itself as an access point.
I still can not fully understand those requirements. If I can understand the essence of these requirements, I will extend the library without inconsistencies in the current functionality.
However, it is not possible to inadvertently capture a combination of actions that are only valid in certain situations. Incorporating actions that conflict with existing features can lead to loss of feature integrity.

The utility of AutoConnect is that you do not have to hard code the AP. WIFI_AP_STA mode of ESP8266 and ESP32 is spoofed access points sharing the same radio circuit and channel. It is unreliable for constant use. So I chose to limit AP_STA activation to the momentary means of connecting to the actual access point.
But people want to ESP module be in AP_STA mode and keep as an access point. An issue #86 is also claiming a similar request to you.

> Try to connect to a hardcoded Wifi Accesspoint, let's say. MY_COMPANY_AP and open an
AP regardless of successful connection or not.

Are you saying "open an AP" is the ESP module in AP mode (or AP_STA mode)? If so, what is the purpose of keeping the ESP module as an access point despite establishing a connection with a hard-coded access point (you call it MY_COMPANY_AP)? 
Keeping AP mode means that there are several WiFi clients that try to connect directly to the ESP module. What is the purpose of connecting directly to the ESP module from another WiFi client device (e.g. smartphone) despite the fact that the ESP module is already joined to the WLAN?
And if channel switching occurs at this time, the connection with the original AP (MY_COMPANY_AP) will be disconnected.
Are you thinking that you can drive the captive portal even under these circumstances and can select the access point and the AutoConnect menu will pop up?

> The user chooses his AP let's say CUSTOMER_AP
> On next restart try first to connect to CUSTOMER_AP and start the config portal at the same time.

About your above intention, What requirements cannot be achieved with code like the following?

```cpp
#include <WiFi.h>
#include <WebServer.h>
#include <AutoConnect.h>

AutoConnect portal;

const char* SSID = "MY_COMPANY_AP";
const char* PASS = "MY_COMPANY_PASS";
bool CUSTOMER_AP = false;

void storeAP() {
  // Here, records that CUSTOMER_AP is selected to EEPROM.
  CUSTOMER_AP = true;
}

bool isStored() {
  // Here, returns that CUSTOMER_AP is selected indicator from EEPROM.
}

void setup() {
  if (isStored()) {
    SSID = nullptr;
    PASS = nullptr;
  }
  portal.begin(SSID, PASS);
}

void loop() {
  portal.handleClient();
  if (WiFi.status() == WL_CONNECTED) {
    if (!CUSTOMER_AP) {
      if (WiFi.SSID() != String("MY_COMPANY_AP"))
        storeAP();
    }
  }
  else
    portal.begin(SSID, PASS);
}
```
1. If a connection to MY_COMPANY_AP is successful, AP mode will not be started (as it is not necessary). You can use Config new and Open SSIDs in the AutoConnect menu even if you are not in AP mode.

2. If a connection fails, the captive portal will launch. Here again, you can use Config new and Open SSIDs.

In either case, you can select CUSTOMER_AP. (However, pre-registration of credentials may be required, but the library function for it has been implemented)

Hieromon, 2019-07-17:
Since there was no response even after one month passed, closed this issue.
When some reports are posted, reopen it.

closed 2019-07-17 by Hieromon