Hieromon/AutoConnect #94: Retrieving the SSID from saved credentials

rgonzalez214, 2019-06-19:
I'm trying to use AutoConnect to set up my esp to a wifi on very first boot.
Looking through the examples I could only find those relating to the html page controls.
Do you have any example that shows how to retrieve the current SSID in use.
My goal is to send an mqtt value only if there is a valid SSID.. and not the default SSID "ESPXXX".
I attempted the AutoConnectCredential function with no luck.
Thanks

Hieromon, 2019-06-20:
You can use the [WiFi.SSID](https://arduino-esp8266.readthedocs.io/en/2.5.2/esp8266wifi/station-class.html#ssid) function if you only want to know the SSID of the access point you have currently established.
However, if you want to enable the captive portal while avoiding connection with a specific SSID, you can try the following logic sequence:
```cpp
AutoConnect portal;
void setup() {
  if (portal.begin()) {
    if (WiFi.SSID() == String("ESPXXX")) {
      WiFi.disconnect();
      while (WiFi.status() == WL_CONNECTED) delay(100);
      portal.begin(); // Or, ESP.restart();
    }
  }
}
```

Hieromon, 2019-09-07:
Since there was no response even after one month passed, closed this issue.

closed 2019-09-07 by Hieromon