Hieromon/AutoConnect #97: About Deletion of certificates

MingKoo, 2019-06-25:
After the first run of the program, the certificate can be obtained, However, after reboot the ESP8266, But the certificate that have been deleted are still maintained. How did it ? Now, count is ZERO.What should I do? Thanks.
The code is:
void delCredential() {
AutoConnectCredential ac(CREDENTIAL_OFFSET);
//AutoConnectCredential();
struct station_config entry;
String content = "";
uint8_t count = ac.entries();
Serial.println(" ");
Serial.print("Wifi Credentials: ");
Serial.println(count);
for (int8_t i = 0; i < count; i++) { // Loads all entries.
ac.load(i, &entry);
content += String((char *)entry.ssid);
ac.del((char *)entry.ssid);
Serial.print("SSID: ");
Serial.print(content);
Serial.println(" is deleted!");
}
Serial.println("ESP RESET...... ");
ESP.reset();
}

MingKoo, 2019-06-26:
I have solved this problem. Reference resources:  #94 
   ###[https://github.com/Hieromon/AutoConnect/pull/91](url)  
The new code is:
void delCredential() {
  AutoConnectCredential  ac(CREDENTIAL_OFFSET);
  //AutoConnectCredential();
  struct station_config  entry;
  String content = "";
  uint8_t  count = ac.entries();
  Serial.println(" ");
  Serial.print("Wifi Credentials: ");
  Serial.println(count);
  for (int8_t i = 0; i < count; i++) {    // Loads all entries.
    ac.load(i, &entry);
    content += String((char *)entry.ssid);
    ac.del((char *)entry.ssid);
    Serial.print("SSID: ");
    Serial.print(content);
    Serial.println(" is deleted!");
  }
  Serial.println("ESP RESET...... ");
  WiFi.disconnect();
  while (WiFi.status() == WL_CONNECTED) delay(100);
    portal.begin(); // Or, ESP.restart();
  Serial.println("ESP RESET......WAITTING FOR SETUP ");
  delay(3000);
  }

Now, after rebooting, it automatically enters SOFTAP mode.I tested it many times and it worked well. Perfect!
I set a button to allow users to reset the parameters so that ESP can enter the SOFTAP state.Equivalent to a return to factory mode.
Thank you very.

Hieromon, 2019-06-27:
That's fine @MingKoo. For your reference:
The ESP8266 will save the previous SSID and password in the flash by behaves of SDK. This area is a different place from the credential data area that AutoConnect stores.
AutoConnect uses this behavior to provide the following options for ESP8266 reconnection control.

1. [AutoConnectConfig::immediateStart](https://hieromon.github.io/AutoConnect/apiconfig.html#immediatestart)
Transit to SoftAP mode forcibly and start the captive portal, ignoring the SSID and password which previously established a connection that ESP8266 (SDK) stored. The SSID and password saved by ESP8266 (SDK) are not cleared.

2. [AutoConnectConfig::autoReconnect](https://hieromon.github.io/AutoConnect/apiconfig.html#autoreconnect)
AutoConnect::begin attempts to reconnect using the saved credentials of AutoConnect when the first WiFi.begin behaving inside will fail. At this time, if the SSID and password are provided to AutoConnect::begin as the parameter, AutoConnect will erase the SSID and password stored in ESP8266 by executing WiFi.disconnect before WiFi.begin; that is, erase the established SSID stored in the ESP8266 flash and then load the SSID from the AutoConnect credentials area.

A description of the logic sequence of AutoConnect:: begin can be found in [Inside AutoConnect::begin](https://hieromon.github.io/AutoConnect/lsbegin.html). (It lacks the description of the execution condition of WiFi.disconnect. I will revise a little.)

So can you close this topic?

MingKoo, 2019-06-27:
Thanks you replay.

closed 2019-06-27 by MingKoo