Hieromon/AutoConnect #98: Will this function be available in the future?Like led Ticker.

MingKoo, 2019-06-28:
What is the way to achieve the LED effect be real-time in parallel? Like Ticker or TaskScheduler. For example, How to represents the current state ?  SOFTAP mode / access the User's Wifi network.   I've learned to look at all the instructions . Actually related to portal. onDetect ().  About LED effects such as: flash, FADE. 
I think in which mode is the current for users, Using LED lights to indicate is a very useful function. Because, Mode switching and implementation, Useing  serial port to print  , is only a programming mode. What's your opinion?Will this function be available in the future?

MingKoo, 2019-06-29:
I modified your autoconnect.cpp using Task Scheduler[https://github.com/arkhipenko/TaskScheduler?utm_source=platformio&utm_medium=piohome](url). 
Although it's not perfect enough to represent each state. But My need is :   In the soft AP state, I made the LED flicker. Everything seems to be working well.

Hieromon, 2019-06-29:
It seems that my reply was a bit late. I was studying the code to implement your proposal after looking at your post. Come to think of it, maybe you are right. I did not notice that need until your post.
It is a useful gimmick to indicate the state of WiFi mode transition with several LED flicker patterns. Flashing LED may be important for the ESP module built into some molding devices.

It can also be implemented using the TaskScheduler as you have chosen, but just only for the blinking LED I can have other implementations that are lighter than including the TaskScheduler in AutoConnect.
I will incorporate this feature in v0.9.11 and prepared a draft specification that shows the three WiFi statuses as follows:
- The ESP module stays in APSTA mode, and the captive portal is active.
- WiFi connection with access point established and data link enabled. (ie WiFi.status = WL_CONNECTED)
- No STA connection state. (ie WiFi.status != WL_CONNECTED)

Each state is identified by changing the flicker cycle. Short blink cycle in AP stay state, Short-on and long-off in WiFi-disconnected state, Turn off in WiFi-connected state. However, the ESP module consumes a lot of power, so LED flicker is optional.
Does the above draft meet your requirements?

MingKoo, 2019-06-30:
Perfect 
I'm glad you accepted my suggestion. I'm looking forward to updating the new version. I think is :  
bool AutoConnect::_indicate(bool enable)
void AutoConnect::_setindicate(int speed_open,int speed_close)

Thanks !

Hieromon, 2019-07-09:
@MingKoo, I staged up RC version of ticker supports to the develop branch named [Enhance/v0911](https://github.com/Hieromon/AutoConnect/tree/Enhance/v0911). You can evaluate the flickering behavior with an external LED.
The outline of the specification is as follows:

1. Specifies the ticker activation
**AutoConnectConfig**
    ```cpp
    bool AutoConnectConfig::ticker
    uint8_t AutoConnectConfig::tickerPort
    uint8_t AutoConnectConfig::tickerOn
    ```
    - **AutoConnectConfig::ticker**
    Specifies ticker activation. The **true** activates the ticker.
    - **AutoConnectConfig::tickerPort**
    Specify the GPIO port number to output the flicker signal of the ticker. The default assumes on the board dependent definition **LED_BUILTIN** macro.
    - **AutoConnectConfig::tickerOn**
    Specifies the logic level of the flicker signal. If the LED attached as a ticker is the active-low, specify **LOW** for this option. In the opposite case, specify **HIGH**. The default is LOW.

2. Flicking cycle time
The values that determine the flickering cycle time of the flicker are defined in `AutoConnectDefs.h`.
    - Period [ms]
    https://github.com/Hieromon/AutoConnect/blob/5ddd49c42a8d9363cbe9268fbe3132a7cd6c2ea7/src/AutoConnectDefs.h#L142

    - Duty rate in the WiFi_AP mode (256 resolution), ie. 128 is 50%.
    https://github.com/Hieromon/AutoConnect/blob/5ddd49c42a8d9363cbe9268fbe3132a7cd6c2ea7/src/AutoConnectDefs.h#L145

    - Duty rate while in WiFi disconnected (256 resolution)
    https://github.com/Hieromon/AutoConnect/blob/5ddd49c42a8d9363cbe9268fbe3132a7cd6c2ea7/src/AutoConnectDefs.h#L148

3. Usage
    ```cpp
    AutoConnect portal;
    AutoConnectConfig config;

    void setup() {
      config.ticker = true;
      portal.config(config);
      portal.begin();
    }

    void loop() {
      portal.handleClient();
    }
    ```
    Connect the LED to the LED_BUILTIN port of your board. (If you connect externally, don't forget the protection resistance around 200 ohms)
    The ticker LED flashes according to the WiFi connection status.

If possible, please you test Enhance/v0911 and return the results. After that, I will prepare the document and release the branch.
Thank you for your contribution.

MingKoo, 2019-07-11:
I'm sorry. I responded slowly. I just tried. Everything is perfect! Thank you. This is a great update! Thank you!

Hieromon, 2019-07-17:
Closed due to release a v0.9.11.

closed 2019-07-17 by Hieromon